realsenseai/librealsense #9550: Out of frame resources  on  android

dota2mhxy, 2021-08-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  android 8  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  android 8  |
| SDK Version                            |  {legacy 2                         | 
| Language                            |  java }                          | 
| Segment			| Smartphone     | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello i have a problem now   
[image]
I searched this problem first and found that issues 9371 has a solution.  frame.close();
I used it and found it to be invalid
[image]
Thank you for your reply

MartyG-RealSense, 2021-08-02:
Hi @dota2mhxy  The RealSense SDK's Java example program for the Android wrapper takes the approach of releasing frames from the pipeline using a **try* statement*.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/app/src/main/java/com/example/realsense_java_example/MainActivity.java#L93

dota2mhxy, 2021-08-03:
@MartyG-RealSense thanks  for  your  help  ，i see  an parameter called frame rate  ，how  to   adjust it  in  android

MartyG-RealSense, 2021-08-03:
I could only find a Java example  - **Capture** - for setting the resolution rather than the frame rate (frames per second / FPS).  Without defining a custom FPS, it would likely default to a rate of 30 FPS. 

**config.enableStream(StreamType.DEPTH, 640, 480);**

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java#L163

closed 2021-08-04 by dota2mhxy