realsenseai/librealsense #9552: About OpenGL using realsense d455 to achieve 3D display

zma69650, 2021-08-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   |  05.12.13.50 | 
| Operating System & Version |  win10 | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC  |
| SDK Version                            | 2.45.0                          | 
| Language                            | c++| 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I compared rs-pointcloud and realsense-viewer, and found that the latter has a better display effect. I want to know the difference between the two and how to use OpenGL to achieve this effect or the key technology to achieve it. In addition, I want to use openni2 to read data, is this feasible?

MartyG-RealSense, 2021-08-02:
Hi @zma69650  Yes, the RealSense SDK does support OpenNI2 through the compatibility wrapper in the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2

The output of the RealSense Viewer tends to look improved compared to other RealSense applications because it applies a range of depth colorization settings and post-processing filters by default.   Applications do not have these filters and settings by default and they have to be deliberately incorporated into scripting in order to replicate the appearance of the Viewer's output.

More information about post-processing filters is provided here:

https://dev.intelrealsense.com/docs/post-processing-filters
https://dev.intelrealsense.com/docs/depth-post-processing

As the RealSense SDK 2.0 is open-source, you have the option of customizing the code of realsense-viewer into your own design if you prefer not to create a completely new application.

https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

In regard to programming OpenGL (implemented via **glfw3** in the RealSense SDK), the guidance in the link below may be helpful.

https://github.com/IntelRealSense/librealsense/issues/2504

zma69650, 2021-08-03:
Thank you for your answer. In addition, do I need to set the depth image to align with the color image in openni? I set the frame rate to be too slow before, is there any solution?

zma69650, 2021-08-03:
There is another question. I need to use openni and nite to implement user tracking, and also retain the 3D effect like relasense-viewer. Is there any solution?

zma69650, 2021-08-03:
Hi@MartyG-RealSense I am using nite2 for user tracking. It uses openni2 to read data. Can I use openni and rs pipes to read data at the same time?

MartyG-RealSense, 2021-08-03:
As far as I am aware it is not mandatory to use depth to color alignment in the OpenNI2 wrapper if you do not require alignment.  Some OpenNI example programs **(SimpleDepth** and **SimpleColor** that show depth only or color only are listed at the link below.

https://www.stereolabs.com/docs/openni2/#linux-examples

A factor that should be considered when choosing the OpenNI2 wrapper for RealSense development is that the wrapper has not been updated in some years (it was created by a group outside of Intel) and although Intel maintain the wrapper in a working As-Is state, further features and performance improvements for this wrapper should not be expected in future.

There is a NITE2 example program at the link below.

https://github.com/UnaNancyOwen/NiTE2Sample

 I am not familiar enough with NITE2 to comment on what may be possible with it though.

closed 2021-08-06 by zma69650