realsenseai/librealsense #9554: realsense-viewer doesn't build on Mac

Timmmm, 2021-08-02:
Using `master` as of today, you cannot build `realsense-viewer` on Mac. I followed the instructions but there are two issues:

1. They don't tell you to install openssl. This is required for libcurl. Easily fixed by `brew install openssl`, however
2. OpenSSL isn't added to the default CMake path due to conflicts with the system libraries (I think), so when you build realsense-viewer in Xcode you get this error when it tries to build `libcurl`.

```
CMake Error at /usr/local/Cellar/cmake/3.21.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
```

It isn't enough to `export OPENSSL_ROOT_DIR=/usr/local/opt/openssl` before you run `cmake` for the first time because the build process itself runs CMake again from within Xcode where that environment variable isn't set.

Timmmm, 2021-08-03:
I'm trying to build the realsense-viewer from source because you don't provide Mac binaries as far as I call tell (it would be great if you did btw!).

librealsense itself compiled fine because it doesn't depend on libcurl/OpenSSL.

This is on an Intel Mac.

MartyG-RealSense, 2021-08-04:
The only reference that I could find regarding the location of realsense-viewer files on Mac is in the link below.

https://github.com/IntelRealSense/librealsense/issues/2811#issuecomment-443909901

Timmmm, 2021-08-08:
Oh ok I didn't know you could install it via `brew`. You guys should really update [this page](https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk) which tells you to compile from source!

This issue probably still needs fixing but it isn't urgent.

MartyG-RealSense, 2021-08-08:
I have added a **Documentation** tag to this case so that your suggestion about adding information to the official Mac installation instructions about the option to install with **brew install librealsense** can be considered by the RealSense team.  This case should be kept open in the meantime, but you do not need to take any further action about this.  Thanks again!

mfkenson, 2021-11-02:
@Timmmm here is the answer of your original question 

after 
`brew install openssl`

and before cmake you may want to set these 2 env var.
You may have different path due to different version of openssl but they should be similar.

```
export OPENSSL_ROOT_DIR=/usr/local/Cellar/openssl@1.1/1.1.1l_1/
export OPENSSL_INCLUDE_DIR=/usr/local/Cellar/openssl@1.1/1.1.1l_1/include
```

I have to build from source because I need the python bindings.
```
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
 -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/opt/anaconda3/envs/rs2env/bin/python
```

MartyG-RealSense, 2021-11-02:
Thanks very much @mfkenson for your advice to @Timmmm  :)

TroikaTronix, 2022-01-17:
While attempting to build the entire librealsense project (v2.49.0) in Xcode on macOS Big Sur, none of the suggestions above worked for me.

I did however discover a workaround that allowed me compile the entire project.

1. Click on the libcurl subproject
2. Click on "Build Phases" tab
3. Add the following line to the the very top of all of the post-build scripts whose title follows the form:"Generate libcurl/src/libcurl-XXXXX/$CONFIGURATION)$(EFFECTIVE_PLATFORM_NAME)/libcurl-XXXXX 

export OPENSSL_ROOT_DIR=**/PATH/TO/OPENSSL**

where /PATH/TO/OPENSSL is the path to OpenSSL on your machine. I had installed OpenSSL with brew, so my path was /usr/local/opt/openssl

After making this change, selecting ALL_BUILD and building built everything properly.

MartyG-RealSense, 2022-01-17:
Thanks so much @TroikaTronix for sharing your own workaround with the RealSense community   :)

TroikaTronix, 2022-01-17:
P.S. If someone knows how to modify the CMakeLists.txt scripts to do this automatically, please post the solution here. (@mfkenson, do you have any ideas?)

MartyG-RealSense, 2026-10-23:
This open GitHub issue is being closed because the realsense-viewer tool has been enabled for MacOS builds.

closed 2026-10-23 by MartyG-RealSense