realsenseai/librealsense #9555: No device detected. Is it plugged in?

vusdoo, 2021-08-03:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.14.50) | 
| Operating System & Version |   {Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  ( 5.4.0-80-generic )                                         | 
| Platform                                 |  NUC10i3FNK  |
| SDK Version                            |  { 2.48.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
issue / 
$ lsusb | grep 8086
Bus 002 Device 003: ID 8086:0b07 Intel Corp. 
$ ls -al | grep /dev/video
(empty)
$ realsense-viewer 
please connect  intel realsense device
$ rs-enumerate-devices 
No device detected. Is it plugged in?

question / 
1. How to downgrade (2.48 -> 2.47)
2. how it works(2.48)

vusdoo, 2021-08-03:
What is the solution to secure boot enable?

MartyG-RealSense, 2021-08-03:
Hi @vusdoo  The Isusb and image that you provided shows that your D435 camera is able to detected correctly.  It does not look right though that your /video list is empty when you use the command ** ls -al | grep /dev/video** as I would expect /video 0, 1 and 2 to be present if a single D435 camera is attached to the computer, representing depth (0), infrared (1) and RGB (2) respectively.

In regard to your questions:

1.  If you installed librealsense from Debian packages then you can uninstall all librealsense packages with the command below:

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

If you installed librealsense from source code with CMake then go to the **Build** directory of your librealsense folder and use the command below:

**sudo make uninstall && make clean**

1.  If you wish to install a specific librealsense version from packages instead of the latest version then the guidance in the link below may be helpful.

https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785

If you are installing from source code then you can obtain the source code folder for SDK 2.47.0 by visiting the link below, scrolling down to the **Assets** file list and left-clicking on the **source code (zip)** link.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.47.0

[image]

2.  Could you provide more information please about the meaning of your question "how it works(2.48)".  Thanks!

closed 2021-08-16 by MartyG-RealSense