realsenseai/librealsense #9557: hardware_reset() name error not defined

Lostosaurus, 2021-08-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.12.11.00) | 
| Operating System & Version |   {ubuntu18.04  | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {/python}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am calling the pyrealsense2 pipeline inside of a thread. The first time it calls it works without issue, but on the second time the pipeline start command returns , a  RuntimeError : No device connected. Looking at other questions it appears that adding a 

dev.hardware_reset() into the front end before initializing the pipeline should work, but my jetson can't seem to find hardware_reset().

what package is hardware_reset included with? do i need to import a specific package first?

MartyG-RealSense, 2021-08-03:
Hi @Lostosaurus  The **hardware_reset()** instruction is supported by pyrealsense2 and does not require an additional package if you already have the pyrealsense2 wrapper installed on your computer.  

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.device.html#pyrealsense2.device.hardware_reset

A simple example of Python code for hardware_reset() to compare to your own code is here:

https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

If a specific camera in a set of multiple cameras needs to be reset then the link below describes how to perform this in Python using a camera serial number.

https://github.com/IntelRealSense/librealsense/issues/5428

Lostosaurus, 2021-08-03:
Thank you for your response. After seting up the reset similarly to this example i am now getting a error in the line
for dev in devices:
RuntimeError: failed to set power state

MartyG-RealSense, 2021-08-03:
You could try adding a **sleep** period directly after the hardware_reset() instruction to give the camera time to restart (which may take a few seconds to complete until the camera is ready to access).

First, add a time import instruction at the top of the script:

**import pyrealsense2 as rs
import time**

Then add the sleep instruction directly after the hardware_reset() instruction.  For example:

**time.sleep(4)**

Lostosaurus, 2021-08-04:
Hello, i tried adding a delay, up to 20s, but still see the power state error. 

Additionally, after using hardware reset, i checked  in realsense-viewer, and was unable to load the stereo module. i'm not sure this is related, but the Error msg is
invalid value in rs2_open_multiple(sensor:etc. Profiles:etc.)

Failed to resolve this request:
     Format:z16 width 848. Height: 480
Info:
    Formats:
          Z16

Physically unplugging the camera and plugging it back in solves the issue with the viewer.

MartyG-RealSense, 2021-08-04:
There is a Python case with similarities to this one in the link below, where the program runs fine the first time but on the next start attempt a **RuntimeError : No device connected** error is produced.  A commenter on that case suggested the solution of stopping the pipeline when the program ends.

https://stackoverflow.com/questions/65686232/how-do-i-restart-the-pipeline-with-pyrealsense2

Since the problem is solved for you when unplugging and re-inserting the camera, this suggests to me that the suggested pipe close fix may have merit, as an unplug-replug would close the pipeline and reset the camera.

Lostosaurus, 2021-08-04:
Hello, I will double check my script. It should have a line to close the pipeline at the end. Are there any commands to check for existing pipelines?

Lostosaurus, 2021-08-05:
I have checked and the thread goes through a pipeline.stop() before it exits.

MartyG-RealSense, 2021-08-05:
In your script, do you define a custom depth resolution of 848x480 using a **cfg** instruction or do you use the camera's default configuration?  

I ask this because if the script was requesting a depth resolution of 848x480 at 15 or 30 FPS instead of letting the camera set the resolution itself then if the camera was mis-detected as being in USB 2 mode instead of USB 3 after the reset, a **Couldn't resolve requests** error could occur.  This is because in USB 2 mode, the 848x480 resolution does not support 15 or 30 FPS and so the program would be unable to provide that stream configuration.

Lostosaurus, 2021-08-10:
I am calling it at 640*480 for the depth and color streams on the 455. I have simplified my code into a sample and have posted it below in case it may help. It is using multithreading to run 2 threads and in the demo after a sleep command has been passed it returns a global variable exiting both threads and returning to the main loop. after a few seconds it tries to start the multithreading again and in theory should do this indefinitely while the script is running

import time
import threading
import cv2
import concurrent.futures
import numpy as np
import pyrealsense2.pyrealsense2 as rs

##########################

WINDOW_NAME = 'cv2Window'

class safetyCheck():

    def __init__(self):
        print('init')

    def camera_feed(self):
        ctx= rs.context()
        devices = ctx.query_devices()
        for dev in devices:
            print(f'this is dev {dev}')
            dev.hardware_reset()
            time.sleep(4)     

        global variable1
        global variable2
        global variable3

        pipeline = rs.pipeline()
        config = rs.config()
        # Get device product line for setting a supporting resolution
        pipeline_wrapper = rs.pipeline_wrapper(pipeline)
        pipeline_profile = config.resolve(pipeline_wrapper)
        device = pipeline_profile.get_device()
        device_product_line = str(device.get_info(rs.camera_info.product_line))

        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

        if device_product_line == 'L500':
            config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, 30)
        else:
            config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
            
        # Start streaming
        profile=pipeline.start(config)
        # get depth scale
        depth_sensor=profile.get_device().first_depth_sensor()
        depth_scale = depth_sensor.get_depth_scale()

        while variable3== 0:
            # Wait for a coherent pair of frames: depth and color
            frames = pipeline.wait_for_frames()

            depth_frame = frames.get_depth_frame()
            color_frame = frames.get_color_frame()

            if not depth_frame or not color_frame:
                continue
            if color_frame is None:
                break
            # Convert images to numpy arrays
            depth_image = np.asanyarray(depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())

            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
            depth_colormap_dim = depth_colormap.shape
            color_colormap_dim = color_image.shape

            # If depth and color resolutions are different, resize color image to match depth image for display
            if depth_colormap_dim != color_colormap_dim:
                resized_color_image = cv2.resize(color_image, dsize=(depth_colormap_dim[1], depth_colormap_dim[0]), interpolation=cv2.INTER_AREA)
                images = np.hstack((resized_color_image, depth_colormap))
            else:
                images = np.hstack((color_image, depth_colormap))

            cv2.imshow(WINDOW_NAME, images)
            variable1=1
            toc = time.time()
            key = cv2.waitKey(1)
            if key == 27:  # ESC key: quit program
                print('exiting')
                break

        print('\nDone.')

        cv2.destroyAllWindows()
        pipeline.stop()

        print(" after pipelinestop ")

    def pan_camera(self):
        global variable1
        global variable2
        global variable3

        try:
            while variable3== 0:

                if variable1==1:
                    time.sleep(4) #wait for 4 seconds then set value to 1
                    if variable2==1:
                        variable3=0
                        variable2 = 0 #reset value

                    elif variable2==0:
                        variable3=1

                else:
                    print('camera not on', variable1)
                    time.sleep(1)
       
        finally:
            print('exit ')

def serverStart():

    processVar1=0
    statusVar1=0
    listen=1

    while True :
        #global variables process1
        global variable1
        global variable2
        global variable3
        
        variable1=0
        variable2=0
        variable3=0

        process1=safetyCheck()

        while True:
            while listen == 1:
                global logger
                print('waiting for process declaration')
                processVar1=0
                statusVar1=0
                time.sleep(4)
                listen = 0
                processVar1 = 1 
                print('doing process declaration')

            while processVar1==1 and statusVar1 == 0:

                try:
                    th1 = threading.Thread(target = process1.camera_feed, args=())
                    th2 = threading.Thread(target = process1.pan_camera, args=())

                    th1.start()
                    th2.start()
                    th1.join()
                    th2.join()

                except KeyboardInterrupt:
                    quit = True

                if variable3 == 1: # jetson side nothing found, send to freedom,

                    processVar1=0                    
                    statusVar1=1
                    listen=1
                    print('process 1 finish')

                    variable1=0
                    variable2=0
                    variable3= 0
                    time.sleep(3)

if __name__ == '__main__':

        serverStart()

MartyG-RealSense, 2021-08-10:
Thanks very much for the script.  Multi-threading is honestly outside of my programming knowledge.  There is an example of a multi-threading Python script in the link below though where they incorporate a mechanism for dealing with a situation where a device is not found.

https://github.com/IntelRealSense/librealsense/issues/6039

[image]

MartyG-RealSense, 2021-08-16:
Hi @Lostosaurus  Do you require further assistance with this case, please?  Thanks!

Lostosaurus, 2021-08-16:
Hello, sorry for my late response. I was unable to work on this for a while. 

1. I am having issues understanding how this code snippet is used. I looked at the linked thread, but did not find the snippet being used. I did find a number of the definitions that the snippet uses here: #7641

currently i am getting a segmentation fault from the :
profile = depth_sensor.start(get_frame_handler(device))

Would you be able to share how the depth_sensor.start syntax works? 
In the linked page they are passing an inner function? I haven't been able to find anything on this command.

MartyG-RealSense, 2021-08-16:
A RealSense team member provides advice about resolving segmentation errors with **depth_sensor.start** in https://github.com/IntelRealSense/librealsense/issues/1952

Lostosaurus, 2021-08-16:
I did run into this example while searching, but regrettably am not familiar enough with c++ in order to convert it to python. Is there any resources for the python wrapper that you would be able to provide me?

MartyG-RealSense, 2021-08-16:
I located a pyrealsense2 script that makes use of depth_sensor.start.

First, it calls a function called **reset_profiles()** that resets profiles instead of resetting the camera with a hardware_reset

https://codechina.csdn.net/mirrors/intelrealsense/librealsense/-/blob/revert-8926-hdr/unit-tests/func/rec-play/test-got-playback-frames.py#L62

After calling reset_profiles to activate the block of code in that function, it performs depth_sensor and color_sensor open and start instructions, followed by a sleep instruction.

https://codechina.csdn.net/mirrors/intelrealsense/librealsense/-/blob/revert-8926-hdr/unit-tests/func/rec-play/test-got-playback-frames.py#L171

Lostosaurus, 2021-08-18:
After taking a look at the code and inserting the above section and running the code I was seeing an issue with rs.pipeline where after resetting the sensors with close/stop it resulted in the pipeline not seeing any device connected.

I have tried to change from the pipeline over to the syncer. Although I am unsure if this is necessary, or if there is a better way.
to abbreviate i swapped out the pipleine txt for the follows
camera_syncer = rs.syncer()
color_sensor.open(dp)
color_sensor.start(camera_syncer)
color_sensor.open(dp)
color_sensor.start(camera_syncer)

fset = camera_syncer.wait_for_frames()
depth_frame = fset.get_depth_frame()

However trying to use what i expected to be a depth_frame later on threw a RuntimeError: null pointer passed for argument "frame_ref". 

is it possible to get frames without using the pipeline? Is that approach the correct direction?

MartyG-RealSense, 2021-08-18:
I researched your question carefully.  If you are ulsing multithreading, you could see how using the pipeline with **callbacks** works for you, like the Python pipeline callback script at https://github.com/IntelRealSense/librealsense/issues/5417

Lostosaurus, 2021-08-21:
After taking apart the code and rewritting it I realized the issue was with the imshow command. Which for some reason hangs after being called into a second thread. This was holding onto the depth stream for some reason resulting in the realsense-viewer throwing errors. I tried to close the thread which resulted in the could not set power state error because the thread closed prematurely.  As such this was not a realsense issue and will now close this issue.

closed 2021-08-21 by Lostosaurus

MartyG-RealSense, 2021-08-21:
Great to hear that you found the cause of the problem, @Lostosaurus - thanks very much for the update!