realsenseai/librealsense #9560: Realsense not working currently in Unreal Engine 4

jaybhanushali99, 2021-08-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |   Win 10  |   
| Platform                                 | PC |
| SDK Version                            |  v2.47.0                        | 
| Language                            |  Unreal Engine 4.26.2                       | 

### Issue Description
I am able to compile the UE4 wrapper and load the Realsense project in Unreal. But on playing the level it does not update the cubes with frames from the camera. The camera is also detected by the Blueprint correctly.

[image]

MartyG-RealSense, 2021-08-03:
Hi @jaybhanushali99  There was a similar problem with an L515 camera in a case in 2020.  A RealSense user provided details (in the link below) of how to manually configure camera settings for USB2 or USB3.  A camera auto-configure function was also added to the changes to the RealSense SDK that added support for Unreal 4.26, though it may be worth checking the configuration manually anyway to see whether it can resolve your problem.

https://github.com/IntelRealSense/librealsense/issues/7180#issuecomment-752377865

jaybhanushali99, 2021-08-10:
Thanks @MartyG-RealSense .
I updated the resolution manually to 1280x720 and it works!

MartyG-RealSense, 2021-08-10:
That's excellent news, @jaybhanushali99  - thanks for the update!

kborgen37, 2021-08-11:
just wanted to chime in as well since i just ran into this problem as well had the same issue and changing the resolutions to the [linked comment](https://github.com/IntelRealSense/librealsense/issues/7180#issuecomment-752377865) worked

MartyG-RealSense, 2021-08-11:
Thanks very much for the confirmation @kborgen37  :)

closed 2021-08-16 by MartyG-RealSense