realsenseai/librealsense #9561: Thermal error of D415 cameras

agustics, 2021-08-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.12.100 | 
| Operating System & Version |   RPi OS Lite Buster  | 
| Kernel Version (Linux Only)    |  5.10.52-v7+                                    | 
| Platform                                 | Raspberry Pi 3B+  |
| SDK Version                            |  2.43.0                         | 
| Language                            | C++                          | 
| Segment			|  Others                          | 

### Issue Description

A dataset of a static scene has been captured with a D415 camera over a temperature range from 0 to 24 ºC with High Accuracy Visual Preset in order to check the consistency of the data captured. The scene is a room of dimension 5x8 meters with one target placed at 4m and the other at 2m.

Unfortunately, we have observed that there is a distance offset that can be as high as 500mm that depends on the temperature and the measured distance. This behavior can be observed in the images below, which show the depthmaps captured at 0ºC and 24ºC and the distance difference between them.
[image]
[image]
In addition, ASIC and projector temperatures for both cases were:

| Ambient Temperature | ASIC Temperature | Laser Temperature |
|------------------|---------------------------|--------------------------|
| 0 ºC               |  14 ºC                     |  14 ºC                     |
| 24 ºC             |  28 ºC                      |  29 ºC                     |

On the other hand, measurements taken at similar temperatures do not show such a difference:
[image]
[image]

**Do you have any additional information about this issue?**

#8434 closed issue states that SDK version 2.43.0 adds "a compensation mechanism intended to mitigate the effect of thermal propagation in optics" but only for D455 cameras. We have checked if this option (RS2_OPTION_THERMAL_COMPENSATION) is supported by D415 camera but unfortunately it is not.

**Do you plan to implement thermal compensation for D415 cameras or recommend a way to mitigate this temperature effect on D415 cameras?**

Thank you in advance.

MartyG-RealSense, 2021-08-03:
Hi @agustics  There is no further information to provide about this feature or future plans for it at the time of writing this.  You could though instead provide external environmental control for the camera by providing a sealed custom housing for it, such as the one for an underwater project in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-805760519

I would not expect to see problems occurring at the low ASIC and Projector temperatures that you reported though.  Is the lighting level controlled and consistent throughout the day in an indoor setting, or is the camera being used outdoors where it is under 24 hour day and night cycle natural lighting conditions?

agustics, 2021-08-03:
Hi MartyG, thanks for your rapid response.

Regarding the lighting conditions, they have been stable during the experiment.

MartyG-RealSense, 2021-08-03:
I note that you are using the **High Accuracy** preset.  That may not be ideal due to the reduction in depth image detail compared to other presets.  A good compromise may be **Medium Density**, which provides a balance between accuracy and depth detail ('fill rate').

MartyG-RealSense, 2021-08-10:
Hi @agustics  Do you require further assistance with this case, please?  Thanks!

agustics, 2021-08-13:
Hi @MartyG-RealSense, 

Thanks for your advice! We do prefer to use High Accuracy preset to get fully accurate depth pixels no matter the fill rate.

We still look for a solution regarding thermal compensation since our use case contemplates temperatures within 0ºC to 30ºC range.

MartyG-RealSense, 2021-08-13:
The easiest solution may be to replace the D415 with a D455 that supports thermal adjustment.  If that is not possible though then another method that you could try is to periodically run an **On-Chip Calibration** operation to update the depth calibration. You could update the calibration for the current streaming session but not write the calibration permanently to the camera hardware.  Intel have published a white-paper document about this calibration method.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

Because an On-Chip depth calibration runs on the camera firmware driver, you do not need a separate calibration software tool installed.  On-Chip calibration can be accessed and activated using C++ scripting.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-d-on-chip-calibration-c-api

agustics, 2021-08-19:
Hi @MartyG-RealSense, nothing else to add, we will try to handle this situation with calibration. 

One last question before closing this, is there a technical constraint to implement thermal compensation on D415 besides D435 or it is just a non-prioritized issue for D415?

MartyG-RealSense, 2021-08-19:
My understanding is that only the D455 has a thermal monitor that makes this feature possible.

closed 2021-08-19 by agustics