realsenseai/librealsense #9565: Public key not found during installation

magicmomo123, 2021-08-03:
Hello,

I'm trying to install the realsense driver using the tutorial at this link, https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. I am trying to register the public key, however, when searching for the key at keyserver.ubuntu.com, the website says its not found. Is there something I am missing? Is there a new version? Also, what is the best way to download the realsense driver.

Thanks

MartyG-RealSense, 2021-08-03:
Hi @magicmomo123  I hope that the advice in the link below will be helpful in resolving your package installation problem.

https://github.com/IntelRealSense/librealsense/issues/9506#issuecomment-888068753

magicmomo123, 2021-08-03:
Thank you very much for your lighting quick response. I will see if this works for me.

closed 2021-08-16 by MartyG-RealSense