realsenseai/librealsense #9566: D415 does not work with PCI-e USB expansion port

ritviksingh9, 2021-08-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version | Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 5.4.0-80-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.48.0                         | 
| Language                            |  realsense-viewer                          | 
| Segment			|  Others                        | 

### Issue Description
I am using Intel D415 cameras and the `realsense-viewer` is able to work perfectly fine when it is plugged into the USB 3.2 port that comes with the computer. However, I want to plug this into the PCI-e to USB 3 expansion port for a multi-camera setup. When I plug **one** camera into it, I encountered errors. I unplugged and re-plugged it many times and sometimes, I only get the RGB stream to show up on the viewer, other times only the depth stream shows up, and sometimes none of the streams show up. I was wondering if there was a way to stream both RGB and depth footage consistently with the PCI-e expansion port.

The following error occurs when I connect the camera into the PCI-e expansion port, successfully turn on the RGB stream, and then try to turn on the stereo stream:

`Backend in rs2_open_multiple(sensor:0x555b95d45810, profiles:0x555b970b3590, count:1): set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out`

MartyG-RealSense, 2021-08-04:
Hi @ritviksingh9  There was a past case in https://github.com/IntelRealSense/librealsense/issues/6899 regarding the camera not working correctly in a pci-e expansion port but working normally when plugged in directly.  In that particular case, it was thought to be due to the **USB controllers** that the pci-e expansion was using compared to the USB controllers that the computer's USB port used.

The RealSense user in that case, who was also using Ubuntu, developed a bash script as a workaround to rebind the USB drivers before enabling the camera stream.

https://github.com/IntelRealSense/librealsense/issues/6899#issuecomment-671984909

ritviksingh9, 2021-08-10:
Okay thanks, will check out!

MartyG-RealSense, 2021-08-17:
Hi @ritviksingh9  Do you require further assistance with this case, please?  Thanks!

ritviksingh9, 2021-08-17:
Unfortunately, the fix did not work. The website also suggested to reset the USB ports but when we tried that the PCI-e expansion card still did not work.

MartyG-RealSense, 2021-08-17:
So to confirm, rebinding the USB drivers does not solve the problem no matter which method is used to do so, please?

ritviksingh9, 2021-08-18:
Yep. When we unbind and then bind the USB drivers it does not solve the problem of the Realsense not being recognized.

MartyG-RealSense, 2021-08-18:
When the PCI-e card was installed into the PC, can you confirm that it was connected up to the power supply unit (PSU) inside the PC?  The power lead supplies the PCI-e board with 5V that enables the board to be to provide ports with sufficient power when USB devices with a high power draw (such as a RealSense camera) are attached.  If the board was simply plugged into the PCI-e slot without the power cable attached then it might only be able to handle low power draw devices such as mouse and keyboard.

The image below is taken from the Amazon product listing for the board that you kindly provided at the beginning of this case.

[image]

ritviksingh9, 2021-08-19:
Yes, we made sure to connect the SATA power cable from the PCI-e board to the PSU.

MartyG-RealSense, 2021-08-19:
If you are using a PCI-e board for the sake of keeping cables tidy, I wonder if you could achieve the same with a mobile USB 3 hub that has a very short USB cable connecting it to the computer  You can find examples of them on Amazon by searching for **mobile usb hub**