realsenseai/librealsense #9568: About D435 camera field of view adjustment

ju-mingyue, 2021-08-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |  Linux (Ubuntu18.04.5)  | 
| Kernel Version (Linux Only)    |  4.9.201                                     | 
| Platform                                 | NVIDIA JetsonNano B01 |
| SDK Version                            |2.48.0                        | 
| Language                            | ROS packages                           | 
| Segment			|  Robot                         | 

### Issue Description
Hello, I want to know how the D435 camera adjusts the camera's field of view. The current situation is like this. Our camera is placed under the robot and tilted up 45 degrees, but the top of the robot shell will block the camera. I think this parameter How to adjust.

MartyG-RealSense, 2021-08-04:
Hi @ju-mingyue  If you are unable to have the camera pointing directly forward or directly downward and need to tilt it, I would not recommend a tilt angle of more than 30 degrees maximum, as problems in the image may occur beyond that angle.

The problem here though is that the robot shell is blocking visibility, so reducing tilt may not help much in this case.  The shelf is likely obscuring the view of both the depth and RGB sensors.

Other than alter the robot design to move the camera further forward so that less of the shell is visible, you would likely have to set up a **depth clamp** to ignore depth values near to the camera lenses (such as the first 0.5 meters) so that the shell is excluded from the image.  As the camera cannot see through the shell though if it is solid and not transparent, any detail on the outer side of the part of the shell that is obscuring the view will not be seen on the image.

If you are using the RealSense ROS wrapper, it does not have a minimum distance clamp (you can only clamp the maximum observable distance with **clip_distance**).  You can set a minimum dstance with **RTABMAP** though by using its **RangeMin** function.

https://github.com/IntelRealSense/realsense-ros/issues/1859#issuecomment-836207835

ju-mingyue, 2021-08-04:
@MartyG-RealSense  
I'm very sorry, I don't understand what you mean by set up a depth clamp, so I hope you can tell me in detail and how to set it.
For rtabmap, I didn’t use it at all, I used the depth information to convert the laser signal。
By the way, what is the best angle of upward tilt for D435?

MartyG-RealSense, 2021-08-04:
A **depth clamp** means setting a minimum and maximum distance for rendering depth data.  Depth coordinates that have distance values outside of this minimum and maximum range will not appear on the depth image.

I would think that if you are converting the depth map to a laser signal then any depth data that a depth clamp removes from the original depth image would not appear on the laser-image conversion.

As mentioned, I would recommend not tilting the camera further than 30 degrees in the up or down direction.  You are not prevented from tilting further to 45 degrees, but it may have unpredictable results on the depth image.

If you cannot move the camera to a different position on the robot, is there any possibility of making the part of the robot shell that is blocking the camera transparent so the camera can see through it?

ju-mingyue, 2021-08-05:
[image]
@MartyG-RealSense 
Hello, my camera also has the problem of "Incomplete video frame detected". I checked the temperature of Asic, as shown in the picture above.
I don't know how to configure it now

MartyG-RealSense, 2021-08-05:
I will link you to a list of resources about the **Incomplete video frame detected** error so that you can see if any of the situations listed might apply to you.

https://github.com/IntelRealSense/realsense-ros/issues/1899#issuecomment-849837579

The ASIC temperature of 40 degrees C is a little high but probably not enough to cause errors.  I would consider 42 degrees C or above to be the range where problems might start to occur in the image.

I note in your Viewer log that you are also experiencing frame counter resets.  The link below provides advice on some possible causes of a counter reset.

https://github.com/IntelRealSense/librealsense/issues/7819#issuecomment-733006800

MartyG-RealSense, 2021-08-11:
Hi @ju-mingyue  Do you require further assistance with this case, please?  Thanks!

ju-mingyue, 2021-08-11:
I'm very sorry, this question does not need help for the time being, I will close it and open it again when needed

closed 2021-08-11 by ju-mingyue

MartyG-RealSense, 2021-08-11:
Okay, thanks very much @ju-mingyue for the update!  Please do feel free to re-open this case at a future date if you need to.