realsenseai/librealsense #9569: Align vs. Pointcloud

snopytas, 2021-08-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } ### Issue Description

Hi,

in my project I detect the center (u,v) of a ball in the color frame. From the depth frame, which is aligned to the color frame, I get the depth at (u,v). With `rs2_deproject_pixel_to_point` I can calculate the (x,y,z) coordinate.

My question is: Does the calculation of the point cloud internally the same as above? If I would use the point cloud to detect the same center of the ball, would I get the same (x,y,z) coordinate or has the point cloud has any improvements? 

Thank you

MartyG-RealSense, 2021-08-04:
Hi @snopytas  The advice from a RealSense team member in the link below may be relevant to answering your question.

https://github.com/IntelRealSense/librealsense/issues/4644#issuecomment-522754917

snopytas, 2021-08-04:
Thanks. The thing is that using the pointcloud would force me to change my code, whereas the current code with aligned frames and `rs2_deproject_pixel_to_point` is nearly ready.  So I need to know if making big changes is worth it.

Now I understand that it is better to align to depth instead of align to color. My question was if the accuracy of x,y,z points in aligned frames is different from the point cloud or if it is negligible. From the thread you mentioned I get that there is only a slightly difference. Could this be correct?

MartyG-RealSense, 2021-08-04:
It is my understanding too that any difference is slight.  For some RealSense users the small difference matters for their particular application, but if you are satisfied with your results from **rs2_deproject_pixel_to_point** then you can certainly continue with that code instead of starting again with a method such as generating a pointcloud with **points = pc.calculate(depth_frame)** and then using **points.get_vertices**

snopytas, 2021-08-05:
@MartyG-RealSense Thank you so much!

closed 2021-08-05 by snopytas