realsenseai/librealsense #9571: g++ -std=c++11 filename.cpp -lrealsense2

ShengningZ, 2021-08-04:
Hey guys I'm really new to these things and..
[image]
I don't under stand what it means..
when I run the code:g++ -std=c++11 filename.cpp -lrealsense2   in the terminal,I get this below
g++: error: filename.cpp: No such file or directory

I am confused about this sentence.Could anyone tell me how to do that step exactly?

ShengningZ, 2021-08-04:
By the way,I followed the instruction in this page
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
and I can open the realsense-viewer
[image]
But when I am using python3
ModuleNotFoundError: No module named 'pyrealsense2'
this error shows up...

MartyG-RealSense, 2021-08-04:
Hi @ShengningZ  When installing the librealsense SDK from packages, you only need to install the first two packages listed to have a working SDK:

**librealsense2-dkms
librealsense2-utils**

The other two packages -dev and -dbg are optional packages for developers.  They contain header files and symbolic links (dev) and debug symbols (dbg).

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#package-details

Once librealsense2-dkms and librealsense2-utils are installed then you can skip past the instructions for the optional packages and run **realsense-viewer**

In regard to pyrealsense2 (Python) compatibility, this would not be installed by following the instructions in the installation.md instructions for building librealsense from source code.  It is possible to build librealsense and Python support at the same time by adding further commands to the CMake build instruction.

Could you tell me which computer or computing device you are using so that I can advise you about how to build the pyrealsense2 wrapper for Python compatibility, please?  Thanks!

ShengningZ, 2021-08-04:
Hi, MartyG
I am using a Jetson nano developer kit.Thanks!

MartyG-RealSense, 2021-08-04:
If you have a Jetson board then Intel's special installation instruction page for Jetson should be used.  If you are installing from packages then the instructions on this page should be used instead of the distribution_linux.md page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

The package installation instructions should be begun at section **5.  Register the server's public key**

For building librealsense from source code on Nano with pyrealsense2 support included in the build, I would recommend the installation guide in the link below.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

MartyG-RealSense, 2021-08-04:
With an RSUSB installation, there should be a working internet connection on your Nano.  If you were able to run the simpler version of the CMake command then it sounds as though your internet connection is okay.

It looks as though on the more complex instruction, there is not a space between cmake ../ and -DFORCE_RSUSB_BACKEND.  It should be:

**cmake ../ DFORCE_RSUSB_BACKEND**

ShengningZ, 2021-08-04:
Yeah I put a space in that and it start building
Then it shows 

-- Configuring incomplete, errors occurred!
See also "/home/shengning/Downloads/librealsense-2.47.0/build/CMakeFiles/CMakeOutput.log".
See also "/home/shengning/Downloads/librealsense-2.47.0/build/CMakeFiles/CMakeError.log".

when I tried this mentioned in that issue..
[image]
It is still not working and get the same result
So I looked up into the error log and I saw this with errors

> Run Build Command:"/usr/bin/make" "cmTC_75200/fast"
> /usr/bin/make -f CMakeFiles/cmTC_75200.dir/build.make CMakeFiles/cmTC_75200.dir/build
> make[1]: Entering directory '/home/shengning/Downloads/librealsense-2.47.0/build/CMakeFiles/CMakeTmp'
> Building C object CMakeFiles/cmTC_75200.dir/CheckSymbolExists.c.o
> /usr/bin/cc   -fPIC  -fPIE   -o CMakeFiles/cmTC_75200.dir/CheckSymbolExists.c.o   -c /home/shengning/Downloads/librealsense-2.47.0/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
> Linking C executable cmTC_75200
> /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_75200.dir/link.txt --verbose=1
> /usr/bin/cc -fPIC     CMakeFiles/cmTC_75200.dir/CheckSymbolExists.c.o  -o cmTC_75200 
> CMakeFiles/cmTC_75200.dir/CheckSymbolExists.c.o: In function `main':
> CheckSymbolExists.c:(.text+0x14): undefined reference to `pthread_create'
> CheckSymbolExists.c:(.text+0x18): undefined reference to `pthread_create'
> collect2: error: ld returned 1 exit status
> CMakeFiles/cmTC_75200.dir/build.make:97: recipe for target 'cmTC_75200' failed
> make[1]: *** [cmTC_75200] Error 1
> make[1]: Leaving directory '/home/shengning/Downloads/librealsense-2.47.0/build/CMakeFiles/CMakeTmp'
> Makefile:126: recipe for target 'cmTC_75200/fast' failed
> make: *** [cmTC_75200/fast] Error 2

and this

> Run Build Command:"/usr/bin/make" "cmTC_154e5/fast"
> /usr/bin/make -f CMakeFiles/cmTC_154e5.dir/build.make CMakeFiles/cmTC_154e5.dir/build
> make[1]: Entering directory '/home/shengning/Downloads/librealsense-2.47.0/build/CMakeFiles/CMakeTmp'
> Building C object CMakeFiles/cmTC_154e5.dir/CheckFunctionExists.c.o
> /usr/bin/cc   -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create -fPIE   -o CMakeFiles/cmTC_154e5.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
> Linking C executable cmTC_154e5
> /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_154e5.dir/link.txt --verbose=1
> /usr/bin/cc -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_154e5.dir/CheckFunctionExists.c.o  -o cmTC_154e5 -lpthreads 
> /usr/bin/ld: cannot find -lpthreads
> collect2: error: ld returned 1 exit status
> CMakeFiles/cmTC_154e5.dir/build.make:97: recipe for target 'cmTC_154e5' failed
> make[1]: *** [cmTC_154e5] Error 1
> make[1]: Leaving directory '/home/shengning/Downloads/librealsense-2.47.0/build/CMakeFiles/CMakeTmp'
> Makefile:126: recipe for target 'cmTC_154e5/fast' failed
> make: *** [cmTC_154e5/fast] Error 2

MartyG-RealSense, 2021-08-04:
Looks like it may be a debugging related error.  Did you put a '-' in front of DCMAKE_BUILD_TYPE=release

**-DCMAKE_BUILD_TYPE=release**

MartyG-RealSense, 2021-08-04:
If you continue to have problems: if the simple CMake instruction built successfully but the more complex one didn't then it may be worth starting with the simple instruction and doing test builds by adding terms from the complex instruction one by one until you find which one is making the build break.  Good luck!

ShengningZ, 2021-08-05:
I tried some more things
First I followed the instruction here [https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049](url)
and that post says I need to update Cmake and I did as this method said[https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977](url)
and failed to do the next steps.
1.So I checked my CMAKE version and found that I did not upgrate successfully
2.Then I Uninstalled the CMAKE and installed Cmake again
3.and I put the pathonpath in that .bashrc file
4.Next, I ran `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`
Then I found it is now compiling...
5.Then I ran ` make -j4` 

But at this step,when the process goes to about 30%, the makefile interrupts

> ^CCMakeFiles/realsense2.dir/build.make:2486: recipe for target 'CMakeFiles/realsense2.dir/src/archive.cpp.o' failed
> make[2]: *** [CMakeFiles/realsense2.dir/src/archive.cpp.o] Interrupt
> CMakeFiles/realsense2.dir/build.make:2366: recipe for target 'CMakeFiles/realsense2.dir/src/tm2/tm-device.cpp.o' failed
> make[2]: *** [CMakeFiles/realsense2.dir/src/tm2/tm-device.cpp.o] Interrupt
> CMakeFiles/realsense2.dir/build.make:2462: recipe for target 'CMakeFiles/realsense2.dir/src/algo.cpp.o' failed
> make[2]: *** [CMakeFiles/realsense2.dir/src/algo.cpp.o] Interrupt
> CMakeFiles/realsense2.dir/build.make:2390: recipe for target 'CMakeFiles/realsense2.dir/src/tm2/tm-info.cpp.o' failed
> make[2]: *** [CMakeFiles/realsense2.dir/src/tm2/tm-info.cpp.o] Interrupt
> CMakeFiles/Makefile2:105: recipe for target 'CMakeFiles/realsense2.dir/all' failed
> make[1]: *** [CMakeFiles/realsense2.dir/all] Interrupt
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Interrupt
>

MartyG-RealSense, 2021-08-05:
On some 'low end' devices, using a -j number beside the **make** instruction to accelerate build time may cause a random crash during building.   Could you try just building with **make** instead of **make -j4** please.

ShengningZ, 2021-08-05:
Thanks! Now it is installed successfully and I can import pyrealsense2 now!
Thanks for your patient!

MartyG-RealSense, 2021-08-05:
That's great news, @ShengningZ - you are very welcome.   :)

closed 2021-08-11 by MartyG-RealSense