realsenseai/librealsense #9572: D435i IR图像容易过曝

meiy-px, 2021-08-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D35i} | 
| Firmware Version                   | (Newest) | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |                            | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
在相同的光照条件下，D435i的IR图像比RGB图像更容易过曝，过曝区域深度测量不准确。
IR图像容易过曝，是因为IR sensor的动态范围比RGB sensor的动态范围低，还是因为IR图像故意将对比度拉的比较高？
我们是否可以调整IR图像的对比度？
[image]

MartyG-RealSense, 2021-08-05:
Translation: "D435i IR image is easy to overexpose.  Under the same lighting conditions, the IR image of the D435i is more likely to be overexposed than the RGB image, and the depth measurement of the overexposed area is not accurate.  IR images are prone to overexposure.  Is it because the dynamic range of the IR sensor is lower than that of the RGB sensor, or because the IR image deliberately increases the contrast?  Can we adjust the contrast of the IR image?"

************

Hello @meiy-px  The exposure effects that you are experiencing with infrared resemble the ones in the case linked to below.  Reading this case should be helpful for you.

https://github.com/IntelRealSense/librealsense/issues/2875

At a later time after the advice in that case was published, Intel also added the **High Dynamic Range (HDR)** feature for controlling exposure and gain.

https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras

************

你好@meiy-px 您使用紅外線體驗到的曝光效果類似於下面鏈接的案例中的效果。 閱讀此案例應該對您有所幫助。

在該案例的建議發布後，英特爾還添加了**高動態範圍 (HDR)** 功能，用於控制曝光和增益。

meiy-px, 2021-08-06:
It seems that D400 uses 2  consecutive frames to combine a HDR image. The performance will reduce when the camera is moving.

MartyG-RealSense, 2021-08-06:
Yes, the possibility of artefacts during camera motion or fast motion in the observed scene is mentioned in the **Limitations** section near the end of the HDR paper.

https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras#section-4-3-limitations

Which version of the RealSense SDK are you using please?  RealSense SDK version **2.42.0** introduced maximum limits for exposure and gain when depth auto-exposure is enabled.  As well as requiring SDK 2.42.0 or newer, this feature also requires **5.12.11.0** or newer.

[image]

meiy-px, 2021-08-09:
Yes, the motion artifacts can be improved by limit the max exposure time. But the interval time between two frames would lead to HDR artifact.

MartyG-RealSense, 2021-08-09:
As long as HDR is not enabled then motion of the camera or observed motion should not lead to HDR artefacts in the depth and infrared images on the D435i (as the left and right infrared sensors have a fast Global Shutter on the D435i model).

You could test this in the Viewer by ensuring that the **HDR Enabled** option is disabled (indicated by a black box beside it) and then using the limit sliders to set a maximum.

MartyG-RealSense, 2021-08-15:
Hi @meiy-px  Do you require further assistance with this case, please?  Thanks!

meiy-px, 2021-08-16:
> Hi @meiy-px Do you require further assistance with this case, please? Thanks!

No, thanks for your help.

closed 2021-08-16 by meiy-px

MartyG-RealSense, 2021-08-16:
You are very wlecome, @meiy-px - thanks for the update!