realsenseai/librealsense #9574: librealsense binary compatibility

Da-Hartl, 2021-08-05:
| Camera Model                       | D400, L500 | 
| Firmware Version                   | any | 
| Operating System & Version |   Win 10 / Ubuntu   | 
| Platform                                 | x64 PC.  |
| SDK Version                            | 2.43.0   | 

### Issue Description

Hello Realsense Team, 
i wanted to ask about the compatibility between different versions of the realsense2.dll / librealsense2.so on windows vs on linux.
We develop a shared library that links against the realsense shared library. 

On windows i noticed the following: if i link against 2.43.0, for example and at runtime older versions of the realsense2.dll are found, compatiblity is denied by the explicit comparison of the version numbers in the header rs.h. However newer versions of the dll like 2.47.x work. This lead me to believe that binary compatibility for newer versions in general can be expected.

Now i found out that if i do the same thing on linux (link against 2.43.0, but provide some newer version at runtime) the operating system will deny compatibility. This is because the SONAME of the librealsense2.so includes the 2.43 part and this is then the requirement for my new library. And if any 2.47 or other version is available on the system it is not enough for the OS to load my binary. (The comparison of the version numbers in the header files is never done in that case)

So, do i understand correctly that compatibility is handled differently on Windows vs Linux?
Can you clarify this topic for me?

Best Regards
Christian Hartinger

MartyG-RealSense, 2021-08-05:
Hi @Da-Hartl  I have carefully researched your question.  Whilst there is not enough reference material available about librealsense2.so on Windows in regard to shared libraries, there is some about shared libraries on Linux.  So I do not have the information to directly answer your Windows vs Linux question about this subject.  However, hopefully by looking at the Linux information it will be possible to provide further clarification of your queries about this subject.

Thank you in advance for your patience as we go through this process.

******************

The subject of compatibility between versions with regard to soname was discussed in https://github.com/IntelRealSense/librealsense/issues/3270 in February 2019.

Following that case, a PR request to add minor versions to soname was merged into the SDK in version **2.19.2**.

https://github.com/IntelRealSense/librealsense/pull/3449

The subject of compatibility between versions was later discussed in https://github.com/IntelRealSense/librealsense/issues/7109#issue-679256982 in August 2020 regard to how a project may not work if it was linked to an older SDK version than the one that was currently installed.  This seems to reflect your experience with problems with linking to 2.43.0 and then experiencing an incompatibility with newer versions.

I hope that the above references will be a good starting point for further analysis in this discussion about the problems that you are facing.

Da-Hartl, 2021-08-06:
Hello @MartyG-RealSense and thanks for responding to my question.

I have read the mentioned issues, and i interpret it in the way that for Linux the minor version is explicitly part of the SONAME because the Realsense development process does not focus on providing binary compatibility between an application linked against an older (minor)version of the shared object, and using a newer one at runtime.

However then the same should be true for windows. Meaning although it seems to work on windows (the operating system does not compare versions for me and the explicit version check in the realsense2.dll does not fail) it is probably not a good idea to rely on anything. So to always be on the save side i would need to make sure that on any system where our software gets deployed to, the realsense2.dll is available there in the specific version we have linked against.

Is this correct?

MartyG-RealSense, 2021-08-06:
This particular question is outside of my personal knowledge, as I am not involved in SDK development activities.  Generally speaking though, I find that with the Windows version of the SDK it consistently works fine to simply copy a pre-built realsense2.dll from the **bin** folder of the SDK and paste the DLL into the folder of the program needs it instead of compiling the DLL myself in Visual Studio.

If I am using the pre-made DLL with an SDK wrapper then I always make sure that I use the DLL from the same SDK version that the wrapper was built from.

**C: > Program files (x86) > Intel RealSense SDK 2.0 > bin > x64**

[image]

Da-Hartl, 2021-08-12:
Hello  @MartyG-RealSense  and sorry for not responding.

We still have to decide internally how to proceed with the information you provided. 
But besides that the case can be closed.

Thanks for your help

MartyG-RealSense, 2021-08-12:
No problem at all, @Da-Hartl - thanks for the update.  On your suggestion, I will close the case.  Good luck!

closed 2021-08-12 by MartyG-RealSense