realsenseai/librealsense #9578: UVC metadata payloads not available

zschanghai, 2021-08-05:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D415|
|**Serial Number**|806312060427|
|**Firmware Version**|05.12.13.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|806413020909|
|**Firmware Update Id**|806413020909|

Please provide a description of the problem

MartyG-RealSense, 2021-08-05:
Hi @zschanghai  Could you provide information about the questions below, please.

-  Which computer or computing device are you using (PC, Raspberry Pi, Nvidia Jetson, etc)

-  Is the warning **UVC metadata payloads not available** appearing in the RealSense ROS wrapper or in the librealsense SDK (in a program such as the RealSense Viewer)?

-  Which method did you use to install librealsense.  Did you install it from packages or build it from source code?

zschanghai, 2021-08-05:
Hi, @MartyG-RealSense, thank for your help!
**1.Whichcomputer or computing device are you using (PC, Raspberry Pi, Nvidia Jetson, etc)**
I use the computer(PC), ubuntu18.04;
**2.Is the warning UVC metadata payloads not available appearing in the RealSense ROS wrapper or in the librealsense SDK (in a program such as the RealSense Viewer)?**
the warning UVC metadata payloads not available appearing in the librealsense SDK (in a program such as the RealSense Viewer)
**3.Which method did you use to install librealsense. Did you install it from packages or build it from source code?**
I git clone the packages from the "https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation"

MartyG-RealSense, 2021-08-05:
Thank you very much @zschanghai 

The installation page that you linked to - **installation_md** - is the source code method of building librealsense with CMake.  Unlike the package method on the **distribution_linux.md** page, support for metadata is not automatically included.  

To add metadata support when building from source code with CMake, you need to patch the kernel using the instruction below.

**./scripts/patch-realsense-ubuntu-lts.sh**

[image]

zschanghai, 2021-08-06:
HI, @MartyG-RealSense, thank for your help!
According to your suggestion, the problem was effectively solved！

MartyG-RealSense, 2021-08-06:
You are very welcome, @zschanghai - it's great to hear that your problem was solved.  Thanks for the update!

MartyG-RealSense, 2021-08-12:
Case closed due to solution achieved and no further comments received.

closed 2021-08-12 by MartyG-RealSense

AndreV84, 2021-09-25:
Is there a solution/patch for 20.04 ubuntu?

MartyG-RealSense, 2021-09-25:
@AndreV84   A response has been posted to your question at https://github.com/IntelRealSense/realsense-ros/issues/1604#issuecomment-927139643