realsenseai/librealsense #9580: rs-enumerate-devices: command not found

luan1412167, 2021-08-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455 } | 
| Firmware Version                   | () | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson   |
| SDK Version                            |  { }                          | 
| Language                            |  {pythony }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi team, 
I have install pyrealsense2 by following steps. 
```sudo rm -rf /tmp/check_install_dependencies_status.txt
sudo apt-get update 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get install -y wget curl 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get install -y libxinerama-dev libxcursor-dev 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get install librealsense2-dev 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get install librealsense2-dkms 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get install librealsense2-utils 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo apt-get install curl
sudo apt-get install libssl-dev libcurl4-openssl-dev

workding_dir=$(pwd)

sudo tee -a ~/.bashrc &>/dev/null <<'EOF'
export CUDA_HOME=/usr/local/cuda
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64:/usr/local/cuda/extras/CUPTI/lib64
export PATH=$PATH:$CUDA_HOME/bin
EOF

source ~/.bashrc

# update Cmake
wget http://www.cmake.org/files/v3.13/cmake-3.13.0.tar.gz 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
tar xpvf cmake-3.13.0.tar.gz cmake-3.13.0/ 
cd cmake-3.13.0/
./bootstrap --system-curl
make -j6 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
echo 'export PATH=$PWD/bin/:$PATH' >> ~/.bashrc
source ~/.bashrc
cd ${workding_dir}
rm -rf cmake-3.13.0.tar.gz

#Install librealsense
which python3 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
# python3_path=$(which python3) 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
# echo ${python3_path} 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
wget https://github.com/IntelRealSense/librealsense/archive/refs/tags/v2.48.0.zip 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
unzip v2.48.0.zip
cd librealsense-2.48.0
mkdir -p build && cd build
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
make -j4 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
sudo make install 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
cd ..
sudo tee -a ~/.bashrc &>/dev/null <<'EOF'
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
EOF

source ~/.bashrc
./scripts/setup_udev_rules.sh 2>&1 | tee -a /tmp/check_install_dependencies_status.txt

cd ${workding_dir}
rm -rf v2.48.0.zip

rs-enumerate-devices 2>&1 | tee -a /tmp/check_install_dependencies_status.txt
```

The installation is done without error. However, when I run `rs-enumerate-devices` I got error `rs-enumerate-devices: command not found` . Anyone can point out my mistake.
Thanks for support

luan1412167, 2021-08-05:
[check_install_dependencies_status_v2.txt](https://github.com/IntelRealSense/librealsense/files/6939573/check_install_dependencies_status_v2.txt)
This is installation log.

MartyG-RealSense, 2021-08-05:
Hi @luan1412167   Could you check the **usr/local/bin** folder of Ubuntu and see whether **rs-enumerate-devices** is located in that folder, please?  If it is, are you able to launch it whilst you are inside that folder using the **rs-enumerate-devices** command in the Ubuntu terminal?

luan1412167, 2021-08-05:
@MartyG-RealSense I worked fine. thanks for your support.

closed 2021-08-05 by luan1412167