realsenseai/librealsense #9581: Trajectory processing and post-processing saved to disk.

jimmek, 2021-08-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D445} | 
| Firmware Version                   | 5.12.10.00 | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |  NA                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.48.0.3381                        | 
| Language                            |  {C/C++                          | 
| Segment			|  imaging plus depth measures                        | 

### Issue Description
Now that the imu data (linear and gyro accelerations) are available via rs-convert, computation of trajectories is possible given some initial condition inputs. There are a number of programs in the SDK tools and sample programs for trajectory but they appear to be for specific conditions (such as T265 pose stream) and outputs primarily for screen display and not for saving the trajectory to a file. 

Are there any programs within the SDK tools and samples that I may have missed for generating and saving a trajectory in real-time for the D455 camera or post-processing a bag file for the D455 pose stream to generate and save the trajectory given initial condition inputs?  

Do you know of any good code or pseudo-code (and documentation) outside of the SDK tools and samples for computing trajectories? I have found one thesis with code similar to that found in rs-trajectory based on using the Quaternion method. 

Again thank you for your assistance.

MartyG-RealSense, 2021-08-06:
Hi @jimmek   Although it is more complex to implement than the **rs-trajectory** example for T265, a RealSense user has published an adaptation of Intel's official D435i SLAM guide for ROS.  The adaptation expands upon the concept of the original guide to demonstrate how to add path tracking and draw the trajectory from position data with **matplotlib**.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

[image]

jimmek, 2021-08-06:
Hello MartyG,

Thank you for comment.  I have looked at some of the references you provided for trajectory computation. As you said, they do look to be more complex to implement. I think the best approach for trajectory computation may be post-processing as the imu data is now available via rs-convert. Do you know of any other good code or pseudo-code (and documentation) for trajectory computation? As I mention in the previous comment, I did find one thesis with code similar to that found in rs-trajectory that was based on using the Quaternion method. Do you or your colleagues know of any other sources?

Thank you for your time and assistance.

MartyG-RealSense, 2021-08-06:
Was the quaternion-using paper that you saw the one in the link below that was published in May 2021 and uses a RealSense D455 to compute the camera trajectory?

https://www.mdpi.com/2071-1050/13/10/5585/htm

Another paper, meanwhile, uses **backward trajectory post-processing** with two D455 cameras.

https://www.mdpi.com/2072-4292/13/16/3099/htm

*******************

An existing platform that may also fit your needs is **SLAMcore** ( https://www.slamcore.com/ ), which is compatible with RealSense D435i and D455.  An article on the official RealSense blog about SLAMcore can be found in the link below.

https://www.intelrealsense.com/slamcore/

**Real-time position estimate**
[image]

closed 2021-08-19 by MartyG-RealSense