realsenseai/librealsense #9586: (jetson-inference) Training Image Classification Models

b2992872, 2021-08-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  NVIDIA Jetson   |
| SDK Version                            |   }                          | 
| Language                            |  python }                          | 
| Segment			|  Robot                       | 

I want to use jetson-inference to train an image classification model. I followed the steps in this video https://www.youtube.com/watch?v=sN6aT9TpltU&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=12, everything went well before I reached the 26th minute and 40th of the video, but it appeared when I started training Error. As shown in the figure, if the picture is not found, it will automatically go to the location of data/tools/val instead of data/tools/train. Although I can execute pictures in val, will it be a problem? Sorry I'm not sure if this question can be asked here。

This problem is similar to mine, but it can't solve my problem. Provide reference
https://github.com/pytorch/examples/issues/236
[image]

MartyG-RealSense, 2021-08-08:
Hi @b2992872  My research found a similar case to yours about this kind of RuntimeError when using **jetson-inference/python/training/classification/data**

https://forums.developer.nvidia.com/t/jetson-inference-collecting-your-own-detection-datasets/158681

In that case, the solution utilized by the jetson-inference was to move the images from the 'train' folder to their empty 'val' folder.  So if in your case the images are being moved from train to val automatically by jetson-inference then it is already performing the solution on your behalf.

b2992872, 2021-08-09:
This is unexpected. Is there a way for me to judge whether the training results have converged? , Because it is 100% successful, even if there is no goal, it is 100%

MartyG-RealSense, 2021-08-09:
There is a pytorch collection example on the jetson-inference GitHub called **pytorch-collect**.  This suggests using a tool called **camera-capture** to set up the train, val and test folders and define subfolders within those three main folders.

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md

Are you able to train data without errors and collect 10 to 20 percent of the image set if you follow this guide, please?

b2992872, 2021-08-09:
Yes, I followed this example exactly. I collected 100 training images and 20 validation images, but each level was 100% during training. According to the example video, it will only approach 100%.

MartyG-RealSense, 2021-08-09:
I researched the question and did not find any statements that having all 100% images is wrong.  Looking at your tabletop images above, they have a well-lit 'studio quality' to them, compared to the dimly lit conditions with changing natural background lighting that the dinosaur animated image and the image of the man apparently have.  I would be inclined to take the word of jetson-inference that they are 100% images.  You could of course randomly choose a different 20 images from the set of 100 and run the training again if you prefer to validate the results though.

b2992872, 2021-08-10:
Thank you for your help, I have tried to take pictures with an additional light source, the result is the same, I will go to jetson-inference for help, I hope you have a good day.

closed 2021-08-21 by MartyG-RealSense