realsenseai/librealsense #9588: Render PointCloud in Unity using SoftwareDevice

abhijaysingh, 2021-08-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |   2.47.0                          | 
| Language                            |  Unity                           | 
| Segment			|  VR                          | 

I have setup a software device that receives raw depth and color frames over the network and uses a modified event system of the RealSense PointCloudColorAndDepth sample provided as part of the RealSense package. I'm able to receive the frames and render the pointcloud but the color texture does not get applied on to the pointcloud. The UV map field of the PointCloudGeom shader does not update (is black). Can someone please point me in the right direction?

images attached for reference : 

[image]

MartyG-RealSense, 2021-08-09:
Hi @abhijaysingh  In the link below, a RealSense Unity user also had a problem with getting the point cloud to display correctly on an adaptation of the **PointCloudDepthAndColor** sample scene.  The found that changing the selected shader on **PointCloudMat** from 'Custom/PointCloudGeom' to the **Custom/Pointcloud** shader resolved the problem.

https://github.com/IntelRealSense/librealsense/issues/4155#issuecomment-501933766

[image]

abhijaysingh, 2021-08-09:
Hi @MartyG-RealSense Thanks for the swift response. I had tried changing the shader earlier as well but the issue persists. The UV Map field is black even with the shader set to Custom/PointCloud. Is it something related to loading the raw texture data from Points (https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/Frames/Points.cs)?

MartyG-RealSense, 2021-08-09:
In the RealSense Unity wrapper, camera data is converted into a mat so that it can be rendered.    I can see in the **PointCloudMat** panel of the Inspector that a color image is being received.  

I find that in Unity, if a texture will not display correctly then simply selecting Unity's default **Standard** shader from the shader list will often render it.  Have you tried Standard, please?

abhijaysingh, 2021-08-09:
Thank you, @MartyG-RealSense! It worked but the texture has a blue tint to it and the texture coordinates are not right. 

[image]

MartyG-RealSense, 2021-08-09:
Are you able to achieve depth-color alignment if you add an **Rs Align** component to your point cloud project, like in the recent RealSense Unity wrapper align project in the link below?

abhijaysingh, 2021-08-09:
It's working! 

Thanks a lot @MartyG-RealSense!!

closed 2021-08-09 by abhijaysingh

MartyG-RealSense, 2021-08-09:
That's great news @abhijaysingh - you are very welcome  :)