realsenseai/librealsense #9589: Last frame's afterimage appears in the current depth image

surefyyq, 2021-08-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | V1. 27 | 
| Operating System & Version |  Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |                         |                  | 
| Platform                                 |  NVIDIA Jetson TX2  |
| SDK Version                            |   2. 41                     | 
| Language                            |  python                          | 
| Segment			|  Robot                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I use a python program to control the camera to obtain the frames, and find that the depth image of the current frame will remain with the depth shadow of the previous frame after the position of targets changed. The pictures says it clearer:
[image]
[image]
Both depth and color exposure mode are automatic. What could be the possible reason, please?

MartyG-RealSense, 2021-08-09:
Hi @surefyyq  This effect of a different frame being shown as a slice on top of the main image has been demonstrated in a very small number of past RealSense cases, typically with the RGB stream rather than the depth stream.  In some of the cases though, abonormalities were reported in the depth stream too at the same time even if the focus of the case was the RGB stream.

In three of the most similar cases that I examined side by side to compare to your case, two of those three cases stated as being on Jetson (Nano and NX).

https://github.com/IntelRealSense/librealsense/issues/5986
https://github.com/IntelRealSense/librealsense/issues/7588

The third case was in Ubuntu and Python, though the computer / computing device that was being used was not specified.

https://github.com/IntelRealSense/librealsense/issues/8205

In the Python case where the depth image was the focus of the problem (but they were using both depth and color), I suggested disabling an RGB option called **Auto-Exposure Priority** whilst keeping Auto-Exposure enabled.  Could you try this please as a starting point in investigating your prob;em?  Thanks!

https://github.com/IntelRealSense/librealsense/issues/8205#issuecomment-764860412

surefyyq, 2021-08-10:
@MartyG-RealSense Thanks for your reply! The first few issues seem to be not helpful to my problem and I will try to disable the RGB Auto-Exposure Priority whilst keep Auto-Exposure enabled.

MartyG-RealSense, 2021-08-10:
Thanks very much for the update.  Please do let me know whether disabling Auto-Exposure Priority was helpful.

surefyyq, 2021-08-11:
@MartyG-RealSense Sorry for the late, but it doesn't solve the problem.

My code is like below:
```
pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth, RSLUX, RSLUY, rs.format.z16, 30)
config.enable_stream(rs.stream.color, RSLUX, RSLUY, rs.format.bgr8, 30)

profile = pipeline.start(config)
    
depth_sensor = profile.get_device().query_sensors()[0]
depth_sensor.set_option(rs.option.enable_auto_exposure, True)

depth_sensor.set_option(rs.option.laser_power, 160)

color_sensor = profile.get_device().query_sensors()[1]
color_sensor.set_option(rs.option.auto_exposure_priority, False)
color_sensor.set_option(rs.option.enable_auto_exposure, True)
```
And actually the RGB image is all ok. There are no lines or strips in the color image, either alignment error with the depth image.

MartyG-RealSense, 2021-08-11:
Do the red-orange bars at the top and bottom of the depth image represent real-world objects please, such as shelves on a bookshelf?

If they are meant to be there, and the main problem that you have is that a trail is left behind when movement takes place, then this could be that the depth frames are not updating fast enough to clear the previous frame from the screen after the image updates to the new position.

If it is a temporal issue, and the FPS of your depth stream is not lagging significantly beneath the target value of '30, then adding a post-processing **temporal filter** may help to exert some control over it.

https://github.com/IntelRealSense/librealsense/issues/1672#issuecomment-387438447

surefyyq, 2021-08-12:
Yes, they are crops on the conveyor belt.

And actually I have added the temporal post-process.

How should I do to confirm the FPS of the depth stream is lagging or not? When I check it using realsense-viewer in win10, all things work noramlly.

MartyG-RealSense, 2021-08-12:
The temporal filter can cause such a trail on the depth image if the image update rate is slowed down by its **alpha** value being low.  The pyrealsense2 documentation describes this temporal filter setting as **smooth_alpha**

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.temporal_filter.html

surefyyq, 2021-08-12:
The smooth_alpha of temporal filter is set to 0.5 and delta 50.

The target is stationary when realsense taking pictures, that is, the conveyor belt is not in continuous motion, but stops and goes. The realsense gets frames only when the conveyor belt stops. There is about 1 minute interval between getting frames.

I will check FPS by `print(rs.frame_metadata_value.actual_fps)`

surefyyq, 2021-08-16:
The FPS seems fine, too.

I replaced it with a new D435 with the same python code and now all works well. It seems the depth function of the camera is broken.

MartyG-RealSense, 2021-08-16:
Thanks very much @surefyyq for the update!

surefyyq, 2021-09-15:
I'm afraid I have to reopen this issue. 

In the early days after I replaced with a new D435, everything worked fine, but after less than a month, the depth frame afterimage problem appeared again.

It comes back to my memory that this is also the case with the older D435.

Help~~

MartyG-RealSense, 2021-09-15:
Were the new camera and the old camera ever placed in storage unused in a cool dry place such as a box or desk drawer for a period of time before the problem occurred on the next use after removal from storage, please?

surefyyq, 2021-09-16:
@MartyG-RealSense Nah，the old and new D435 are installed above the conveyor belt all the time and the equipment has been continuously tested. The equipment is located in a dry and ventilated room.

MartyG-RealSense, 2021-09-16:
So the camera works normally when used with the Viewer, as confirmed in https://github.com/IntelRealSense/librealsense/issues/9589#issuecomment-897328274 earlier in this discussion.  Also, the Python code was working fine for for a few weeks after the camera was replaced but then the problem returned again.  This suggested that the Python code is also okay.

You could perhaps stop the pipeline after a capture is taken and then start the pipeline again before taking the next capture.  Stopping the pipeline should clear out frames that were in the pipeline at the time of stopping.

surefyyq, 2021-09-20:
Thanks for your advice, after various tests, I remove the post-process of the temporal filter, and the trail in depth image disappears, for now...

Don't know why, but you can close the issue, temporarily.

MartyG-RealSense, 2021-09-20:
Thanks very much for the update, @surefyyq - as suggested, I will close the issue.  Please feel free to re-open it or create a new issue at a future date if the problem re-occurs.

closed 2021-09-20 by MartyG-RealSense