realsenseai/librealsense #9590: Transformation Parameters' details in ICP Registration

yiyangd, 2021-08-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 5.12.13.50 | 
| Operating System & Version |    Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    | 5.4.0-77-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.44.0                        | 
| Language                            |  open3d/python                          | 
| Segment			|  others                           | 

### Issue Description
Hi,

I am trying to simulate the example below with my RealSense D435i Depth Camera, I would like to know how could I understand the parameters for initial transformation's parameters and how to define a set of transformation on my own experiment?

[image]

My own experiments (a chair with wall)  is a little bit mess:
[image]

Thanks!

MartyG-RealSense, 2021-08-10:
Hi @yiyangd   I must emphasize that I am not an Open3D specialist.  I have researched your question carefully though and will do my best to help you with the information that you require.

ICP registration and its use with Python is introduced in a tutorial guide (linked to below) in the **Basics** section of the Open3D documentation.

http://www.open3d.org/docs/0.9.0/tutorial/Basic/icp_registration.html

An extract from this guide about initial alignment is shown below.

[image]

In regard to using Open3D and ICP with a RealSense camera, the discussion below should be a good entry point into the subject.

https://github.com/isl-org/Open3D/issues/362

As mentioned at the beginning of this case, my knowledge of Open3D is limited but I will be happy to work through any further questions that you have that arise from the information resources highlighted above.

yiyangd, 2021-08-11:
Hi MartyG,

Thanks, so now my biggest problem is I didn't know the parameters of init_trans arrays, I guess I need to set one since I try to capture my two point clouds in different positions?

MartyG-RealSense, 2021-08-11:
You could try using the **trans_init** values from the earlier example.

[image]

The Open3D tutorial in the link below suggests though a method for calculating trans_init values with **user-provided correspondences**.

http://www.open3d.org/docs/latest/tutorial/visualization/interactive_visualization.html#registration-using-user-correspondences

[image]

MartyG-RealSense, 2021-08-17:
Hi @yiyangd  Do you require further assistance with this case, please?  Thanks!

yiyangd, 2021-08-18:
Hi @MartyG-RealSense，I just got some important information from [here](https://github.com/isl-org/Open3D/issues/3891#issuecomment-900937843),  and thank you for your information. 

> Hi @yiyangd Do you require further assistance with this case, please? Thanks!

MartyG-RealSense, 2021-08-19:
Thanks very much @yiyangd for the update.  If you are satisifed with the outcome of this case and do not require further assistance, please feel free to use the **Close issue** button beneath the comment writing box to close the case.  Thanks!

yiyangd, 2021-08-19:
# #

closed 2021-08-19 by yiyangd