realsenseai/librealsense #9591: The key to calibration tool cannot be used

luccosta, 2021-08-09:
> Hi
> 
> The 2.11.0.0 for Linux is already on AWS, same location as before. The documentation will be uploaded soon.
> 
> Here are the simplified steps:
> 
> echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
> sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE
> sudo apt-get update
> sudo apt-get install librscalibrationtool
> sudo dpkg -L librscalibrationtool
> 
> Hope it's helpful.
> 
> Thanks
> 
> **Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

hello,
This key "C8B3A55A6F3EFCDE"  cannot be used. We cannot get the calibration tool from the aws.Is there any other way we can get the  Dynamic Calibration Tool v2-11-0-0 available for Linux?
Thanks!
(Executing: /tmp/apt-key-gpghome.2nRpwLBFRl/gpg.1.sh --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE
gpg: Failed to receive from the public key server：No name)

_Originally posted by @JiangweiNEU in https://github.com/IntelRealSense/librealsense/issues/6772#issuecomment-878246427_

The key "C8B3A55A6F3EFCDE" didn't worked for me neither.

MartyG-RealSense, 2021-08-09:
Hi @luccosta  The **gnupg.net** keyserver was phased out.  I hope that the information in the link below about an alternative address (keyserver.ubuntu.com) will be helpful.

https://github.com/IntelRealSense/librealsense/issues/9506#issuecomment-888068753

MartyG-RealSense, 2021-08-22:
Case closed due to no further comments received.

closed 2021-08-22 by MartyG-RealSense

bhomaidan1990, 2022-01-17:
I wasn't able to download the tool on Ubuntu 20.04:
```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/focal/InRelease  403  Forbidden [IP: 52.218.85.74 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

`E: Unable to locate package librscalibrationtool`

MartyG-RealSense, 2022-01-17:
Hi @bhomaidan1990  the Dynamic Calibration tool supports Ubuntu 16.04 and 18.04 but not 20.04.

[image]

You can instead calibrate the camera using the **On-Chip** and **Tare** calibration functions.  This software is built into the camera firmware driver instead of requiring installation of a separate calibration software tool.  It is accessible from the **More** menu option near the top of the RealSense Viewer's options side-panel, or can alternatively be accessed through program scripting.

[image]

Intel have published a white-paper guide about the On-Chip and Tare calibration functions at the link below.  On-Chip can improve depth image quality, whilst Tare improves depth measurement accuracy.  The tool can also calibrate both intrinsics and extrinsics, though they have to be performed as separate calibration operations rather than both at the same time.  The normal version of the Dynamic Calibration tool only calibrates extrinsics.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

shphilippe, 2022-03-08:
The repository is not accesible anymore: 
```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/focal/InRelease  403  Forbidden [IP: 52.218.117.185 80]
```
I'm on ubuntu 18.04. Where can we download the latest version of the dynamic calibration tool ?

MartyG-RealSense, 2022-03-08:
Hi @shphilippe  Given that the **gnupg.net** keyserver was phased out last year, are you able to access the package using the ubuntu.com keyserver address?

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key 6F3EFCDE

shphilippe, 2022-03-08:
the gpg key is not the problem, it's already in my system:
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key 6F3EFCDE
Executing: /tmp/apt-key-gpghome.3qDpoHDw9x/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key 6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```
The problem is the S3 bucket is returning 403 Forbidden for all the requests.

feliyur, 2022-03-08:
I'm currently also unable to get the key from keyserver.ubuntu.com - getting "Not Found"
```
> curl -sSL 'http://keyserver.ubuntu.com/pks/lookup?op=get&search=F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE'
Not Found
```

MartyG-RealSense, 2022-03-08:
Hi @feliyur  The key **F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE** that you are using indicates that your particular problem is with installing the librealsense SDK from packages.  Is that correct, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

feliyur, 2022-03-08:
Hello, 
No, I'm trying to install the calibration tool librscalibrationtool following the "Intel® RealSenseTM Product Family D400 Series Calibration Tools" user guide pdf and #6772 (to replace keys.gnupg.net with keyserver.ubuntu.com). 

The manual lists 
> Register the server's public key:
> sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE

The same key is listed in #6772.

I'm trying to install from the provided apt repo as I didn't find a downloadable `librscalibrationtool_2.11.0.0_amd64.deb` (as per the pdf manual),  and the tool does not seem to be part of the SDK 2.0, which I was able to build locally.

feliyur, 2022-03-08:
I will try, however I think that the current calibration likely already is factory calibration - which needs to be adjusted.
Thanks!

MartyG-RealSense, 2022-03-09:
Intel responded with feedback.  They looked at the problem and identified that the Dynamic Calibration (DC) repo was okay but the keyserver had changed.  Until an update is released, they provided the steps below to use in the meantime.

***********

For Ubuntu 16.04 and Ubuntu 18.04, install directly from the Intel repo.
    
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key 6F3EFCDE
    
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
    
sudo apt-get update
    
sudo apt-get install librscalibrationtool

The same DC package from Ubuntu 16.04 / Ubuntu 18.04 already supports Ubuntu 20.04.  So for Ubuntu 20.04, just configure the Intel repo server for Ubuntu 16.04 or Ubuntu 18.04, then install it.  For example:
   
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key 6F3EFCDE
   
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u
   
sudo apt-get update
   
sudo apt-get install librscalibrationtool

As a workaround, you can always download the package on Ubuntu 16.04 / Ubuntu 18.04 and then move to Ubuntu 20.04 to install it directly.
   
sudo apt-get download librscalibrationtool
   
sudo dpkg -i librscalibrationtool_2.11.1.0_amd64.deb

feliyur, 2022-03-09:
@MartyG-RealSense 
Thank you for the reply, but the key 6F3EFCDE does not currently seem to exist on key server `keyserver.ubuntu.com` - which is the reason I was writing here in the first place - I'm getting "Not Found" from the server.

shphilippe, 2022-03-09:
indeed, it work with the librealsense.intel.com repository. I guess I had a very old Intel repository in my config files which is not working anymore

MartyG-RealSense, 2022-03-09:
@shphilippe  Thanks very much for the update.

@feliyur  The Dynamic Calibration tool is not open source.

feliyur, 2022-03-10:
Issue is solved for me. The reason I've been getting "Not Found" from `keyserver.ubuntu.com` is that I missed that I needed to prepend '0x' to any search on it. Searching for the key '0x6F3EFCDE` finds the key and allows it to be added manually. The key can be found here
https://keyserver.ubuntu.com/pks/lookup?search=0x6F3EFCDE&fingerprint=on&op=index
I needed to add the key manually in the first place because I was behind a proxy at the workplace, which blocked access to the keyserver.
Thanks @MartyG-RealSense .

MartyG-RealSense, 2022-03-10:
That's great to hear, @feliyur - thanks so much for the update and for sharing your method with the RealSense community  :)