realsenseai/librealsense #9594: Failed to make with architecture error  on M1 Mac

huckle-mems, 2021-08-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Firmware Version                   |  | 
| Operating System & Version | MacOS  BigSur 11.4 |                                      | 
| Platform                                 | M1 Macbook air  |
| SDK Version                            |  2.48.0                        | 
| Language                            |  C++    | 
| Segment			|  others                           | 

### Issue Description
  I followed the official installation guide at 
[https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md](url)
and added "set(CMAKE_MACOSX_RPATH 1)" to librealsense/CMakelists.txt in accordance with [#9330 ](url).
Additionaly,  I added "-DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl" and "-DOPENSSL_LIBRARIES=/opt/homebrew/opt/openssl/lib" as Cmake argument in accordance with [#8090](url). 

I tried under command and successed to create Makefile.
`mkdir build && cd build`
`xcode-select --reset`
`cmake .. DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl/ -DOPENSSL_LIBRARIES=/opt/homebrew/opt/openssl/lib/`

However, I get the following error when I try "make -j2" command:

Error Message:
> [ 51%] Linking CXX executable rs-software-device
ld: warning: ignoring file /opt/homebrew/lib/libglfw.3.3.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture x86_64:
  "_glfwCreateWindow", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwDestroyWindow", referenced from:
      window::~window() in rs-software-device.cpp.o
  "_glfwGetFramebufferSize", referenced from:
      window::operator bool() in rs-software-device.cpp.o
  "_glfwGetWindowSize", referenced from:
      window::operator bool() in rs-software-device.cpp.o
  "_glfwGetWindowUserPointer", referenced from:
      window::window(int, int, char const*)::'lambda'(GLFWwindow*, int, int, int)::operator()(GLFWwindow*, int, int, int) const in rs-software-device.cpp.o
      window::window(int, int, char const*)::'lambda'(GLFWwindow*, double, double)::operator()(GLFWwindow*, double, double) const in rs-software-device.cpp.o
      window::window(int, int, char const*)::'lambda0'(GLFWwindow*, double, double)::operator()(GLFWwindow*, double, double) const in rs-software-device.cpp.o
      window::window(int, int, char const*)::'lambda'(GLFWwindow*, int, int, int, int)::operator()(GLFWwindow*, int, int, int, int) const in rs-software-device.cpp.o
  "_glfwInit", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwMakeContextCurrent", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwPollEvents", referenced from:
      window::operator bool() in rs-software-device.cpp.o
  "_glfwSetCursorPosCallback", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwSetKeyCallback", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwSetMouseButtonCallback", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwSetScrollCallback", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwSetWindowUserPointer", referenced from:
      window::window(int, int, char const*) in rs-software-device.cpp.o
  "_glfwSwapBuffers", referenced from:
      window::operator bool() in rs-software-device.cpp.o
  "_glfwTerminate", referenced from:
      window::~window() in rs-software-device.cpp.o
  "_glfwWindowShouldClose", referenced from:
      window::operator bool() in rs-software-device.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [examples/software-device/rs-software-device] Error 1
make[1]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2

> [ 55%] Linking CXX shared library librealsense2-gl.dylib
ld: warning: ignoring file /opt/homebrew/lib/libglfw.3.3.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture x86_64:
  "_glfwCreateWindow", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
  "_glfwDestroyWindow", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
  "_glfwGetCurrentContext", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
  "_glfwGetProcAddress", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
  "_glfwInit", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
  "_glfwMakeContextCurrent", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
  "_glfwSwapInterval", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
  "_glfwWindowHint", referenced from:
      l___const.rs2_gl_init_rendering.binding in rs-gl.cpp.o
      l___const.rs2_gl_init_processing.binding in rs-gl.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/gl/librealsense2-gl.2.48.0.dylib] Error 1
make[1]: *** [src/gl/CMakeFiles/realsense2-gl.dir/all] Error 2
make: *** [all] Error 2

M1 apple arhchitecture is arm64, but its setting look like x86_64. so I think this problem is occurred.
Has anyone had similar problem?
If someone has experience that fix this error, could you give me your advice on  how to deal with this problem? 

Any help would be greatly appreciated!

MartyG-RealSense, 2021-08-10:
Hi @huckle-mems. Another RealSense M1 Mac owner resolved this architecture error by following the advice in https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-837115091 to clean the CMake cache and then generate the project again.

As the reply in the comment below that one indicates, it may take more than one try of this particular procedure to achieve a successful build on M1.

huckle-mems, 2021-08-10:
@MartyG-RealSense Thanks for the response. I tried CMake cache clear( rm -r build and do same  process), however it doesn't work.
Besides libusb pig-config and  Vulkan-sdk , should I install other package?

[image]

MartyG-RealSense, 2021-08-10:
I would recommend posting the above question on https://github.com/IntelRealSense/librealsense/issues/8090 as that is where most Mac M1 installation issues are collectively solved by M1 owning RealSense users.

https://github.com/IntelRealSense/librealsense/issues/8090

closed 2021-08-24 by MartyG-RealSense