realsenseai/librealsense #9600: RGB摄像头的实际曝光时间，以及AE最大曝光时间设置

meiy-px, 2021-08-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (05.12.14.50) | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { Newest }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
我们想使用RGB摄像头的信息进行识别，使用中发现当使用较低的帧率时，运动物体有明显的拖影，但是我们无法获取当时的实际曝光时间，也没办法设置AE的最大曝光时间。请问是否可以获取到RGB摄像头的实际曝光时间和设置AE的最大曝光时间？

MartyG-RealSense, 2021-08-10:
Translation:  "The actual exposure time of the RGB camera, and the AE maximum exposure time setting.  We want to use the information of the RGB camera for identification.  During use, we found that when using a lower frame rate, the moving objects have obvious smear, but we can't get the actual exposure time at that time, and we can't set the maximum exposure time of AE. 
 Can I get the actual exposure time of the RGB camera and set the maximum exposure time of AE?"

******************

Hello @meiy-px  The D435i camera model has a fast shutter on its depth sensor but a slower shutter on its RGB shutter.  When capturing fast motion with the RGB sensor, there may therefore be blurring / smearing of the RGB image, or misalignment when performing alignment between color and depth.

It is possible to retrieve a real exposure value for depth from the camera metadata.  A RealSense team member advises in https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920 though that an actual exposure metadata value is not able to be retrieved for RGB.

There are a couple of different approaches that you could try for reducing RGB blurring, depending on whether auto-exposure is enabled or disabled.

If auto-exposure is enabled and depth has an FPS speed of 30, then setting RGB to 60 FPS will reduce blurring.  This phenomenon is discussed in https://github.com/IntelRealSense/librealsense/issues/3554

If auto-exposure is disabled and you use manual exposure values then you can instead set RGB exposure to 70 and the RGB FPS to 6 (that is correct - 6 FPS, not 60) to reduce blurring.  

******************

你好@meiy-px D435i 相機型號在其深度傳感器上有一個快速快門，但在其 RGB 快門上有一個較慢的快門。因此，當使用 RGB 傳感器捕捉快速運動時，RGB 圖像可能會模糊/拖尾，或者在執行顏色和深度之間的對齊時會出現錯位。

可以從相機元數據中檢索深度的真實曝光值。 RealSense 團隊成員在 https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920 中提出建議，儘管無法為 RGB 檢索實際曝光元數據值。

您可以嘗試幾種不同的方法來減少 RGB 模糊，具體取決於是啟用還是禁用自動曝光。

如果啟用了自動曝光並且深度的 FPS 速度為 30，則將 RGB 設置為 60 FPS 將減少模糊。這種現像在 https://github.com/IntelRealSense/librealsense/issues/3554 中有討論

如果自動曝光被禁用並且您使用手動曝光值，那麼您可以改為將 RGB 曝光設置為 70，將 RGB FPS 設置為 6（這是正確的 - 6 FPS，而不是 60）以減少模糊。

meiy-px, 2021-08-16:
@MartyG-RealSense Increasing frame rate can reduce the maximum exposure time, and result in reducing motion blure. But we don't need 30fps or more. If we use 30fps, we need to sample from the received 30 frames. So we want to limit the maximum exposure time when using 15fps.

MartyG-RealSense, 2021-08-16:
If you require a custom frame rate that is lower than the FPS speed that has been set, one method for achieving this is to only use every 'nth' frame (e.g every 2nd frame out of 30 when using 30 FPS to simulate 15 FPS).   The link below that demonstrates this is a Python script, but the structure of the script may provide some insights for replicating it in C++.

https://github.com/IntelRealSense/librealsense/issues/9480#issuecomment-886218633

[image]

meiy-px, 2021-08-17:
Will you plan to add the max exposure time tuning function of RGB camera in the future?

MartyG-RealSense, 2021-08-17:
At the time of writing this, I am not aware of future plans to add the depth auto-exposure limit feature for RGB too.

meiy-px, 2021-08-17:
OK，got it. Thanks very much.

closed 2021-08-17 by meiy-px