realsenseai/librealsense #9603: Intel Realsense 265 make mouse very slow on Jetson TX2/Xavier NX

deeprun, 2021-08-11:
Hi, I am using Intel Realsense T265 camera on Jetson TX2 and Xavier NX via USB. There was also a wireless USB dongle for mouse and keyboard. TX2 first is connected to an USB hub, and T265 and mouse were plugged into the hub. When I ran T265 I found the mouse and keyboard response is very slow, and sometimes I could totally lost the muse. If I used wired mouse or keyboard, they worked fine.

How can I fix the mouse slow response issue?

deeprun, 2021-08-13:
Does anyone has the same problem?

codesian, 2021-12-16:
In my experience USB3 connections have issues with EMI interferences cause port speed. Use a hub to add space between camera usb conection and mouse adapter.

closed 2026-10-23 by MartyG-RealSense