realsenseai/librealsense #9604: Can't open Ue4 wrapper project. Failed to load because module 'RealSense couldn't be found.

MatthewLKomar, 2021-08-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |   {Win 10)  | 
| Platform                                 | PC  |
| SDK Version                            |  2   | 
| Language                            |  unreal   4.26.2 and 4.19      | 

I'm following the instructions on this link: 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unrealengine4
However, when building it, I get this error message. 

`Library 'realsense2.lib' was not resolvable to a file when used in Module 'RealSense', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
`
Then after launching it, I get this error 
[image]

MartyG-RealSense, 2021-08-11:
Hi @MatthewLKomar  I note that your kindly provided information at the top of this case states **unreal 4.26.2 and 4.19**.  If you are using librealsense SDK **2.42.0** or newer then although Unreal engine 4.26 is supported, Unreal versions older than 4.26 may experience breakages. 

[image]

As SDK 2.42.0 is the version that introduces 4.26 support, using an older SDK version will lose that support.  So you may unfortunately need to leave 4.19 behind if that is the Unreal version that you are experiencing the error in and you need to use 4.26.

If you need support for 2.19, a compromise may be to use 4.25 instead of 4.26 and librealsense version **2.41.0**.  Doing so would prevent your project from being updated to a newer Unreal and librealsense version in future though.

MatthewLKomar, 2021-08-11:
I have the latest librealsense SDK and I got the error for Ue4 4.26. 
I just tried it for 4.19.2 just in case.

MartyG-RealSense, 2021-08-11:
If the latest librealsense version that you are using is 2.49.0, the most recent report from the RealSense community in which I have seen a successful build with Unreal 4.26.2 is librealsense **2.47.0**.  

https://github.com/IntelRealSense/librealsense/issues/9560

So it may be worth trying that slightly older SDK version with Unreal 4.26.2.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.47.0

MatthewLKomar, 2021-08-13:
Brain confused Intel.Realsense.dll with realsense.dll
Using realsense.dll makes project work.

closed 2021-08-13 by MatthewLKomar

MartyG-RealSense, 2021-08-13:
Excellent to hear that you found the solution, @MatthewLKomar - thanks for the update!