realsenseai/librealsense #9605: Cannot obtain RGB video stream in YOLOv4 model using RealSense (D435)  Camera

VAIBHAVPATEL97, 2021-08-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.9.201-tegra  | 
| Platform                                 | NVIDIA Jetson AGX Xavier |
| SDK Version                            |   v2.48.0                          | 
| Language                            |  Python                       | 
| Segment			|  Others                     | 
| Jetpack |   Jetpack v4.5.1 |

Hello,
I trying to use Intel RealSense D435 camera as an input to the YOLOv4 darknet model but it shows a grayscale version of the captured video. How can I convert the input video stream to an RGB version as a normal camera ?. (Attached screenshot of the input video captured using RealSense).

[image]

MartyG-RealSense, 2021-08-11:
Hi @VAIBHAVPATEL97  Because of the dot pattern projection on your kindly provided image, it looks as though you have captured the monochrome **infrared** stream of the camera rather than the camera's RGB stream.

VAIBHAVPATEL97, 2021-08-12:
Hi @MartyG-RealSense, I was able to fix the issue by changing some code in the python script. Another question is it possible to change the resolution of the Intel RealSense during RGB video capturing ?

MartyG-RealSense, 2021-08-12:
Hi @VAIBHAVPATEL97   I'm pleased to hear that you resolved the problem with the stream type.

To change the resolution whilst the program is running, you could stop the pipeline, set a new resolution with a **cfg.enable_stream** stream configuration instruction and start the pipeline again to apply the new resolution setting.

closed 2021-08-24 by MartyG-RealSense