realsenseai/librealsense #9607: librealsense2-udev-rules 2.49.0-0~realsense0.5199 fails on Ubuntu 20.04 docker

wngreene, 2021-08-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | NA | 
| Operating System & Version |   {Ubuntu 20.04 Docker)  | 
| Kernel Version (Linux Only)    |                  5.8.0-63-generic                         | 
| Platform                                 | PC  |
| SDK Version                            |         2.49.0-0~realsense0.5199                    | 
| Language                            |  NA                          | 
| Segment			|  NA                         | 

### Issue Description
We have a CI server that installs `librealsense2-dev` inside a Docker container. With the update to version 2.49.0-0~realsense0.5199, the installation of `librealsense2-udev-rules` fails with the following error:

```bash
root@f4fb0717a804:/workdir# sudo DEBIAN_FRONTEND=noninteractive apt install librealsense2-udev-rules
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-udev-rules
0 upgraded, 1 newly installed, 0 to remove and 12 not upgraded.
Need to get 7608 B of archives.
After this operation, 30.7 kB of additional disk space will be used.
Get:1 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 librealsense2-udev-rules amd64 2.49.0-0~realsense0.5199 [7608 B]
Fetched 7608 B in 1s (12.6 kB/s)                   
Selecting previously unselected package librealsense2-udev-rules:amd64.
(Reading database ... 134631 files and directories currently installed.)
Preparing to unpack .../librealsense2-udev-rules_2.49.0-0~realsense0.5199_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5199) ...
Setting up librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5199) ...
+ echo \nPostinst script activated...

Postinst script activated...
+ mkdir -p /home/root/Documents/librealsense2
+ rsync -avh --ignore-errors usr/share/librealsense2/presets /home/root/Documents/librealsense2
sending incremental file list
rsync: change_dir "/usr/share/librealsense2" failed: No such file or directory (2)

sent 18 bytes  received 12 bytes  60.00 bytes/sec
total size is 0  speedup is 0.00
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
dpkg: error processing package librealsense2-udev-rules:amd64 (--configure):
 installed librealsense2-udev-rules:amd64 package post-installation script subprocess returned error exit status 23
Errors were encountered while processing:
 librealsense2-udev-rules:amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Seems like the issue is that `/usr/share/librealsense2` isn't created for some reason.

wngreene, 2021-08-11:
The previous version (2.48.0-0~realsense0.4976) seems to install fine:

```bash
root@f4fb0717a804:/workdir# sudo apt install librealsense2-udev-rules=2.48.0-0~realsense0.4976
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dctrl-tools dkms linux-headers-5.4.0-80 linux-headers-5.4.0-80-generic linux-headers-generic
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  librealsense2-udev-rules
0 upgraded, 1 newly installed, 0 to remove and 12 not upgraded.
Need to get 7432 B of archives.
After this operation, 29.7 kB of additional disk space will be used.
Get:1 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 librealsense2-udev-rules amd64 2.48.0-0~realsense0.4976 [7432 B]
Fetched 7432 B in 1s (13.6 kB/s)                   
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 1.)
debconf: falling back to frontend: Readline
Selecting previously unselected package librealsense2-udev-rules:amd64.
(Reading database ... 164517 files and directories currently installed.)
Preparing to unpack .../librealsense2-udev-rules_2.48.0-0~realsense0.4976_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.48.0-0~realsense0.4976) ...
Setting up librealsense2-udev-rules:amd64 (2.48.0-0~realsense0.4976) ...

```

singhsarvagya, 2021-08-11:
I am having the same issue

singhsarvagya, 2021-08-11:
I run the following commands in Ubuntu 18.04 environment
```
apt-get update 
apt-get install -y apt-utils software-properties-common rsync 

apt-key adv --keyserver keyserver.ubuntu.com --recv-keys C8B3A55A6F3EFCDE
add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u

apt-get update
apt-get install -y librealsense2

```

And I get the following error:

```
Setting up librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5198) ...
+ echo \nPostinst script activated...

Postinst script activated...
+ mkdir -p /home//Documents/librealsense2
+ rsync -avh --ignore-errors usr/share/librealsense2/presets /home//Documents/librealsense2
sending incremental file list
rsync: change_dir "/usr/share/librealsense2" failed: No such file or directory (2)

sent 18 bytes  received 12 bytes  60.00 bytes/sec
total size is 0  speedup is 0.00
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
dpkg: error processing package librealsense2-udev-rules:amd64 (--configure):
 installed librealsense2-udev-rules:amd64 package post-installation script subprocess returned error exit status 23
Processing triggers for libc-bin (2.27-3ubuntu1.4) ...
Errors were encountered while processing:
 librealsense2-udev-rules:amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MartyG-RealSense, 2021-08-11:
Given that the error is related to presets, I wonder whether this problem is related to changes in the preset system that were made in the new **2.49.0** SDK.  They are described in the extract below from the 2.49.0 release notes.  Especially worth noting may be the change of the CMake target for presets to `$ENV{HOME}/Documents/librealsense2/presets`

[image]

daenny, 2021-08-11:
We also have a CI server that installs librealsense2-dev inside a Docker container based on Ubuntu 20.04. 
We encountered first the error that `rsync` was not found. It seems not to be listed asa a dependency of the `udev-rules` and we have a pretty minimal docker. 
Afterwards we have encountered this error: 
```
Errors were encountered while processing:
 librealsense2-udev-rules:amd64
Exit code: 100
Stderr: debconf: delaying package configuration, since apt-utils is not installed
dmesg: read kernel buffer failed: Operation not permitted
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Not sure if that is related?

GiulioRomualdi, 2021-08-11:
Hi all, I think it also happens on Linux-based Github actions when `librealsense2-dev` is installed.

Please check here: https://github.com/dic-iit/bipedal-locomotion-framework/runs/3299490851

workingenius, 2021-08-11:
Tried on ubuntu 18.04, similar error occurs. Maybe they are exactly the same error.
And I reverted to 2.48.0.

MartyG-RealSense, 2021-08-11:
Hi @GiulioRomualdi  If you are installing from packages for an x86 / x64 computer with the instructions on the **distribution_linux.md** page then unless you are compiling librealsense with a special method such as **g++ -std=c++11 filename.cpp**, you may be able to install without using **librealsense2-dev** as that package is optional by default.

robbel, 2021-08-11:
+1 on the same error on Ubuntu 18.04 
```
 installed librealsense2-udev-rules:amd64 package post-installation script subprocess returned error exit status 23
Errors were encountered while processing:
 librealsense2-udev-rules:amd64
```

xperroni, 2021-08-11:
> Hi @GiulioRomualdi If you are installing from packages for an x86 / x64 computer with the instructions on the **distribution_linux.md** page then unless you are compiling librealsense with a special method such as **g++ -std=c++11 filename.cpp**, you may be able to install librealsense without using **librealsense2-dev** as that package is optional by default.

Notice however that the issue is related to package `librealsense2-udev-rules`, which the basic `librealsense2` package depends on. Even if they could live without the `dev` package that would still not help them.

xperroni, 2021-08-11:
I was able to circumvent the `No such file or directory` error message by creating the required directories before installing the `librealsense2-udev-rules` package:

    $ sudo mkdir -p /usr/share/librealsense2/presets
    $ sudo apt install -y librealsense2-udev-rules

This seems to have done the trick:

    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    The following NEW packages will be installed:
      librealsense2-udev-rules
    0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
    Need to get 0 B/7,606 B of archives.
    After this operation, 30.7 kB of additional disk space will be used.
    Selecting previously unselected package librealsense2-udev-rules:amd64.
    (Reading database ... 238953 files and directories currently installed.)
    Preparing to unpack .../librealsense2-udev-rules_2.49.0-0~realsense0.5198_amd64.deb ...
    Unpacking librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5198) ...
    Setting up librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5198) ...
    + echo \nPostinst script activated...

    Postinst script activated...
    + mkdir -p /home/erios/Documents/librealsense2
    + rsync -avh --ignore-errors usr/share/librealsense2/presets /home/erios/Documents/librealsense2
    sending incremental file list

    sent 58 bytes  received 17 bytes  150.00 bytes/sec
    total size is 0  speedup is 0.00

Obviously this is not an acceptable solution, but I hope it helps pinpoint the source of the issue.

MartyG-RealSense, 2021-08-11:
I have highlighted this case to Intel.  Thanks everyone for your reports.  And thanks very much @xperroni for your workaround!

ev-mp, 2021-08-13:
@wngreene , a patch that addresses the installation issue and potentially resolves the Docker compatibility has been released.
Please upgrade the Debians and update.

wngreene, 2021-08-13:
@ev-mp Thanks for the quick response. Just tried the update (2.49.0-0~realsense0.5268). Unfortunately doesn't seem to address the issue(s). Tried installing librealsense2-udev-rules on a fresh Ubuntu:20.04 container and there appear to be at least two problems: 

1. rsync isn't listed as a dependency
2. `/usr/share/librealsense2` isn't created

Here's the output after installing rsync:
```bash
root@3c3a15c1ff30:/# sudo apt install librealsense2-udev-rules
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-udev-rules
0 upgraded, 1 newly installed, 0 to remove and 17 not upgraded.
Need to get 7788 B of archives.
After this operation, 30.7 kB of additional disk space will be used.
Get:1 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 librealsense2-udev-rules amd64 2.49.0-0~realsense0.5268 [7788 B]
Fetched 7788 B in 0s (16.1 kB/s)                   
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 1.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (Can't locate Term/ReadLine.pm in @INC (you may need to install the Term::ReadLine module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.30.0 /usr/local/share/perl/5.30.0 /usr/lib/x86_64-linux-gnu/perl5/5.30 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.30 /usr/share/perl/5.30 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at /usr/share/perl5/Debconf/FrontEnd/Readline.pm line 7, <> line 1.)
debconf: falling back to frontend: Teletype
Selecting previously unselected package librealsense2-udev-rules:amd64.
(Reading database ... 10946 files and directories currently installed.)
Preparing to unpack .../librealsense2-udev-rules_2.49.0-0~realsense0.5268_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5268) ...
Setting up librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5268) ...

Postinst script activated...
Prerequisites are met, copying files
sending incremental file list
rsync: change_dir "/usr/share/librealsense2" failed: No such file or directory (2)

sent 18 bytes  received 12 bytes  60.00 bytes/sec
total size is 0  speedup is 0.00
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
dpkg: error processing package librealsense2-udev-rules:amd64 (--configure):
 installed librealsense2-udev-rules:amd64 package post-installation script subprocess returned error exit status 23
Errors were encountered while processing:
 librealsense2-udev-rules:amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

ev-mp, 2021-08-17:
@wngreene, another Debian drop was published yesterday.
Can you retest and update?
Thanks

wngreene, 2021-08-17:
@ev-mp Great - looks like that update removed the errors:

```bash
root@9e80262f682e:/workdir# sudo apt install librealsense2-udev-rules
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-udev-rules
0 upgraded, 1 newly installed, 0 to remove and 254 not upgraded.
Need to get 7860 B of archives.
After this operation, 30.7 kB of additional disk space will be used.
Get:1 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 librealsense2-udev-rules amd64 2.49.0-0~realsense0.5307 [7860 B]
Fetched 7860 B in 1s (15.2 kB/s)                   
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 1.)
debconf: falling back to frontend: Readline
Selecting previously unselected package librealsense2-udev-rules:amd64.
(Reading database ... 142944 files and directories currently installed.)
Preparing to unpack .../librealsense2-udev-rules_2.49.0-0~realsense0.5307_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5307) ...
Setting up librealsense2-udev-rules:amd64 (2.49.0-0~realsense0.5307) ...

Postinst script activated...
Presets deployment:
rsync processes found -  1
no src dir
tgt dir exists
Presets deployment is skipped
```

Thanks!

ev-mp, 2021-08-17:
@wngreene , thank you for the tests you've done to confirm this.
Is there anything else with this release?

daenny, 2021-08-18:
We are still running into this issue:
```
Presets deployment:
rsync processes found -  1
src dir exists
tgt dir exists
Deploy presets files
/var/lib/dpkg/info/librealsense2-udev-rules:amd64.postinst: 17: rsync: not found
dpkg: error processing package librealsense2-udev-rules:amd64 (--configure):
 installed librealsense2-udev-rules:amd64 package post-installation script subprocess returned error exit status 127
 ```
 I think `rsync` is still not listed as a dependency of the apt package?
 ```
$ apt-cache show librealsense2-udev-rules
Package: librealsense2-udev-rules
Source: librealsense2
Version: 2.49.0-0~realsense0.5307
Architecture: amd64
Maintainer: Raikhel, Evgeni <evgeni.raikhel@intel.com>
Installed-Size: 30
Depends: at  <--- here I would expect rsync as dependency
Section: kernel
Priority: extra
Multi-Arch: same
Homepage: https://github.com/IntelRealSense/librealsense
Description: Intel(R) RealSense(tm) Camera Capture API - udev rules
 This package contains udev rules employed by realsense devices.
Description-md5: 33a7b6c11165e311534fa2d862c196f3
Filename: pool/focal/main/librealsense2-udev-rules_2.49.0-0~realsense0.5307_amd64.deb
SHA1: a9a8a7fb3b6ef99a054d591c822118cee5e73593
SHA256: 871b4ded125eed5b872ae0fd2ee2ec2cb0fec43555702650c61a282199044702
Size: 7860

```

MartyG-RealSense, 2021-08-24:
@daenny   It's great to hear that you had success after installing rsync manually.

@wngreene  Do you require further assistance with this case, please? Thanks!

wngreene, 2021-08-24:
@MartyG-RealSense No further assistance on our end. Thanks!

MartyG-RealSense, 2021-08-24:
Thanks very much @wngreene for the confirmation - I will therefore close the case.

closed 2021-08-24 by MartyG-RealSense