realsenseai/librealsense #9609: Use c# to record video, can't set some parameters？

nevynwong, 2021-08-11:
| Camera Model                       |D415 | 

| Operating System & Version |   {Win 10}  | 

I am using c# to call sdk to record video, can not set some parameters, such as advanced control in Depth table-> Depth units:100; disparity shift: 70；Disparty modulation -> A Factor:0.08 。 And 
[image]
Spatial Filter in Controls, etc.

MartyG-RealSense, 2021-08-11:
The best way to set Advanced Mode parameters in C# would be to define them in a **json** camera configuration file and load the json file with scripting, like in the example cases linked to below.

https://github.com/IntelRealSense/librealsense/issues/2854
https://github.com/IntelRealSense/librealsense/issues/4664

Disparity Shift
**"param-disparityshift": "70"**

A-Factor
**"param-amplitude-factor": "0.08"**

Depth Units
**"param-depthunits": "100"**

nevynwong, 2021-08-12:
Can post-processing be set directly?Or post-processing after recording?

nevynwong, 2021-08-12:
The depth units in Control is set to 0.0001,how to set it in C#?

MartyG-RealSense, 2021-08-12:
To access Advanced Mode in C# you can get to it indirectly through **NativeMethods.cs**

https://support.intelrealsense.com/hc/en-us/community/posts/360033065534-How-to-retrieve-advanced-mode-options-in-C-

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/pinvoke.md

In regard to depth unit scale in C#, I only know of a method for retrieving the current value with scripting, not setting it.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#get-depth-units

Because the depth unit scale of the camera does not change unless deliberately altered though, you can just use a fixed numeric value instead of retrieving it from the camera in real-time (for example, calculating real-world distance by multiplying the 16-bit pixel depth value by 0.0001).

closed 2021-08-13 by nevynwong