realsenseai/librealsense #9614: Compression by MPEG VPCC/GPCC

imapoorvaholla, 2021-08-11:
Hello everyone,
A point cloud(.ply) file is captured from camera D435i. This generated .ply is provided as an input file for MPEG compression VPCC/GPCC, but reconstruction of .ply file is not appropriate.

I have tried the same with sample .ply files from the internet (long dress) and it's working fine. 
Any suggestions what parameter from camera D435i or input file format is missing?

MartyG-RealSense, 2021-08-11:
The section of Intel's white-paper document about ethernet networking that is linked to below states in part **3.5.3 Compression** that lossy compression can be destructive to depth data, and that it is preferable to use lossless compression.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-5-design-considerations

A RealSense user successfully compressed a RealSense .ply file to **Draco** format to send it to a server, because this format has specific support for 3D mesh and point cloud data.

https://github.com/IntelRealSense/librealsense/issues/6865
https://docs.fileformat.com/3d/drc/

Another approach to RealSense compression, albeit with bag files instead of ply, is to record depth data to a bag with the RealSense SDK's **Z16H** compression format.

https://github.com/IntelRealSense/librealsense/issues/9295

MartyG-RealSense, 2021-08-18:
Hi @imapoorvaholla  Do you require further assistance with this case, please?  Thanks!

imapoorvaholla, 2021-08-18:
Hi @MartyG-RealSense Thanks a lot for your reply. We already tried Draco compression. But currently we are interested for VPCC and GPCC  lossy compression only.  Please let us know if you have any example where intel real sense is integrated MPEG VPCC GPCC algorithms?
Thank You.

MartyG-RealSense, 2021-08-18:
There was a paper and accompanying YouTube video published in 2021 about lossy point cloud compression with **PCGC**. that the authors claimed had a significant performance margin over GPCC.

https://www.youtube.com/watch?v=c5T2kUrxUEc
https://njuvision.github.io/PCGCv1/

Whilst the paper does not mention RealSense specificially, it does make use of **TensorFlow**, which the RealSense SDK is compatible with via a TensorFlow wrapper.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow

I was unable to find any RealSense-relevant resources about GPCC and VPCC.

MartyG-RealSense, 2021-08-25:
Thanks very much @imapoorvaholla for the update - good luck!