realsenseai/librealsense #9615: Problem Lidar L515 No Frames Recieved

ClaudeWN, 2021-08-11:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

When I am connecting Lidar L515 i see no picture but get the message "No Frames recieved".

I am using windows 10 pro - I have tried on two different computers with same problem. I am using USB 3.2 port and the original USB wire following with the product.

I have tried to change varios settings - but it does not solve the problem. the software is allowed to use the camera.

MartyG-RealSense, 2021-08-11:
Hi @ClaudeWN  As a starting point in investigating your case, could you unplug the USB cable from the base of the camera (not from the computer end of the cable), turn it around the opposite way and re-insert it into the camera (USB-C cables are two-way insertion at the micro sized end of the cable). Does this make any difference to the camera's performance, please?

MartyG-RealSense, 2021-08-12:
Thank you for testing the connector orientation.

If you are able to access the camera in the RealSense Viewer program and experience the **No Frames Received** message, which stream types (Depth, RGB, Motion Module or all of them) is the message appearing on?

Could you also paste the text of the Viewer's log window into a comment on this discussion, please. The log can be displayed by left-clicking on a small upward arrow in the bottom corner of the Viewer window.

ClaudeWN, 2021-08-12:
Hi
 It is in the depth sensor and RGB stream I got the No Frames Received message
The motion stream seems to work

I will try to copy into toe log window

ClaudeWN, 2021-08-12:
07:43:06 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:06 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\global_timestamp_reader.cpp:239 - Error during time_diff_keeper polling: hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:06 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:06 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:06 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\error-handling.cpp:78 - Error during polling error handler: hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:07 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:07 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\global_timestamp_reader.cpp:239 - Error during time_diff_keeper polling: hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:07 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:07 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\error-handling.cpp:78 - Error during polling error handler: hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:08 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:08 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\global_timestamp_reader.cpp:239 - Error during time_diff_keeper polling: hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:08 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"
07:43:08 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\error-handling.cpp:78 - Error during polling error handler: hr returned: HResult 0x80004002: "En sådan grænseflade understøttes ikke"

MartyG-RealSense, 2021-08-12:
Thanks very much for the log.  There is a past case of this particular set of errors occurring with the L515 camera model in https://github.com/IntelRealSense/librealsense/issues/7736

With the RealSense user in that case, the camera worked on their personal Windows PC and on an Android tablet.  Are both of the PCs that you tried the camera with at a workplace, please?

ClaudeWN, 2021-08-12:
Yes – it is both work place computers

MartyG-RealSense, 2021-08-12:
Are you able to take the camera out of the workplace to test it on a personal computer at home if you have access to one?

If you must keep the camera in the workplace, could you try resetting the camera in the RealSense Viewer to see if it makes a positive difference.  This can be done by going to the **More** option near the top of the Viewer's options side-panel and selecting the **Reset Hardware** option from its drop-down menu.

[image]

Also, could you use the **Info** option that is on the same row as the More option to check which **Firmware Version** is currently installed in your camera?

ClaudeWN, 2021-08-12:
Hi

I did bring the camara home and Connect it to one of my kids personal computer and now it works so it must be the reason

As we are Working with the cameras for robot vision it will be best if we can use them in professional setup.

Do you have any suggestions to get around it? It does not help to reset the firmware

Sendt fra min iPhone

[image]<https://user-images.githubusercontent.com/41145062/129161886-ea543439-86d5-497a-bacd-746c31969884.png>

MartyG-RealSense, 2021-08-12:
Excellent to hear that the camera worked on a home personal PC!

On the workplace PCs, is the camera being plugged into a USB 3 port or slower USB 2?  The firmware that is installed on the L515 camera when new is only compatible with USB 3.  The L515 can now be used with USB 2 ports **but** the firmware has to be updated with the camera on a USB 3 connection first.

ClaudeWN, 2021-08-12:
It should be USB 3 but i will try Update the firmware from the home pc

Sendt fra min iPhone

MartyG-RealSense, 2021-08-12:
I edited my message after the comment about USB 2 to acknowledge that your opening message states that the workplace PCs are USB 3.2.  It would not rule out the possibility that the camera was being mis-detected as being on a USB 2 connection on those PCs though, which would be a problem for the original camera firmware.

Your home PC is likely using Windows 10 Home, so something in Windows Pro on the workplace PCs might be affecting the camera.

MartyG-RealSense, 2021-08-12:
I had a recollection of a previous problem with L515 on Windows 10 Pro.  A RealSense team member recommended checking whether a Windows 10 **group policy** (a Windows Pro feature) could be preventing the camera from working correctly.  

https://github.com/IntelRealSense/librealsense/issues/7952#issuecomment-757500368

MartyG-RealSense, 2021-08-18:
Hi @ClaudeWN  Do you require further assistance with this case, please?  Thanks!

ClaudeWN, 2021-08-18:
We are right now running it on a private PC – I will check with IT organization if we have any administrative limitations. If this is not the problem – I will maybe come back

Thank you

MartyG-RealSense, 2021-08-18:
Thanks very much @ClaudeWN for the update - please do feel free to return if a group policy is not the cause of your problem.

closed 2021-08-30 by MartyG-RealSense