realsenseai/librealsense #9617: Get the velocity of the Realsense D455

Noctstar, 2021-08-11:
| Camera Model                       | D455 | 
| Operating System & Version | Win10| 
| Language                            | python | 

I want to find the velocity of realsense D455 using the motion module.
Is there a function for this in librealsense?

MartyG-RealSense, 2021-08-12:
Hi @Noctstar   The velocity at which the camera is moving would be registered by the IMU's accelerometer, referred to in the librealsense SDK as **accel**.  The link below provides a Python script for accessing and displaying the accel value, and also the IMU's **gyro** value.

https://github.com/IntelRealSense/librealsense/issues/3409#issuecomment-471673994

Noctstar, 2021-08-18:
thank you.  I solved

closed 2021-08-18 by Noctstar

MartyG-RealSense, 2021-08-18:
Thanks very much @Noctstar for the update - it is great to hear that you achieved a soluton!