realsenseai/librealsense #9618: Any noise reduction method in c#?

m-1s, 2021-08-12:
<html>
<body>

Required Info |  
-- | --
Camera Model | L515
Operating System & Version | Win 10
Platform | PC
Language | C#

<h3 style="box-sizing: border-box; margin-bottom: 16px; margin-top: 24px; font-size: 1.25em; font-weight: 600; line-height: 1.25; color: rgb(36, 41, 46); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Issue Description</h3>
</body>
Hi, I'm trying to create a 3d model for some object via c# pointcloud. I noticed that the L515 returns data with very high noise points.
For an example, when I take a shot of a table top, the result I get is a concave, uneven surface
I guess that for a stationary object, we can capture multiple frames, then somehow compare the frames to get a better quality result.
So, just want to know the noise reduction method for librealsense. Or any c# lib to deal with this problem?
</html>

MartyG-RealSense, 2021-08-12:
Hi @m-1s  A range of factors can affect the quality of the depth image, such as environmental / lighting conditions, reflective / glass surfaces and distance of the observed object / surface from the camera.  The L515 in particular works best in controlled indoor lighting conditions with an absence of infrared light sources (sunlight through windows or other cameras that emit infrared, for example).

Would it be possible please to provide an RGB image of the scene that your camera is observing.  This will help to identify possibilities for image improvement before considering additional methods such as applying post-processing filters.  Thanks!

m-1s, 2021-08-12:
Thanks for your reply
 below is the RBG and depth image 2d, depth 3d from my camera.
[image]
[image]
[image]

At  you can see, in the last image, the surface is not good.
When i use them in c# with cloudpoints. These protrusions have a negative effect on my measurements

MartyG-RealSense, 2021-08-12:
The box itself is a good depth image that looks accurate.

What objects do you have on the table if any that are just outside the range of the bottom of the RGB image that might account for the black areas on the bottom of the 2D depth image, please?

The metallic object next to the box may not be showing up well due to reflections.  In the case of the water bottle that is invisible, there was a similar recent case in https://github.com/IntelRealSense/librealsense/issues/9475 that explains why depth detail for the transparent area of the bottle may not register on the L515 camera model.

The black rectangle in the top corner of the depth image would be because of the equivalent black triangle in that location on the RGB image.  It is a general physics principle (not specific to RealSense) that black or dark gray colors absorb light.  the darker the shade, the less light that is returned to the camera.  So such areas may appear as black empty areas without depth values on a depth camera image.  

Projecting a strong light source onto black surfaces can help to bring out depth detail from them.  An example of this of a black office chair is demonstrated in an L515 case in https://github.com/IntelRealSense/librealsense/issues/6757

Also, do you experience improvement in the depth quality if you use the **Short Range** preset in the RealSense Viewer?  

[image]

m-1s, 2021-08-12:
thanks for you support. I will try with the Short range and the noise filtering. 
Have a nice day @MartyG-RealSense

MartyG-RealSense, 2021-08-19:
Hi @m-1s   Do you have an update about this case that you can provide, please?  Thanks!

m-1s, 2021-08-19:
i applied some filter á follow this article, the output is better: 
https://dev.intelrealsense.com/docs/rs-post-processing

MartyG-RealSense, 2021-08-19:
Thanks very much @m-1s for the update - it's excellent to hear that you achieved improvement.

closed 2021-08-31 by MartyG-RealSense