realsenseai/librealsense #9619: RGB stream :No frames Received build with cuda (cuda 11.3) on ubuntu 20.04 (both l515 and d455)

cdb0y511, 2021-08-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455/L515 } | 
| Firmware Version                   | (latest,update from realsense-viewer 2.49) | 
| Operating System & Version |   {Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  (5.11.0-25-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.<4>.<8> & 2.49 }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
If the build with cuda is enable, the realsense-viewer RGB stream can not enable. It states  "No frames Received" and
"ERROR [140611214614528] (synthetic-stream.cpp:48) Exception was thrown during user processing callback:".
I test with L515 and D455, with or without RSUSB backend. The same ERROR occurs RGB stream: No frames Received.
If I disable build with Cuda, then everything is fine.
So I think this may be related to the Cuda 11.3 and Nvidia driver 470.57.02.

cdb0y511, 2021-08-12:
I hope the official release 2.49 could fix it.

MartyG-RealSense, 2021-08-12:
Hi @cdb0y511  May I first confirm whether you have a computer or computing device that is equipped with an Nvidia graphics GPU.  This could include an Nvidia Jetson developer board or a computer that has an Nvidia graphics chip or video card.  CUDA support is only available for Nvidia GPUs.

cdb0y511, 2021-08-13:
> Hi @cdb0y511 May I first confirm whether you have a computer or computing device that is equipped with an Nvidia graphics GPU. This could include an Nvidia Jetson developer board or a computer that has an Nvidia graphics chip or video card. CUDA support is only available for Nvidia GPUs.
Hi,@ MartyG-RealSense 
I test on a high-end laptop equipped with RTX 2080. It used to work smoothly. But after I update the 2.48 or 2.49. The build with Cuda results in No frames Received error. Also, I update the latest Nvidia driver 470.57.02. I am not sure if it is the latest driver result in the error or  SDK 2.48, 2.49. So maybe it needs some investigation.

MartyG-RealSense, 2021-08-13:
A table in the **CUDA** Toolkit release notes suggests matching the 11.3 versions with Toolkit Driver version **265.89**.

https://docs.nvidia.com/cuda/cuda-toolkit-release-notes/index.html

[image]

cdb0y511, 2021-08-13:
> A table in the **CUDA Toolkit** release notes suggests matching CUDA 11.3 with Toolkit Driver version **465.19.01** on Linux.
> 
> https://docs.nvidia.com/cuda/cuda-toolkit-release-notes/index.html
> 
> [image]

Thanks, maybe I should update Cuda to 11.4 update.
I will give a report later.

cdb0y511, 2021-08-13:
Hi,@MartyG-RealSense.

I remove all cuda and old drivers. Then I install cuda 11.4.1 with 470.57.02-1 from
https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=20.04&target_type=deb_local

then, enable SDK 2.49 build with Cuda option, the same error occurs. I doubt the sdk2.48-2.49 may not compatible with Cuda 11.4 and 470.57.02-1 driver.

MartyG-RealSense, 2021-08-13:
Okay, for now I suggest either using SDK 2.47.0 or building 2.49.0 without CUDA support and we can wait and see if anyone else reports the same problem.

cdb0y511, 2021-08-13:
Hi,@MartyG-RealSense 
I solved the problem.
After I "dpkg -l | grep nvidia", and purge all drivers in the list. And reinstall cuda through recover mode, SDK 2.49 can work with the build with cuda.  
The problem is Cuda 11.4 and specific driver compatible issues. 
Thanks

closed 2021-08-13 by cdb0y511

MartyG-RealSense, 2021-08-13:
Great news @cdb0y511 - thanks very much for sharing your solution with the RealSense community to help those who experience a similar situation in future!