realsenseai/librealsense #9622: cmd.xml not found

mehmet-duramaz, 2021-08-12:
I read  this directory which librealsence/tools/terminal/readme. Its says like that command 
$rs-terminal -l cmd.xml -c sequence.txt, Butt i don't understand where is cmd.xml file.

MartyG-RealSense, 2021-08-12:
There is little information available about this tool, but my understanding is that the tool is intended for Intel support team members to send debug commands to the firmware themselves by using cmd.xml or to enable RealSense customers to do it themselves in 'hex' mode by providing the customer with specific debugging hex-codes to input.  An example of a code provided by a RealSense team member can be found in https://github.com/IntelRealSense/librealsense/issues/321#issuecomment-353867626

So unless you are debugging a particular problem with the help of a RealSense team member, the rs-terminal tool may not be helpful to you.

mehmet-duramaz, 2021-08-12:
İ want to code performance programming on jetson tx2 cuda. I have to find ram location raw depth data.

MartyG-RealSense, 2021-08-12:
If you are investigating efficient memory usage on Jetson in librealsense in regard to CUDA and the GPU then the discusssion at https://github.com/IntelRealSense/librealsense/issues/7824 may be worth reading.

MartyG-RealSense, 2021-08-19:
Hi @mehmet-duramaz  Do you require further assistance with this case, please?  Thanks!

mehmet-duramaz, 2021-08-19:
No thanks.

MartyG-RealSense, 2021-08-19:
Thanks very much @mehmet-duramaz for the update.  I will now close this case.  Thanks again!

closed 2021-08-19 by MartyG-RealSense