realsenseai/librealsense #9626: illegal CPU instruction

boom9807, 2021-08-13:
| | |
|---|---|
|**librealsense**|2.43.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

MartyG-RealSense, 2021-08-13:
Hi @boom9807  This is related to an issue that was introduced in SDK **2.36.0** when OpenVINO face detection was incorporated into the Viewer.  Since then, computers that had a processor chip older than the Intel Core 6th Generation (Skylake) from around 2016 experienced the error that you received.  The solution can be to download the standalone .exe version of the Viewer (**Intel.RealSense.Viewer.exe**) from the SDK **Releases** page, as this does not contain the OpenVINO functions.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0

[image]

This standalone Viewer should be launched with a double-leftclick on the Intel.RealSense.Viewer.exe file at the folder location that you downloaded it to, and not with the shortcut icon on the Windows desktop.  If you want to be able to launch the standalone Viewer with a desktop shortcut, delete the original launch icon and then right-click on the Intel.RealSense.Viewer.exe file and select the **Create shortcut** menu option.

[image]

The problem with certain processors (mainly old ones) triggering the error that you experienced and requiring use of the standalone Viewer was dealt with by a change introduced in SDK **2.47.0**.  As you are using the SDK version 2.43.0 though, that version does not contain this fix.

[image]

MartyG-RealSense, 2021-08-20:
Case closed due to correct solution provided and no further comments received.

closed 2021-08-20 by MartyG-RealSense