realsenseai/librealsense #9628: Unreal Engine Point Cloud updates (at least) once a minute

MatthewLKomar, 2021-08-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |   {Win 10)  | 
| Platform                                 | PC  |
| SDK Version                            |  2   | 
| Language                            |  unreal   4.26.2      | 

I managed to compile the ue4  wrapper and going data going out of the camera and into the engine.
For some reason, the point cloud updates once a minute. Is there a way to get it going 30 fps like in the real sense viewer or in Unity? 
In the image below, notice in the D Raw, D Colorized, Color that my hand (bottom left) is raised, but point cloud doesn't reflect it. 

[image]

MatthewLKomar, 2021-08-13:
[image]
[image]

MartyG-RealSense, 2021-08-13:
Hi @MatthewLKomar   In the **Point Cloud** settings, does performance improve if you use the Unreal wrapper documentation's suggested settings of having the **Enable Pcl** setting unticked?  The **Known Issues** section of the documentation advises about performance issues with the PCL pointcloud.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unrealengine4#known-issues

MatthewLKomar, 2021-08-13:
@MartyG-RealSense Yes, unticking Enable Pcl, leads to improved performance (120 FPS now vs 70 FPS before), but I need the point cloud. Unticking it simply doesn't render it.

MartyG-RealSense, 2021-08-13:
And how does it run with PCL enabled and the documentation's suggested voxel size of 1 compared to your 4, please?

MatthewLKomar, 2021-08-13:
120 FPS, but it takes about the same time to switch frames of the point cloud. 
I timed it: it takes 45 seconds to update the PCL frame.

MartyG-RealSense, 2021-08-13:
And if you untick Align Depth to Color, is there any difference in update rate?

MatthewLKomar, 2021-08-13:
No, same thing.

MartyG-RealSense, 2021-08-13:
And if you tick **Enable Polling** in the Device settings?

I realize that this advice amounts to educated guesses of things to try and I apologize, but there are no references for similar situations with the Unreal wrapper to compare with, unfortunately.

MatthewLKomar, 2021-08-13:
Nah, that doesn't change anything either. 
I've been looking at the C++ code and changing it to run at every tick doesn't do anything either, as it updates after 45 seconds as usual.

MatthewLKomar, 2021-08-13:
[image]
got it to work

Commenting out line 477, 478 and 498 
`        //if (bEnablePcl && RsPointCloud.Get() && PclCalculateFlag)
        //{
            if (PclScopedMx.TryLock(&PointCloudMx))
            {
                NAMED_PROFILER("rs2::pointcloud::calculate");
                *RsPoints = RsPointCloud->calculate(DepthFrame);
            }
        //}
`
Makes it work.

MartyG-RealSense, 2021-08-13:
Awesome news - I'm very pleased that you were able to develop a solution.  :)  Thanks for sharing your method with the RealSense community too!

closed 2021-08-13 by MatthewLKomar