realsenseai/librealsense #9630: camera d455 reset to factory default

Owenyu034, 2021-08-14:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D455|
|**Firmware Version**|05.12.15.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|

I misadjusted my  camera d455
I want to return to the original factory calibration
I use Intel.Realsense.CustomRW, but it crash
How to restore the original factory settings

MartyG-RealSense, 2021-08-14:
Hi @Owenyu034  You can repeat the calibration as many times as you wish and then save the calibration with the **Apply New** button once you achieve a good one that you are satisifed with.

Alternatively, you could go to the **Calibration Data** option of the RealSense Viewer's 'More' menu (the option directly beneath 'Tare Calibration' to bring up the **Camera Calibration** window.  

Left-click on the black box in the bottom corner beside **I know what I'm doing** to change the box to blue (On).  This unlocks the 'Restore Factory' option at the top of the Camera Calibration window.  

Left-click on 'Restore Factory'.  You may see a brief blue flash in the window as the camera's factory defaults for calibration replace the calibration values that were previously displayed.

[image]

Owenyu034, 2021-08-14:
Thank you  for your suggestions! My camera has been successfully restored to the original factory settings.

MartyG-RealSense, 2021-08-14:
Great news @Owenyu034   :)  Thanks for the update!

closed 2021-08-21 by MartyG-RealSense