realsenseai/librealsense #9635: depth  bitmap value Missing and  point_3D value  all are 0

dota2mhxy, 2021-08-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|-------------------------------- |------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   android8  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  android8  |
| SDK Version                           |  sdk2.0 }                          | 
| Language                              | android }                          | 
| Segment			        | tablet PC                         | 

### Issue Description
Hello, now when I take pictures outdoors with my d435 camera, I want to save the color map and depth map. The depth map is also the function getdata() to obtain the data and convert it into bitmap. At the same time, I also convert each pixel into three-dimensional coordinates and save it.this is my  code from  github
[image]
[image]

 I could work sometiomes,。The indoor depth data acquisition is relatively complete, that is, the color map is relatively dark。
this  is  indoor result
[image]

these  are outdoor results
[image]
[image]
this  is   i  saved     three-dimensional coordinates data
[image]

 but more photos were taken . The depth maps were black and the three-dimensional coordinates were all 0. 
[image]
[image]

At that time, the light was strong and the distance was 1-4 meters, or for other reasons. Thank you for your reply

MartyG-RealSense, 2021-08-16:
Hi @dota2mhxy  Can you confirm whether the problem with zero depth and the broken depth image occurs on the second bitmap capture after having successfully captured the first time, please?

Do the color bitmaps save correctly on the second capture?

dota2mhxy, 2021-08-16:
Hello, I can correctly capture the color map and depth map indoors and outdoors in the evening of last Saturday. Today, I experimented again. There are frequent problems before. The color map can be saved normally, but it is a little overexposed because of the strong sunlight, but the depth map is mostly dark or only a few colors, At the same time, the three-dimensional coordinates are basically 0. I consider that in addition to the influence of illumination and distance, it may take me too much time to save the depth coordinates and write the file, about 80s. At the same time, I have a counter in the code, which may also have an impact
`     private int framesCounter = 1;
    Runnable mStreaming = new Runnable() {
        @Override
        public void run() {
            try {
                try(FrameSet frames = mPipeline.waitForFrames()) {
                    try(FrameSet processed = frames.applyFilter(mColorizer)) {
                        mGLSurfaceView.upload(processed);
                        if(++framesCounter % 16 == 14) {
                            // getting depth frame
                            DepthFrame depthFrame = frames.first(StreamType.DEPTH).as(Extension.DEPTH_FRAME);
                          // DepthSensor depthSensor = frames.first(StreamType.DEPTH).as(Extension.DEPTH_SENSOR);
                            // getting color frame/
                            // ColorFrame depthFrame = frames.first(StreamType.DEPTH).as(Extension.DEPTH_FRAME);
                            Intrinsic depthFrameIntrinsic = depthFrame.getProfile().getIntrinsic();
                            // getting depth frame field of view
                            float fov[] = Utils.getFov(depthFrameIntrinsic);
                            Log.d(TAG, "fov = (" + fov[0] + ", " + fov[1] + ")");
                            // getting color frame
                            VideoFrame colorFrame = frames.first(StreamType.COLOR).as(Extension.VIDEO_FRAME);
                            Intrinsic colorFrameIntrinsic = colorFrame.getProfile().getIntrinsic();`

MartyG-RealSense, 2021-08-16:
Assuming that auto-exposure is enabled, you could test whether the infrared sensors that generate the depth frame are being saturated by the strong light if you cover the front of the camera with your hand for several seconds to block the camera's view before removing the hand and activating the capture.  If it is related to light saturation, covering the camera should cause the auto-exposure to automatically correct the problem.

dota2mhxy, 2021-08-17:
I take pictures  indoors. The photos are often very dark. I block the camera with my hand for a few seconds and there is no automatic exposure. 
this  is   d435  
[image]

this  is   tablet  camera
[image]

What’s more important is the lack of depth data when taking pictures outdoors. I don’t know if you have any suggestions.

MartyG-RealSense, 2021-08-17:
The darkness of the RGB image suggests that RGB auto-exposure is not enabled.  This could account for the difference between your indoor and outdoor RGB images at the beginning of this case, as the RGB image will be brighter when RGB auto-exposure is disabled if there is more light present in the observed scene.

You can configure camera configuration in Android by loading a custom json file with the instruction **loadPresetFromJson** - an example of such an implementation in Android can be found in https://github.com/IntelRealSense/librealsense/issues/5303#issuecomment-558898195

Quoting from the json section of the script in that link:

```
 try {
                    byte[] presetData = getByteArrayFromResource(FacialProcessingApp.getContext(), R.raw.high_res_high_density_preset, "high_res_high_density_preset.json");
                    device.loadPresetFromJson(presetData);
                    Log.d(TAG, "Preset Set!");
                } catch (IOException e) {
                    Log.e(TAG, "failed to set preset, failed to open preset file, error: " + e.getMessage());
                } catch (Exception e) {
                    Log.e(TAG, "failed to set preset, error: " + e.getMessage());
                }

                config.disableAllStreams();
                config.enableStream(StreamType.COLOR, 0, CommonConstants.getInstance().colourStreamWidth, CommonConstants.getInstance().colourStreamHeight, StreamFormat.YUYV, CommonConstants.getInstance().depthCamFPS);
                config.enableStream(StreamType.DEPTH,0,CommonConstants.getInstance().depthStreamWidth,CommonConstants.getInstance().depthStreamHeight,StreamFormat.Z16,CommonConstants.getInstance().depthCamFPS);
            }
        }

```

dota2mhxy, 2021-08-17:
i  used  this code,It is not easy to use, there are some classes and methods that are not available,CommonConstants class  and
getByteArrayFromResource  method missed，and  i  only  use     if (!device.isInAdvancedMode()){
                                                                                                       device.toggleAdvancedMode(true);
and pic  is  still  dark

MartyG-RealSense, 2021-08-17:
How do the RGB and depth images look if you use the Android wrapper's **Capture** example that displays the depth and color images in the same window?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture

[image]

dota2mhxy, 2021-08-18:
Thank you for your thoughts. I ran the capture example and found that the color map is bright, not grayed out, and the depth map data is quite complete. Then combined with my own code, I think it may be caused by using the stop() method to end the pipeline and then using start() to open it. I made the corresponding changes and wrote a button to trigger the independent save function.

MartyG-RealSense, 2021-08-18:
Thanks very much @dota2mhxy for the update.  Does your updated cide now perform satisfactorily?

dota2mhxy, 2021-08-19:
Hello, can the d435 camera zoom in to zoom in part of the area？

MartyG-RealSense, 2021-08-19:
2D RGB graphics could possibly be zoomed in Java using the **graphics2d** Java function, though I do not have an example of doing so in Java in librealsense.

https://www.baeldung.com/java-resize-image

In regard to scaling a live-streamed non-pointcloud depth image like the **Capture** image above, changing the **depth unit scale** could achieve this effect.  For example, changing the depth scale from the default '0.001' to '0.0001' can cause the depth view to zoom in.

closed 2021-08-24 by dota2mhxy