realsenseai/librealsense #9636: Can I Create a framefrom a pre-allocated buffer?

mik0der, 2021-08-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / (v2.47.0) }                          | 
| Language                            | C++ and C#                        | 
| Segment			|                            | 

### Issue Description
Hi, I am trying to get depth frame from C# WPF aplication to C++ native library, where I want to compute pointclouds. Is there a way to recreate frame object from pre-alocated buffer? Something like this, for Azure Kinect (https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gaf84f2a271bcf6afae429bbccd47071b3.html#gaf84f2a271bcf6afae429bbccd47071b3). Because sending whole DepthFrame object from C# to C++ doesnt work.

MartyG-RealSense, 2021-08-16:
Hi @mik0der   I researched your question carefully but could not find a reference about buffer pre-allocation in the RealSense SDK.  However, RealSense users who wish to create a WPF capture application are usually referred to the RealSense SDK's C# **Capture** example project in the link below.  It captures depth and color and renders them to side-by-side windows

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/tutorial/capture

mik0der, 2021-08-16:
Hi, thanks for reply. Is it possible to compute point cloud in c# wrapper?

MartyG-RealSense, 2021-08-16:
Yes, it is possible.  The RealSense SDK's **C# Cookbook** page of C# example scripts has a point cloud script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#generics-support

MartyG-RealSense, 2021-08-23:
Hi @mik0der  Do you require further assistance with this case, please?  Thanks!

mik0der, 2021-08-23:
Hey, thank you again for your previous answer, I managed to solve my problem within c# wrapper and didnt have to do things related to my original question. You can close this thread.

MartyG-RealSense, 2021-08-23:
Excellent news that you found a solution, @mik0der - thanks very much for the update!  As suggested, I will close this issue.  Thanks again.

closed 2021-08-23 by MartyG-RealSense