realsenseai/librealsense #9637: rs-server on Windows

terranas, 2021-08-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |

### Issue Description
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras
I would like to apply the contents of this article to a solution I am currently designing.
However, this article seems to assume a Linux PC (Raspberry Pi 4) as the server machine, but I want to use Windows 10 for the server.
Is there any plan to make rs-server workable on Windows?

I built rs-server on Windows to try:
1. git clone https://github.com/IntelRealSense/librealsense.git
2. cd librealsense\tools\rs-server
3. Open CMakeLists.txt and delete 'if(WIN32) message(...)'
4. mkdir build; cd build; cmake ..
5. Open rs-server.sln, add several path and build
6. I got a lot of link errors for live555

Do these link errors have anything to do with why rs-server is not supported on Windows?
Is there any possibility to build completely to the end without link errors on Windows?
Thank you.

MartyG-RealSense, 2021-08-17:
Hi @terranas  **rs-server** currently can only be compiled on Linux.  I do not have information about the reason for this.

Would it be practical for you to set up your remote computer to be **dual boot** (both Windows and Linux installed on it) so that you can set up rs-server on the Linux OS and boot into Linux mode when running the computer as a remote server?

https://opensource.com/article/18/5/dual-boot-linux

MartyG-RealSense, 2021-08-29:
Case closed due to no further comments received.

closed 2021-08-29 by MartyG-RealSense

terranas, 2021-09-09:
Sorry for the delay.
After that, I made some changes to the original code, and run successfully rs-server on Windows 10.
Thank you very much for releasing useful whitepaper.

[image]

MartyG-RealSense, 2021-09-09:
That's excellent news @terranas that you were successful - thanks very much for the update   :)

jbruno79, 2021-09-29:
Hi @terranas,
I am trying to build the rs-server on Windows 10 and it does not work for me, could you tell what changes you have made to make it work?
Best regards,
Julian

terranas, 2021-09-30:
Hi @jbruno79 
Please apply following patch to the tag `v2.49.0`.
Most important is linking `ws2_32.lib`. (I wonder why the development team didn't do this)
I hope it works :)
```diff
 tools/CMakeLists.txt           |  6 +--
 tools/rs-server/CMakeLists.txt | 83 ++++++++++++++++++----------------
 2 files changed, 46 insertions(+), 43 deletions(-)

diff --git a/tools/CMakeLists.txt b/tools/CMakeLists.txt
index b1b967af2..13f05a984 100644
--- a/tools/CMakeLists.txt
+++ b/tools/CMakeLists.txt
@@ -19,10 +19,8 @@ add_subdirectory(terminal)
 add_subdirectory(recorder)
 add_subdirectory(fw-update)
 
-if(NOT WIN32)
-    if(BUILD_NETWORK_DEVICE)
-        add_subdirectory(rs-server)
-    endif()
+if(BUILD_NETWORK_DEVICE)
+    add_subdirectory(rs-server)
 endif()
 endif()
 
diff --git a/tools/rs-server/CMakeLists.txt b/tools/rs-server/CMakeLists.txt
index 617a72379..bbb5a0be0 100644
--- a/tools/rs-server/CMakeLists.txt
+++ b/tools/rs-server/CMakeLists.txt
@@ -20,46 +20,51 @@ SET(CMAKE_EXE_LINKER_FLAGS ${CMAKE_EXE_LINKER_FLAGS} "-pthread")
 
 set(LIVE ${CMAKE_BINARY_DIR}/third-party/live)
 
+file(GLOB LIVE_SOURCES LIST_DIRECTORIES false CONFIGURE_DEPENDS
+"../../src/ipDeviceCommon/*.c*"
+"${LIVE}/*.c*"
+"${LIVE}/groupsock/*.c*"
+"${LIVE}/BasicUsageEnvironment/*.c*"
+"${LIVE}/liveMedia/*.c*"
+"${LIVE}/UsageEnvironment/*.c*"
+)
+source_group("LIVE555" FILES ${LIVE_SOURCES})
+file(GLOB RS_SERVER_SOURCES LIST_DIRECTORIES false CONFIGURE_DEPENDS
+"*.c*"
+)
+list(APPEND RS_SERVER_SOURCES ${LIVE_SOURCES})
+list(REMOVE_ITEM RS_SERVER_SOURCES "${LIVE}/liveMedia/ADTSAudioStreamDiscreteFramer.cpp")
+if (${BUILD_SHARED_LIBS} AND ${BUILD_EASYLOGGINGPP})
+    list(APPEND RS_SERVER_SOURCES ../../third-party/easyloggingpp/src/easylogging++.cc)
+endif()
+add_executable(${PROJECT_NAME} ${RS_SERVER_SOURCES})
+add_definitions(-DELPP_NO_DEFAULT_LOG_FILE)
+
+include_directories(
+../../common
+../../src
+../../src/ipDeviceCommon
+../../include/librealsense2
+../../third-party/tclap/include
+../../third-party/easyloggingpp/src
+${LIVE}/groupsock/include
+${LIVE}/liveMedia/include
+${LIVE}/UsageEnvironment/include
+${LIVE}/BasicUsageEnvironment/include
+)
+
+set_property(TARGET ${PROJECT_NAME} PROPERTY CXX_STANDARD 11)
+
+set(DEPENDENCIES ${DEPENDENCIES} realsense2 Threads::Threads realsense2-compression ${ZLIB_LIBRARIES} ${JPEG_LIBRARIES})
+
 if(WIN32)
-  message(SEND_ERROR "rs-server supports Linux only")
-else()
-  file(GLOB RS_SERVER_SOURCES LIST_DIRECTORIES false CONFIGURE_DEPENDS
-    "*.c*"
-    "../../src/ipDeviceCommon/*.c*"
-    "${LIVE}/*.c*"
-    "${LIVE}/groupsock/*.c*"
-    "${LIVE}/BasicUsageEnvironment/*.c*"
-    "${LIVE}/liveMedia/*.c*"
-    "${LIVE}/UsageEnvironment/*.c*"
-  )
-  list(REMOVE_ITEM RS_SERVER_SOURCES "${LIVE}/liveMedia/ADTSAudioStreamDiscreteFramer.cpp")
-  add_executable(${PROJECT_NAME} ${RS_SERVER_SOURCES})
-  add_definitions(-DELPP_NO_DEFAULT_LOG_FILE)
-
-  include_directories(
-    ../../common
-    ../../src
-    ../../src/ipDeviceCommon
-    ../../include/librealsense2
-    ../../third-party/tclap/include
-    ../../third-party/easyloggingpp/src
-    ${LIVE}/groupsock/include 
-    ${LIVE}/liveMedia/include  
-    ${LIVE}/UsageEnvironment/include 
-    ${LIVE}/BasicUsageEnvironment/include     
-  )
-
-  set_property(TARGET ${PROJECT_NAME} PROPERTY CXX_STANDARD 11)
-
-  set(DEPENDENCIES ${DEPENDENCIES} realsense2 Threads::Threads realsense2-compression ${ZLIB_LIBRARIES} ${JPEG_LIBRARIES})
-  
-  target_link_libraries(${PROJECT_NAME} ${DEPENDENCIES})
-  
-  set_target_properties(${PROJECT_NAME} PROPERTIES FOLDER "Tools")
-
-  target_compile_definitions(${PROJECT_NAME} PUBLIC RESPONSE_BUFFER_SIZE=100000)
-
-  install(TARGETS ${PROJECT_NAME} RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})
+	set(DEPENDENCIES ${DEPENDENCIES} ws2_32)
 endif()
 
+target_link_libraries(${PROJECT_NAME} ${DEPENDENCIES})
+
+set_target_properties(${PROJECT_NAME} PROPERTIES FOLDER "Tools")
+
+target_compile_definitions(${PROJECT_NAME} PUBLIC RESPONSE_BUFFER_SIZE=100000)
 
+install(TARGETS ${PROJECT_NAME} RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})
```