realsenseai/librealsense #9638: "No frames received" for RGB Module

asb2111991, 2021-08-16:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|920312070806|
|**Firmware Version**|05.12.07.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|925623027108|
|**Firmware Update Id**|925623027108|

Please provide a description of the problem

MartyG-RealSense, 2021-08-17:
Hi @asb2111991  Are you using the RealSense Viewer program, please?  And if you are using the Viewer, are you enabling the RGB stream on its own, or do you have Depth, RGB and Motion Motion all enabled at the same time?

asb2111991, 2021-08-19:
Hi @MartyG-RealSense, Thank you for responding. Yes, I am using the Viewer. I have enabled Depth and the RGB, and also tried to enable RGB without enabling the Depth. In both cases, 'No frames received".
I have tested on a different computer, same result. I have tried installing the 'Recommended Firmware', nothing seems to work
Need help badly

MartyG-RealSense, 2021-08-19:
Thank you for the information.  Next, could you confirm please that access to the camera is set to On in the Windows privacy settings for both camera hardware and camera applications.  This can be done with instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

asb2111991, 2021-08-19:
Yes, that was one of the very first things that I checked and it was already enabled.

asb2111991, 2021-08-19:
Just installed the latest SDK and 'Recommended Firmware', it doesn't help. If it wasn't clear from earlier correspondence, the Stereo Module works fine and give me the depth map

MartyG-RealSense, 2021-08-19:
Please use the instructions in the link below to check if RGB video can be displayed in the Windows **Camera** app.

https://github.com/IntelRealSense/librealsense/issues/8711#issuecomment-811945394

If you are using a laptop that has a built-in webcam, you could also try temporarily disabling the webcam in the Windows Device Manager to eliminate the possibility that there is a conflict between that built-in camera and the RealSense camera.

If you are using a laptop with such a webcam, you can go to the **Cameras** section of the Device Manager , right-click on the entry representing the webcam and select the menu option **Disable device**.  

If disabling it does not make a difference then you can re-enable it by right-clicking on the webcam again and selecting **Enable device**.

asb2111991, 2021-08-19:
Disabling my webcam did not make a difference and when I opened the camera app to see if that works with realsense, it throws an error saying 'Device could not be started'.

MartyG-RealSense, 2021-08-19:
Has the RGB stream ever worked since you purchased the camera, or was it working correctly up until now and then suddenly stopped working?

asb2111991, 2021-08-19:
It has worked perfectly for months. I have developed an object detection solution using that in Python. It has not been connected to the computer in the past three months, during when it has been sitting in the warehouse.

MartyG-RealSense, 2021-08-19:
There have been a couple of cases this year where this problem has occurred with No Frames Received the first time that the camera is used after a period of inactive storage in a cool dry location for weeks / months, and worked correctly before storage.  I will seek advice from Intel about your particular case.  I recommend putting the camera aside for now in the meantime.  Thanks very much for your patience.

MartyG-RealSense, 2021-08-19:
I have received a response from Intel.  Could you send me an email please so that I can pass your email address to Intel with your permission.  They can then follow up with you about your camera problem.  You can contact me at:

martyx.grover@intel.com

asb2111991, 2021-08-20:
I have sent the email, please confirm here or by responding via email. Thank you again for your help.

MartyG-RealSense, 2021-08-20:
Thanks very much, I can confirm that I have received your email and will forward it to the appropriate Intel RealSense team member.

closed 2021-09-15 by RealSenseCustomerSupport

robomotic, 2021-09-21:
Just as proof

[image]

asb2111991, 2021-09-21:
> Yes just tested now (disabled by camera) and same issue as before argh!

In my case, it was due to humidity. The stock Realsense cameras are tested only up to 35 Celcius. If the ambient temperatures go beyond and the powered-off camera is left in such an environment for too long, this can happen.

Again, this was the cause for MY camera. Keep in indoors in a dark moisture-free place for 4 or 5 hours and if your situation is similar to mine, it should go away.