realsenseai/librealsense #9639: fatal error: 'common/fw/target.h' file not found

david-sullivan, 2021-08-16:
trying to install on mac os big sur. 
following the directions on the osx install docs.
when i encountered this problem i tried both of the suggestions in this issue:
[https://github.com/IntelRealSense/librealsense/issues/6377](url)
neither worked.

if the mac install was easier, i might be more willing to buy some sensors for our lab.

MartyG-RealSense, 2021-08-17:
Hi @david-sullivan  I hope that the advice about the **'common/fw/target.h' file not found** error in the Mac installation case https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-690595112 will be helpful to you.

Some RealSense Mac users have found it easier to just use the instruction **brew install librealsense**

https://formulae.brew.sh/formula/librealsense

david-sullivan, 2021-08-17:
Thank you @MartyG-RealSense  the homebrew instructions `brew install librealsense` did exactly what i needed and was much simpler. Why isn't that mentioned in the docs?
of course, the reason for the fatal error isn't solved, but my interest in that problem is dissolved.

MartyG-RealSense, 2021-08-17:
I'm pleased to hear that the advice for helpful to you!

A note has recently been made for the RealSense team to consider adding information about **brew install librealsense** to the MacOS installation instructions page at a future date.

closed 2021-08-24 by MartyG-RealSense