realsenseai/librealsense #9643: Can't get d410 to work with vision board

GrubbyHalo, 2021-08-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-08-17:
Hi @GrubbyHalo   May I first ask the questions below, please:

-  Have you connected the Vision Processor D4 board and and the D410 Depth Module Kit board together with a rigid **interposer board** or with a flexible interposer cable?

-  If you have successfully joined the two boards together with an interposer mechanism, have you calibrated the camera afterwards?

-  What computer / computing device is the camera attached to (PC, Raspberry Pi, Nvidia Jetson, etc)

-  If you are using the librealsense SDK on Linux, did you build the SDK from source code or build it from Debian packages?

GrubbyHalo, 2021-08-18:
Hi @MartyG-RealSense 

The Rigid interposer is being used

W.r.t. the calibrating, I can't run any software to calibrate the camera's as they don't get recognised by the realsense-viewer etc..

I have only connected this to a Raspberry pi. However I have other vision boards and camera modules thad DO contain the IR projector that do work... It just seems like the module without the IR projector does'nt work.

The realsense SDK was built from source.. again, I have other camera's that do work on this rig but not the one i recently purchased that does not contain the IR projector...

Regards

MartyG-RealSense, 2021-08-18:
The interposer part number confirms that you have the correct part.

Can you confirm please that the two boards are fully seated in their 50 pin connectors on the rigid interposer board, and that the Vision Processor D4 board is on the lower white connector and the D420 module on the upper connector? 

The lack of an IR projector should not make a difference to whether the camera works successfully as the projector is a separate component from the IR imaging sensors on the depth module board.

GrubbyHalo, 2021-08-18:
Hi @MartyG-RealSense 

Yes the modules are correctly seated 

I managed to update the firmware on the vision boards buy plugging a camera module (450) into the vision board (That was the only way i could update it)

After I did this, the camera was recognised by my own custom software but bombed out after a minute as it had not yet recieved any frames from the module. I updated the realsense SDK to the latest version and all the tools and examples also don't receive any frames..

MartyG-RealSense, 2021-08-18:
Is the D420 now at least able to be detected correctly by the RealSense Viewer in its options side-panel, even if the depth streaming does not work when enabled?

GrubbyHalo, 2021-08-18:
I get no frames received in the realsense viewer

[image]

MartyG-RealSense, 2021-08-18:
Can you open the Viewer's debug log by left-clicking on the small upward triangle in the bottom corner and provide an image of the log when you have No Frames Received on the depth stream, please.

[image]

GrubbyHalo, 2021-08-18:
Here we go[image]

GrubbyHalo, 2021-08-19:
It seems like the issue might have been a faulty transposer. After swapping it out it seems to work.

closed 2021-08-19 by GrubbyHalo

MartyG-RealSense, 2021-08-19:
Excellent news that you found a likely cause, @GrubbyHalo - thanks for the update!