realsenseai/librealsense #9645: UWP Driver

electro-logic, 2021-08-17:
Hello,

Do you know if there is a way (maybe with the RS400 device MFT KS controls) to do the following with the D400 DMFT UWP driver?
- Change advanced parameters (ex. Disparity Shift)
- Get the extrinsic calibration data for the depth/color camera
- Load a JSON preset

Thank you,
Leonardo

MartyG-RealSense, 2021-08-17:
Hi @electro-logic   May I begin by asking please if you are planning to use the camera to interface with a product that supports Microsoft's **Universal Windows Platform (UWP)** standard, such as the Microsoft HoloLens headset?  If you do not need to interface with such a product then you do not need to have the UWP driver installed to use the camera with Windows.

electro-logic, 2021-08-17:
Hi @MartyG-RealSense, yes I need to interface the camera with UWP-only software.

MartyG-RealSense, 2021-08-17:
There are not many librealsense UWP cases to refer to, but the one in the link below may provide useful insights, as the RealSense user in that case also wanted to access extrinsics via UWP.

https://github.com/IntelRealSense/librealsense/issues/2702

Within the discussion, an Intel RealSense team member provided advice about how it might be achieved.

https://github.com/IntelRealSense/librealsense/issues/2702#issuecomment-438586001

electro-logic, 2021-08-17:
@MartyG-RealSense thank you very much, I managed to get the projection matrix from depth to RGB and is matching values from rs-sensor-control. Definitively a D400-specific approach, but it works great.

MartyG-RealSense, 2021-08-18:
Excellent to hear that you achieved a solution, @electro-logic - thanks for the update!

MartyG-RealSense, 2021-08-24:
Case closed due to solution achieved and no further comments received.

closed 2021-08-24 by MartyG-RealSense

electro-logic, 2021-08-30:
Hello @MartyG-RealSense, I am now trying to update the Depth Control Table.

I'm sending HWRST after EN_ADV to enable the Advanced Mode and the camera is correctly resetting.

The problem is that the follow GET_ADV command is returning the wrong number of bytes (1310720).

Is the Advanced Mode working under UWP or is the driver not supporting reset operations?

Even the UAMG (get advanced mode status) command is returning the wrong number of bytes.

SENT 20,0,171,205,48,0,....,0
RECEIVED 48,0,0,0,1,16,3,0,6,...0,1,0,0,0

so in this case I am receiving the right opcode back (UAMG=48) but the wrong result size (0,1,0,0->0,0,1,0->65536)

Thank you,
Leonardo