realsenseai/librealsense #9646: Segmentation fault when trying to calculate pointcloud from depth

banuprathap, 2021-08-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Ubuntu 14.04.5  | 
| Kernel Version (Linux Only)    |  4.4.0-31-generic                                         | 
| Platform                                 | Com Express module with i7-6600U |
| SDK Version                            |  v2.48.0                          | 
| Language                            |  C++                         | 
| Segment			|  Robot                          | 

### Issue Description

Reopening #8031 to find a permanent solution. With the workaround I made in #8031 performance takes a significant hit (sometimes taking >300ms to compute pointcloud for a single frame).

I am trying to read a depth image from the device and generate pointcloud from the depth data. I followed the example code to create a test program but I get segfault when I call `pc.calulate(depth)`.

``` cpp
    rs2::pipeline p;
    rs2::pointcloud pc;
    rs2::config c;
    c.enable_device(serial_number);
    p.start();

    while (true)
    {
        rs2::frameset frames = p.wait_for_frames();
        rs2::depth_frame depth = frames.get_depth_frame();
        points = pc.calculate(depth);
    }
```
GDB shows that segfault is at [`librealsense::depth_to_points` at line 95](https://github.com/IntelRealSense/librealsense/blob/v2.48.0/src/proc/sse/sse-pointcloud.cpp#L95) but I am not familiar with SSE stuff to troubleshoot further. Please help.

Full backtrace from GDB: 

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x9b4eab40 (LWP 612)]
librealsense::pointcloud_sse::depth_to_points (this=<optimized out>, output=..., depth_intrinsics=..., depth_frame=..., depth_scale=<optimized out>)
    at librealsense-2.48.0/src/proc/sse/sse-pointcloud.cpp:95
95      librealsense-2.48.0/src/proc/sse/sse-pointcloud.cpp: No such file or directory.
(gdb) bt full
#0  librealsense::pointcloud_sse::depth_to_points (this=<optimized out>, output=..., depth_intrinsics=..., depth_frame=..., depth_scale=<optimized out>)
    at librealsense-2.48.0/src/proc/sse/sse-pointcloud.cpp:95
        y_z0 = <optimized out>
        z_x1 = <optimized out>
        xyz13 = <optimized out>
        depth0 = <optimized out>
        y_z1 = <optimized out>
        xyz01 = <optimized out>
        d0 = <optimized out>
        x_y0 = <optimized out>
        xyz02 = <optimized out>
        xyz11 = <optimized out>
        d1 = <optimized out>
        z_x0 = <optimized out>
        xyz03 = <optimized out>
        x_y1 = <optimized out>
        xyz12 = <optimized out>
        i = 0
        size = <optimized out>
        point = 0xa5e04008
        mapx = 0xa4d8ee50
        mapy = <optimized out>
#1  0xb72c8fab in librealsense::pointcloud::process_depth_frame (this=0xa4d18824, source=..., depth=...)
    at librealsense-2.48.0/src/proc/pointcloud.cpp:257
        res = {<rs2::frame> = {frame_ref = 0xa4d25e60}, _size = 19200}
        points = <optimized out>
        vid_frame = {<rs2::frame> = {frame_ref = 0x9b4e8ecc}, <No data fields>}
        mapped_intr = {width = -1689350368, height = -1208040703, ppx = -2.69298762e-05, ppy = 0, fx = 1.40129846e-45, fy = 7.00649232e-45, model = RS2_DISTORTION_NONE, coeffs = {
            -1.40331686e-05, -1.40331686e-05, -1.70863233e-22, -1.70861315e-22, -1.40458287e-05}}
        extr = {rotation = {-9.123286e-17, 2.69049305e-41, -1.70861618e-22, 58.8245544, 95.7468643, -7.05332423e-06, 5.60519386e-45, 0, 0}, translation = {-1.40331686e-05, 
            -1.40331686e-05, -1.70862779e-22}}
        map_texture = <optimized out>
#2  0xb72cb093 in librealsense::pointcloud::process_frame (this=0xa4d18824, source=..., f=...)
    at librealsense-2.48.0/src/proc/pointcloud.cpp:382
        composite = {<rs2::frame> = {frame_ref = 0x0}, _size = 0}
        rv = {frame_ref = 0x0}
#3  0xb72d628d in operator() (source=..., f=..., __closure=0xa4d0074c)
    at librealsense-2.48.0/src/proc/synthetic-stream.cpp:75
        res = {frame_ref = 0x0}
        f = {frame_ref = 0xa977b9f0}
        __for_range = std::vector of length 1, capacity 1 = {{frame_ref = 0xa977b9f0}}
        frames_to_process = std::vector of length 1, capacity 1 = {{frame_ref = 0xa977b9f0}}
        results = std::vector of length 0, capacity 0
        out = {frame_ref = 0x1}
        this = 0xa4d18824
#4  rs2::frame_processor_callback<librealsense::generic_processing_block::generic_processing_block(char const*)::<lambda(rs2::frame, const rs2::frame_source&)> >::on_frame(rs2_frame *, rs2_source *) (this=0xa4d00748, f=<optimized out>, source=0xa4d00738)
    at librealsense-2.48.0/include/librealsense2/hpp/rs_processing.hpp:128
        src = {_source = 0xa4d00738}
        frm = {frame_ref = 0x0}
#5  0xb72d4c20 in librealsense::processing_block::invoke (this=0xa4d18824, f=...)
    at librealsense-2.48.0/src/proc/synthetic-stream.cpp:43
        ptr = <optimized out>
        callback = {invocation = 0xa4d81408, owner = 0xa4d81408}
#6  0xb73e28c4 in rs2_process_frame (block=<optimized out>, frame=<optimized out>, error=0x9b4e920c)
    at librealsense-2.48.0/src/rs.cpp:2102
        __FUNCTION__ = "rs2_process_frame"
#7  0xb730f47a in invoke (f=..., this=0x9b4e9358)
    at librealsense-2.48.0/include/librealsense2/hpp/rs_processing.hpp:273
        ptr = <optimized out>
        e = 0x0
#8  rs2::filter::process (this=0x9b4e9358, frame=...)
    at librealsense-2.48.0/include/librealsense2/hpp/rs_processing.hpp:354
        f = {frame_ref = 0x78}
#9  0xb730f9b1 in rs2::pointcloud::calculate (this=0x9b4e9358, depth=...)
    at librealsense-2.48.0/include/librealsense2/hpp/rs_processing.hpp:433
        res = {frame_ref = 0x78}
#10 0xb227cd64 in (anonymous namespace)::RsSensorServer::processDepth (this=0x820c2e0, frame=..., tFrame=...) at src/RsSensorServer.cpp:894
```

MartyG-RealSense, 2021-08-18:
Hi @banuprathap  After defining a custom config term with **rs2::config** you should put your config term in the brackets of the pipe start instruction, otherwise the pipeline will ignore the custom configuration.  so in the case of your particular script, it would be:

**p.start(c);**

banuprathap, 2021-08-18:
Thanks @MartyG-RealSense  for pointing the config usage. I'll make sure to pass it as a parameter. Do you have any info on the SSE related segfault?

MartyG-RealSense, 2021-08-18:
I cannot find any previous cases of segfault with SSE other than your own ones at https://github.com/IntelRealSense/librealsense/issues/8031 and here.  SSE support is incorporated into processors and very old CPUs may not have support for it, but your i7-6600U is new enough to have SSE.

https://ark.intel.com/content/www/us/en/ark/products/88192/intel-core-i7-6600u-processor-4m-cache-up-to-3-40-ghz.html

[image]

I note though that you have not defined the **points** reference at the start of your test script:

rs2::pipeline p;
rs2::pointcloud pc;
**rs2::points points;**

banuprathap, 2021-08-24:
Thanks for highlighting the rs2::points missing in my snapshot. I verified it is declared in our application code as shown above.

I can see the SSE supported at OS level in the contents of `/proc/cpuinfo` file.

`
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1 dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
`

I'm not sure what else I can try next. Any chance we can escalate this to dev/engineering group?

MartyG-RealSense, 2021-08-25:
The SDK made changes in **2.45.0** that added unit tests for SSE.  If you build SDK **2.44.0** and your program does not have this problem in that SDK version then it may suggest that the changes that were introduced in 2.45.0 are causing your problem.

[image]

Your case is the only one that I know of where this issue has occurred though, there have been no reports of similar problems since 2.45.0 was introduced and you are using an  i7-6600U Intel processor that should support SSE instructions.

I wonder if use of Ubuntu 14.04 is a factor, since very few people use it for RealSense projects.

MartyG-RealSense, 2021-08-31:
Hi @banuprathap  Do you require further assistance with this case, please?  Thanks!

banuprathap, 2021-09-01:
Hi Marty, As you can see in my previous reporting of the issue #8031, the issue was observed in `2.13.0` as well. I also noticed the issue with `2.40.0`.

I agree that the 14.04 OS and 4.4 kernel might have something to do with the issue but I don't understand how using a different USB backend will effect the result for pointcloud generation step.

Please provide any insights that intel dev team might have on this. I'm happy to provide more data and/or run testing as needed.

MartyG-RealSense, 2021-09-02:
I would not expect to see significant differences between a kernel-patched librealsense and an RSUSB-built one in regard to point cloud generation.  What RSUSB **can** do is bypass problems that are caused by conflicts with the Linux version or with the kernel.

banuprathap, 2021-09-02:
Thanks for clarifying about the pointcloud generation. Look forward to the inputs from the dev team.

MartyG-RealSense, 2021-09-02:
I would recommend checking first whether building librealsense with RSUSB can resolve the SSE problem to your satisfaction.  Given how you are using Ubuntu 14.04 and kernel 4.4, the development opinion may be to update to a newer Ubuntu version.  I understand though that this may not be possible for your particular project.

closed 2021-09-19 by MartyG-RealSense