realsenseai/librealsense #9648: RealSense production shutting down - Long Term Support

sam598, 2021-08-17:
Several news outlets have reported that Intel's RealSense division is beginning the process of shutting down and ceasing production.
https://www.crn.com/news/components-peripherals/intel-says-it-s-shuttering-realsense-camera-business
https://www.engadget.com/intel-realsense-shutdown-211525552.html
https://www.theverge.com/2021/8/17/22629528/intel-realsense-3d-camera-tech-wind-down-business-product-demos

If this is true what are the end of life/long term support plans?

Will the shipments in October be the final batches of cameras manufactured?

csr-kick, 2021-08-18:
Completely expected based on their technical support, and it is too good of an idea to survive being at Intel. Too bad they won't just spin it off into its own business. Robotics takes a huge step backwards.

Chris45215, 2021-08-19:
This is horrible news for us, as we just added support for RealSense to our product and are in the process of scheduling early rollouts. It means at least few months of work wasted. I was on the verge of ordering a handful more, to ensure we will have a supply on hand for installations, but that's now in limbo. We can do the planned installs with the Realsense (assuming we can get the sensors), as we've focused so much of our work towards them, but it is a major setback nonetheless.

agrunnet, 2021-08-19:
I can share this more detailed communication from Intel:  It is true that Intel has decided to wind down the "RealSense" business unit, and is announcing the EOL of LiDAR (L515), Facial Authentication (F450) and Tracking (T265) product lines this month. For these specific products, we will provide a 6 month EOL and Last Time Buy period for these EOL products.  

However, as for the Stereo Product Line, Intel WILL continue to provide the following stereo products. The following Stereo products will be continue: D410, 415, 430, 450, modules and D415, 435, 435i integrated product lines. D455 will be EOL’ed (but not the module).

agrunnet, 2021-08-19:
We are also continuing the open source SDK, as well as continuing to expand the capabilities.

gachiemchiep, 2021-08-20:
Hello @agrunnet 
What does this mean: "D455 will be EOL’ed (but not the module)." 
will D455 be EOL ???

agrunnet, 2021-08-20:
Yes. But a new version will come…

gachiemchiep, 2021-08-20:
@agrunnet 
Thanks.

doisyg, 2021-08-20:
Thanks for the info @agrunnet , so the end of production of  D410, 415, 430, 450, modules and D415, 435, 435i integrated product lines is not planned yet ?

agrunnet, 2021-08-20:
Correct. Those continue. We are building them as fast as we can.

dsvua, 2021-08-20:
D455 will be discontinued as I see, but will be there replacement camera with global shutter color sensor? Since D435 has rolling shutter sensor.

csr-kick, 2021-08-20:
> We are also continuing the open source SDK, as well as continuing to expand the capabilities.

Is there any chance of open-sourcing the T265 firmware? You guys did an amazing job solving an incredibly hard robotics problem, but doomed the product through a USB flaw. My company would have bought hundreds of T265s had someone actually addressed that one issue. I have sufficient work-arounds in place now that I can use a few for hobby-level projects, but I would love to actually try and fix it someday.

max-verem, 2021-08-22:
Is it possible to use D435i with **Intel® Neural Compute Stick 2** to fully emulate T265 ? or may be a T265's firmware port exist for **Intel® NCS 2** ?

> Is there any chance of open-sourcing the T265 firmware?

i would be happy if it got ported to librealsense library/sdk

RealSenseSupport, 2021-08-24:
Correction: We will continue to provide current D455 camera.

jonpol01, 2021-08-28:
@RealSenseSupport  may I ask until when?? or are you guys just emptying the pile out of the barn then call it off once all sold??
TBH at this moment, it is a **big risk** investing time on any realsense product line without assurance of continuous and "real" person behind the support desk.

*cross finger* reaslsense sdk team is not down to 10 man person right?? rofl
**Good thing we abandoned realsense since the damned r200 bugs.. we use Zed2 cameras**

hernot, 2021-08-28:
Does the shutdown of the tracking products also affect the short range coded light SR300 products (no explicit mention in any announcement, statement and post) or will we have to start searching for a replacement. The SR300 coded light series offers a depth resolution about 1mm and less compared to D400 wit several mm depth resolution. Resolution of about 1mm and less is crucial for localizing electrodes on the patients torso for inverse bioelectric mapping applications. 
Or will there be shortrange stereo devices offering depth resolution comparable to SR300 coded light products or will there be even an update of the rather aged series with up to date IR depth and color sernsors?

max-verem, 2021-08-28:
> **Good thing we abandoned realsense since the damned r200 bugs.. we use Zed2 cameras**

are you using that as *replacement* for **T265**?

mdisg, 2021-09-06:
@RealSenseSupport what will happen to the RealSenseID F455 Firmware? Will it receive support/fixes or not? Would also be good to have the FA available for the remaining cameras in host mode.

mcgrathm, 2021-09-25:
Hi @RealSenseSupport. As with many have a product using a T265 almost ready for release.  If Intel is not continuing production will they release teh Myriad V-SLAM library source so we can roll our own with teh D4XX family we are also using?  Note: There were suggestopns a year ago that a D435i  camera and internal IMU based V-SLAM may have been made available?

closed 2022-01-26 by RealSenseCustomerSupport

karrson, 2023-06-03:
Release the damn Myriad V-SLAM library, intel. Don't make us pay for this twice!

MartyG-RealSense, 2024-02-25:
Hi @chilikenji  That game was released in 2015, so it likely only works with the original **RealSense F200** camera model from that year and not any modern RealSense models.  The source code would also only be relevant to the RealSense SDK software from that year, which is incompatible with the modern SDK.

https://jennyatkins.me/move-it