realsenseai/librealsense #9653: Best RealSense replacements?

pkrush, 2021-08-18:
It's a huge issue that RealSense is now gone. What other lightweight depth cameras and lidars do I build my career on now after Intel decided to shut down RealSense?  

What else is lightweight, low cost, available, and somewhat mature? 
 
Velodyne Velabit has been vaporware. Microsoft seems like a good idea with a 10 billion contract with the Hololens, but with Microsoft Kinect DK, they are now saying Oct 2022 for shipments, and the Kinect DK is too heavy anyway. Oak AI uses Intel Movidius, and I just can't trust Intel after buying and basically killing Movidius, and now killing off Realsense. Livox lidars (Now owned by DJI) are awesome and low cost, but too heavy. 
 
Anyone else have experience with all the other stereo camera companies or lightweight lidar startups out there?

pkrush, 2021-08-18:
This issue I posted made the front page of hacker news and people are giving advice there: 
https://github.com/IntelRealSense/librealsense/issues/9653
 
I hate to bash Velodyne on the Velabit, as this has to be a CoVid19/Chipageddon thing. 
 
ZED Stereolabs is popular. Any comments? 
 
If you are a manufacturer and you’re really serious about putting stereo vision into your product the Embedded Vision Summit is a great place to learn that there are no great baby step routes to do this, like Realsense. It’s a place to learn that you have to spend 5+ digits to buy by IP for a custom ASIC. inuitive-tech.com and admarila? (I spelled this wrong). OK, this is 3 year old advice, but I am guessing it’s still true. 
 
I need to look again at all the companies including Structure
 
I was told to look at https://www.ids-imaging.us/ensenso-stereo-3d-camera.html and  devices based on the Infineon lidar, such as the pieye Nimbus 3D camera

ThatGeoGuy, 2021-08-18:
Full disclosure: I used to work for Occipital from 2018 to 2020. I also maintain the present [Realsense Rust](https://gitlab.com/tangram-vision-oss/realsense-rust/) package. Lastly, I'm pulling a lot of this from my current employer's [industry map](https://www.tangramvision.com/blog/the-2021-perception-sensor-industry-map#depth).

This is a hard issue, and I think what you're looking for is going to depend on your use-case. Especially so given some of your reasoning. Anyways, I'll dig into some alternative options in a bit more detail:

1. Don't leave RealSense. Any hardware you have still works. The obvious and immediate trade-off is that firmware updates won't continue, and that someone will need to continue to build & maintain librealsense as time goes on. If you're just looking to maintain existing sensors for small projects or exploratory purposes, this is probably okay. However, if you look at the existing issue list, you might find that somewhat of a dealbreaker that support won't be forthcoming.
2. [Structure Core](https://structure.io/structure-core). Again, see my disclaimer above. This is the closest thing on the market today to a D435i. However, the drawbacks are that the SDK is not open-source as librealsense is, and there will be a learning curve associated with learning Occipital's Structure SDK as well (although, that's going to be true regardless of the sensor you choose here). Additionally, the sensor is more expensive than a D435i but it does cross off a lot of your needs specifically: it doesn't use an Intel ASIC and does not weigh much differently than a D435i. 
3. [Luxonis OAK-D](https://www.luxonis.com/). You did mention that you didn't want to use OAK AI, which I presume means this, but I'm not sure. It's gotten a fair bit of attention recently though, and the hardware doesn't seem on its way out in the next few years so I don't see why it would be a huge drawback. 
4. [PMDTech](https://pmdtec.com/en/). If you're looking specifically to replace ToF sensors like the L515 you might be able to find something here. Unfortunately they don't have an SDK at all, much less an open one like librealsense, so I'm not  sure what the software story is there.
5. [Orbec](https://orbbec3d.com/). Their SDK exists, is not open, and is only C++11 as far as I know. That may be a game-changer if you're using librealsense for language support. I'll admit I haven't done extensive review on Orbec, but I can attest that peers have said good things about them. 
6. ZED Stereolabs, as previously mentioned. I've also never used these and don't know much about them. If I had one on hand I could let you know my impressions, but I regrettably don't.

Those are the top 5 current options as far as I can tell, to replace something like the D435 or D435i. The topic is right: this segment of the market is pretty harsh and there will definitely be a gap with regards to RealSense disappearing. I'd make a pitch here for my current employer ([TangramVision](https://tangramvision.com)) to say that we actually want to solve this on the software end (making it possible to use all these disparate sensors in a single system), but I don't want to turn this into a direct ad for myself, and I couldn't make any promises regardless.

Luxonis-Brandon, 2021-08-18:
Thanks @ThatGeoGuy for the thorough list here!  

Also in full disclosure - I am the founder/CEO of Luxonis.  And actually our advisor is Vikas Reddy - co-founder of Occipital (small world).

I wanted to chime in that we're actually integrating PMDTech's ToF into OAK (example of it working below) so that our DepthAI open-source (MIT-Licensed) ecosystem can be used - which addresses the SDK aspect of PMDTech, and allowing an L515-type solution in the OAK ecosystem - using the same API/etc.  So this will soon allow using the same API/etc. with passive stereo depth, ToF, and/or active stereo depth (laser dot projector and IR flood illumination).

Below is one of the PMD ToF sensors working on DepthAI OAK-FFC-3P:
[image]

And @pkrush - I understand the concern WRT things being killed off.  Nervana was also killed off ([here](https://www.networkworld.com/article/3519354/intel-ends-nervana-ai-chip-in-favor-of-habana-labs-product.html)) but didn't make as much news (as it was used in a lot fewer things).

Fortunately Luxonis' has your back should you use our tech.  We guarantee availability of the OAK-D series through 2028.  (And likely well past then, but at a bare minimum through 2028).  So very likely our current series will be only used in "Wow, that's legacy!" by the time we do a last-time buy/build (which we will do - to give even more continuity). 

Thanks,
Brandon

scaldara55, 2021-08-19:
Realsense is not getting out of the business completely. They are restructuring and dropping the LiDAR development. They will continue with their more popular cameras.

Luxonis-Brandon, 2021-08-19:
> Realsense is not getting out of the business completely. They are restructuring and dropping the LiDAR development. They will continue with their more popular cameras.

Ah - good to know @scaldara55 !  I must have misread the articles about this.

sam598, 2021-08-19:
> They will continue with their more popular cameras.

Do you have a source for that? All of the articles I’ve read say that the development team is being disbanded and moved elsewhere within Intel.

rzw2, 2021-08-19:
At Chronoptics we design iToF depth cameras, we are developing a camera that has depth processing on board (unlike the Azure Kinect where you need an Intel i5 equivalent), with VGA resolution and up to 45fps (depth). Checkout our example filtering to clean up iToF point clouds, https://www.youtube.com/watch?v=MPbTN6AwNtA

TomasMichalik, 2021-08-19:
@pkrush it would be great to hear more about your use case. I understand you are looking for something within the boundaries of "lightweight, low cost, available, and somewhat mature" but it might be worth checking out the structured light technology on steroids: https://www.photoneo.com/motioncam-3d/
It combines the best from both worlds of ToF and Structured light.
Review here:
https://www.youtube.com/watch?v=9lsr7aTMHcs

civerachb-cpr, 2021-08-19:
I generally agree with everything in https://github.com/IntelRealSense/librealsense/issues/9653#issuecomment-901387674

My own thoughts and recommendations based on my experiences working with various depth-sensing cameras:

1) In the short-term, RealSense is still around and your existing hardware isn't going to suddenly evaporate.  Anyone with an existing camera can probably be assured of several more years of use out of the hardware they've already purchased.  Companies that use RealSense in products (e.g. Xaomi with their robot dog) are in more of a bind though, so if you fall into that camp keep reading.

2) [The Microsoft Azure Kinect](https://azure.microsoft.com/en-us/services/kinect-dk/) isn't widely available, but is pretty good.  The form-factor is a little funky (I always thought it looked like a nice camera with someone's grad-school project bolted to the back of it), but performance-wise I've never had problems.  It's also supported by ROS on both Linux and Windows, which is a major benefit if you're using RealSense on robots.

3) [Orbbec Astra](https://orbbec3d.com/). (Disclosure: a company I used to work for had a demo in Orbbec's booth at CES a few years ago and have interacted with a lot of the devs in-person.  They're nice people!)  These cameras have been around longer than the RealSense D4xx family.  Unlike RS, Astra works over USB2, which isn't a huge deal but means it still works pretty well on older/low-spec hardware.  Personally, I like the Astra Mini with their Duripod enclosure, as it's a little more rugged and has better mounting options than the standard Astra.

4) [StereoLabs Zed/Zed2](https://www.stereolabs.com/zed/).  The biggest down-side to these cameras is that they need a GPU that supports CUDA to work at all.  But they work well and are pretty reliable, especially at longer ranges.

5) [Mynt Eye](https://www.mynteye.com/).  I'm a little surprised these haven't been mentioned already.  Technologically they seem pretty similar to the RealSense.  I unfortunately don't have as much personal experience with these cameras (yet), but from everything I've read I'd put them near the top of my list of alternatives to try out.

6) [Luxonis OAK-D](https://store.opencv.ai/products/oak-d).  (Disclosure: I backed the OAK-D on Kickstarter.)  If all you need is a camera with RGB + Depth sensors, the OAK-D is probably overkill.  But the fact that it can run object-detection & classification on the sensor itself I think makes it a compelling alternative for specific applications.

Obviously if all you need is the depth data and you don't care about the RGB component, there are lots of 3D lidars out there.  But most are pretty expensive and I wouldn't consider them a good replacement for the RealSense for most users.

pkrush, 2021-08-19:
Thank you Chris for yet another awesome qualified response. I have learned so many little details with all the discussion that is going on. Details that you are not going to get from the manufacturers websites. 
 
There also is a great discussion going about this on the ROS boards:
https://discourse.ros.org/t/intel-cancelling-its-realsense-business-alternatives/21881
 
Those REAL3 Infineon TOF sensors look like they might work for me. It’s nice that they work in the sun. I ordered the PMD Pico Flexx reference board for the Infineon sensor. My use case needs something like what I am getting after downsampling the D455 data stream. I need under 3 watts, under 30 grams. 
 
I am still hoping this is just fake news or a manager talking out of line and RealSense announces something. It’s annoying that all the articles all lead back to one person talking unnamed to crn.com(with a CNN looking logo). I would blow it off if it weren't for all of the industry hearsay.

scaldara55, 2021-08-20:
> > Realsense is not getting out of the business completely. They are restructuring and dropping the LiDAR development. They will continue with their more popular cameras.
> 
> 
> 
> Ah - good to know @scaldara55 !  I must have misread the articles about this.

I get the sarcasm. You did not misread the articles. They are all sourced off of the one erroneous press release. In time the correct info will come out.

pkrush, 2021-08-20:
OK,  RealSense CTO put a comment in #9648. Thanks!  Yes, stereo cameras are mostly sticking around. This addresses my issue and I call this closed.

closed 2021-08-20 by pkrush

scaldara55, 2021-08-20:
> > They will continue with their more popular cameras.
> 
> 
> 
> Do you have a source for that? All of the articles I’ve read say that the development team is being disbanded and moved elsewhere within Intel.

Yes I have a good source. All of the articles are based off of the same erroneous press release. RealSense is being restructured. LiDAR is EOL. Same with some application products and certain cameras. The remaining team will focus on cameras for the AMR market.

Luxonis-Brandon, 2021-08-20:
> I get the sarcasm. You did not misread the articles. They are all sourced off of the one erroneous press release. In time the correct info will come out.

I wasn’t being sarcastic.  I figured you must have known something I didn’t - which it turned out was the case.  I should have phrased it differently sorry - I can definitely see how that can off as sarcastic.

ddiakopoulos, 2021-08-20:
The CTO of the RealSense group, Anders Grunnet-Jepsen posted this updated a short while ago

[image]

odubovsky, 2021-08-20:
Hello everyone,
My name is Oded Dubovsky and I am the SW Director of Intel® LibRealSense.
I am posting the following to help alleviate a lot of the stress that the press announcement has caused.

It is true that Intel has decided to wind down the "RealSense" business unit, and is announcing the EOL of LiDAR (L515), Facial Authentication (F450) and Tracking (T265) product lines this month. For these specific products, Intel plans to provide a 6 month EOL and Last Time Buy period for these EOL products. 

However, the Stereo Product Lines WILL continue with the following products: D410, D415, D430, D450 modules and D415, D435, D435i cameras to our current distribution customers (new distributors will be evaluated on a case by case basis).  Support will be continued for the following Stereo products: D410, 415, 430, 450, modules and D415, 435, 435i integrated product lines. We will also continue the work to support and develop our LibRealSense open source SDK.

In the coming future Intel and the RealSense team will focus our new development on advancing innovative technologies that better support our core businesses and IDM 2.0 strategy.

ericycutlip, 2021-08-21:
> Full disclosure: I used to work for Occipital from 2018 to 2020. I also maintain the present [Realsense Rust](https://gitlab.com/tangram-vision-oss/realsense-rust/) package. Lastly, I'm pulling a [...]

Hi, PMD has an excellent SDK available to include a C++ library and Python wrapper. There is a development kit with SDK available today from Emcraft (https://emcraft.com/products/1111). Later this year, there will be a USB development kit called the PicoFlexx 2 (to replace the now EOL PicoFlexx) with SDK for use with Windows/Linux/Android/Mac OS.

scaldara55, 2021-08-22:
> > > > Realsense is not getting out of the business completely. They are restructuring and dropping the LiDAR development. They will continue with their more popular cameras.
> 
> > > 
> 
> > > 
> 
> > > Ah - good to know @scaldara55 !  I must have misread the articles about this.
> 
> > 
> 
> > I get the sarcasm. You did not misread the articles. They are all sourced off of the one erroneous press release. In time the correct info will come out.
> 
> 
> 
> I wasn’t being sarcastic.  I figured you must have known something I didn’t - which it turned out was the case.  I should have phrased it differently sorry - I can definitely see how that came off as sarcastic. 

NP, sorry to misinterpret your comment as sarcasm. Given the circumstances I was prevented from giving real details. I'm glad that Ander's (CTO) has weighed in.

nerian-vision, 2021-08-31:
We ([Nerian Vision](https://nerian.com)) continue to manufacture our line of real-time stereo vision sensors. There is also a low-cost sensor in the planning that will be released mid/end next year and should be a great replacement for realsense (more details will follow).

Our current sensors are:

- [Scarlet](https://nerian.com/products/scarlet-3d-depth-camera/): A high-performance depth camera with 5 megapixel resolution and up to 120 fps
- [SceneScan](https://nerian.com/products/scenescan-stereo-vision/): A stand-alone stereo image processing system for highest flexibility with off-the-shelf USB3 Vision cameras our our own Karmin3 stereo camera

[Example Video](https://youtu.be/r0pECtAApzI)

[image]
[image]

FPSychotic, 2022-04-15:
For ROS, Except if something new comes, or you have unlimited resources, time, and amazing ROS  and programming skills and a heart armoured against  frustration and smoke given from companies.
Stick to d435i (no other, no d455 or any other) or Livox Avia.
After extense use of all of them, any other thing, in a way in other, is put your money in the bin, and worse , your time.
Specially true for the newer released stereo cameras and brands.
D435i, Livox Avia or both at same time.
I could make a long long list of why, but it is lose the time. Talking about ROS.

s-trinh, 2022-04-15:
@FPSychotic 

I am curious to know why you would not recommend the d455 (and rather stick to the d435i)?

From what I have gathered, the d455 has a bigger baseline and is also global shutter. But looks like you would need to add a filter if you want to use the d455 outdoor.

FPSychotic, 2022-04-15:
Because it is a faulty product that should be recalled because issues of quality with the IR filter. They invented later to avoid recall , complaints returns, a PDF document with a filters study and other bullshit like that. 
Don't fall in that, as said , it is bullshit, the camera is not made to add or attach any filter and they change other params that make it unreliable and apart even with the recommended filters still have issues.
I returned 2, problably the d455 issues, after the failed t265 (almost faulty too,with the USB recognition issue and wheel odometry input?) had some weight in the decision. In fact, they said d455 would be EoL, not sure if finally it is.

https://youtu.be/LzL1Vo-Bow8

Check not only the redish colourization, check the Frame rate how fall, how get crazy the exposure control.

For ROS, d435 and Livox AVIA (which software and support is not nice, in fact barely acceptable, quite bad in fact, but works, and will work, lidars are simple). The rest, any other brand said here, or sell at double price that should, sell smoke, program the obsolescence , or give the half of features, or the half of possible uses, or is usable in the half of environments than a d435. You want wider FOV, get 2 d435i. Do you want stereo camera, turn of the IR laser. Do you want depth + AI in ROS, get a d435 + coral, or nano, or xavier.
Mynteye is a great camera, and right priced, but d435 is more usable.

s-trinh, 2022-04-19:
For further reference, the topic mentioning reddish color image on the **d455**: https://github.com/IntelRealSense/librealsense/issues/7155

A topic mentioning some IR filter for the **d435**: https://github.com/IntelRealSense/librealsense/issues/2875

---

On my experiments with the **d455**, RGB images are reddish but the depth reconstruction is ok:
[image]

Some testing at night:
[image]

It is hard to tell if the stereo reconstruction is correct or not without some reference data.

FPSychotic, 2022-04-20:
Stereo and depth is nice in d455, and the color os ok en some environments. The problem is the color at daylight or with some lights in indoors.
It is not the reddish only, the sensor gets saturated and exposure ,fps won't work as should be. Imagine colourize a pointcloud with that, or ML CV colour recognition

RanpakHoward, 2022-10-05:
> @pkrush it would be great to hear more about your use case. I understand you are looking for something within the boundaries of "lightweight, low cost, available, and somewhat mature" but it might [...]

Cannot agree on this one. Photoneo has fantastic cameras but not the price lol. It's 20+ times what Realsense series cost... (40 times the original L515 cost)

ebritanica, 2023-03-12:
What difference with D455 and F455? In my application i want simple scan objects to make 3d mesh. Which one to choose?

MartyG-RealSense, 2023-03-12:
Hi @ebritanica  The D455 is the correct camera to choose.  The F455 is for human facial recognition and authentication and is not for object scanning.

ebritanica, 2023-03-12:
> Hi @ebritanica The D455 is the correct camera to choose. The F455 is for human facial recognition and authentication and is not for object scanning.
Yes, but... F455 is locked to 3d scaning objects and meshing them?

MartyG-RealSense, 2023-03-12:
F455 is not compatible with the RealSense SDK software and has its own separate Facial Authentication SDK, so cannot be used for object scanning.  Also, F455 is a retired product and its SDK software is no longer updated, whilst the D455 and the RealSense SDK software continue to be supported and updated.

There are also commercial D455-compatible scanning software packages available for Windows computers such as **DotProduct Dot3D** and **RecFusion**.

With the D455 model, you can take advantage of the open-source **rs-kinfu (KinectFusion)** program that enables 360 degree scanning of an object and export to file with a single camera.  

More information can be found at https://github.com/IntelRealSense/librealsense/issues/9832#issuecomment-936313161

ebritanica, 2023-03-13:
I do not know what it is, but...
https://github.com/luxonis/depthai-hardware

MartyG-RealSense, 2023-03-13:
I am unable to offer advice about Oak cameras or other non-RealSense cameras, unfortunately.

FPSychotic, 2023-03-13:
> I do not know what it is, but... https://github.com/luxonis/depthai-hardware

Clearly you don't have the right info.

bridru14, 2023-10-10:
I work a PreAct Technologies.  If you have not seen the Mojave Near Field Flash LiDAR, it is worth taking a look at the advancements we bring to the market.  

https://www.linkedin.com/posts/preact-tech_robotics-automotive-adas-activity-7111014437267939329-71Ly?utm_source=share&utm_medium=member_ios

Clay-sign, 2024-01-07:
I work with signed languages, the languages of the Deaf community, and I was wondering if I could ask your collective advice – given peoples expertise on cameras and types of motion capture.

Signed languages use movement, hand shape, space, arm, hand, face, eyes, mouth etc – and even body motion.
If you aren’t familiar with a signed language this vid will give you an idea of the dynamic nature of what we’d like to capture as effectively and in as much detail as possible.

https://youtu.be/04tOoXCqyEM?si=pieDljlHV0VNl7nAon. 

Our goal is to create an online dictionary of signs… but we also want to do more in depth linguistic analysis down the road. 
Some sign researchers use Azure Kinect DK, which is how I found a discussion Intel RealSense Depth Camera D455 and then this site.

Given peoples expertise  – what kind of camera or device would you recommend?  We will be using a studio set up with a backdrop for most recording, other times, with focus sessions at Deaf events, it may be more uneven lighting – but with a backdrop. We’d like to capture as much fine-grained data as possible.   

We don’t have an unlimited budget – but any suggestions would be greatly appreciated.
Thanks.

FPSychotic, 2024-01-07:
For gestual recognition for sure kinect is the right tool.
I don't think any other Realsense has the depth resolution that task needs.
Orbbec is making them now after Microsoft deprecated them, I would use the
Orbbec one, it has a jetson nano inside, which would allow you some AI
support maybe useful for that task as background removal.
Orbbec has also some high definition/precision cameras but they are close
range, maybe good for a videoconferencing setup, but probably can't to work
in full body , or room setup.
I don't think there is any any Realsense suitable to substitute a Kinect
Azure

Nir-Az, 2024-01-07:
@Clay-sign / @FPSychotic 
RealSense cameras does have experience with hand gestures, I advise you to search about it on the issues and get demo links. 
About resolutions checkout the [D455 camera](https://www.intelrealsense.com/depth-camera-d455/) which has similar resolutions as the Orbbec camera

FPSychotic, 2024-01-07:
No, they doesn't

Clay-sign, 2024-01-07:
FPSychotic thank you for your reply, suggestions and for the reply to Nir-Az.  And Nir-Az thanks as well.  I checked out the specs on the orbbec Persee N1 and it looks promising.   I will need to contact thier support for more details.  The Azure Kinect also seems to tick all the boxes as well.  I need to capture finger, hand, chest, arm, and facial gestures.  and torso movments as well as eye gaze. Signed langauges are complex and the more we can capture the better for post analysis.   Yes, we will start with the sign but being able to do the lingusitic analysis down the road will be just as important.

odubovsky, 2024-01-07:
Clay-sign, as Nir-Az suggested, RealSense cameras have a wide variety of devices that can likely support such use cases. Based on our experience, it is also very easy to set up and alter with the support of LibRS open source SDK. For your consideration.

Nir-Az, 2024-01-14:
> Anyone know some good alternatives to the F455 ? I'm doing facial authentication and it seemed like the perfect device, only for me to learn that it's discontinued.

Hi @mateen-demah ,
Actuality F455 is back, 
checkout the website: https://www.intelrealsense.com/get-started-intel-realsense-id/

mateen-demah, 2024-01-14:
Thanks @Nir-Az .

I actually found that site before I learned it was discontinued so I had assumed they were seeking off devices they had in stock.

Nir-Az, 2024-01-14:
I am happy to say that this is not the case, the product came back to life :)

mdisg, 2024-03-14:
@Nir-Az what do you mean with "the product came back to life"? The thing I see is that the page changed mentioning a license which was not there in the old version of the page. 

There is no way to buy it on the intel store and on 04.03.2024 the customer support said to me that the product is still discontinued. This is what I got from a response to a ticket.

`For your information, Facial Authentication camera (F455) and module (F450) has been discontinued since February 2022. Hence, we have moved our focus to the Stereo products (D400 series camera) and, consequently, we will not be addressing this issue in the F455.`

mdisg, 2024-03-14:
@MartyG-RealSense thanks for that hint. Can you provide a link to the published email?

MartyG-RealSense, 2024-03-14:
https://app.plan.intel.com/e/es?s=334284386&e=17755219&elqTrackId=64d1a1cb1dbd4944a4bd08861778a285&elq=c7ec4d7d4b83400c90f5b13584ce2bd5&elqaid=101927&elqat=1

mdisg, 2024-03-14:
@MartyG-RealSense thank you. We will keep an eye on that webinar

MartyG-RealSense, 2024-04-07:
Hi @mdisg  The RealSense F450 and F455 facial ID products are now listed in the official RealSense Store.

https://store.intelrealsense.com/

mdisg, 2024-04-08:
@MartyG-RealSense thanks I already saw that a new firmware and software was release earlier.

Laitaps, 2024-07-17:
> For gestual recognition for sure kinect is the right tool.
> 
> I don't think any other Realsense has the depth resolution that task needs.
> 
> Orbbec is making them now after Microsoft deprecated them, I would use the
> 
> Orbbec one, it has a jetson nano inside, which would allow you some AI
> 
> support maybe useful for that task as background removal.
> 
> Orbbec has also some high definition/precision cameras but they are close
> 
> range, maybe good for a videoconferencing setup, but probably can't to work
> 
> in full body , or room setup.
> 
> I don't think there is any any Realsense suitable to substitute a Kinect
> 
> Azure
> 
> 
> 
> 
> 
> 
> > I work with signed languages, the languages of the Deaf community, and I
> 
> > was wondering if I could ask your collective advice – given peoples
> 
> > expertise on cameras and types of motion capture.
> 
> >
> 
> > Signed languages use movement, hand shape, space, arm, hand, face, eyes,
> 
> > mouth etc – and even body motion.
> 
> > If you aren’t familiar with a signed language this vid will give you an
> 
> > idea of the dynamic nature of what we’d like to capture as effectively and
> 
> > in as much detail as possible.
> 
> >
> 
> > https://youtu.be/04tOoXCqyEM?si=pieDljlHV0VNl7nAon.
> 
> >
> 
> > Our goal is to create an online dictionary of signs… but we also want to
> 
> > do more in depth linguistic analysis down the road.
> 
> > Some sign researchers use Azure Kinect DK, which is how I found a
> 
> > discussion Intel RealSense Depth Camera D455 and then this site.
> 
> >
> 
> > Given peoples expertise – what kind of camera or device would you
> 
> > recommend? We will be using a studio set up with a backdrop for most
> 
> > recording, other times, with focus sessions at Deaf events, it may be more
> 
> > uneven lighting – but with a backdrop. We’d like to capture as much
> 
> > fine-grained data as possible.
> 
> >
> 
> > We don’t have an unlimited budget – but any suggestions would be greatly
> 
> > appreciated.
> 
> > Thanks.

A CNN based sign language recognition system would be really awesome.

TomasMichalik, 2024-07-17:
Photoneo MotionCam-3D can do highly accurate 3D scanning of moving objects, even with the color point cloud. Nvidia onboard. https://www.photoneo.com/motioncam-3d/

alexeden, 2024-09-05:
> Photoneo MotionCam-3D can do highly accurate 3D scanning of moving objects, even with the color point cloud. Nvidia onboard. https://www.photoneo.com/motioncam-3d/

I'm sure they do. But folks are looking for an alternative to RealSense, insofar as it is/was a consumer, price-transparent product. Any product whose page's CTA says "Get a Quote" is not at all a consumer, available product.

janzontsing, 2026-10-17:
Another option to consider is DOMI’s [[DM-RGBD5002A](https://domisensor.com/products/dm-rgbd5002a)](https://domisensor.com/products/dm-rgbd5002a) and [[DM-RGBD5003A](https://domisensor.com/products/dm-rgbd5003a)](https://domisensor.com/products/dm-rgbd5003a), which combine ToF depth sensing with RGB imaging.

Both models offer:
- Depth up to 640 × 480 at 25 fps
- RGB at 1280 × 960 at 30 fps
- A working range of 0.2 to 5 m
- USB 2.0 connectivity
- Dimensions of 79 × 35 × 15.7 mm

For projects using point clouds or ROS, DM-RGBD5003A also offers X/Y/Z point-cloud output, depth-to-RGB registration, and C/C++, Python and ROS 1 / ROS 2 integration options.

These specifications make them worth evaluating for robot perception and object measurement when the required range, field of view and frame rate fit the application. Existing RealSense software would need adaptation to the DOMI camera interface.

DOMI’s [[RealSense alternatives guide](https://domisensor.com/product-insights/realsense-alternatives-with-domi-vga-tof-rgb-d-cameras)](https://domisensor.com/product-insights/realsense-alternatives-with-domi-vga-tof-rgb-d-cameras) includes D435/D455 specification comparisons and software migration considerations. For the original weight and cost requirements, the next useful details would be unit weight, evaluation pricing and current lead time.