realsenseai/librealsense #9654: L500 Depth Sensor Malfunctioning

hunterlineage1, 2021-08-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L500 (L515) } | 
| Firmware Version                   | 01.05.08.01| 
| Operating System & Version |   {Linux (Ubuntu 20.04 LTS)  | 
| Kernel Version (Linux Only)    |  (5.8.0-43-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            | N/A                          | 
| Language                            |  python                          | 
| Segment			| Robot                       | 

### Issue Description
I am trying to use my L515 depth sensor. When I try to toggle on the 'L500 Depth Sensor' button in the realsense viewer, I get the following error:
[image]

my /dev/ directory looks like this. Essentially there are two 'iio:device0' and 'iio:device1':

[image]

When I enter 'v4l2-ctl --list-devices' I get:

Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-5):
	/dev/video0
	/dev/video1

Intel(R) RealSense(TM) 515: Int (usb-0000:39:00.0-1):
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8
	/dev/video9
	/dev/media1
	/dev/media2

What Can I do to fix this urgently?

stevemartinov, 2021-08-19:
Man haven't you heard the news? Realsense is dead, no one will probably reply here from Intel

MartyG-RealSense, 2021-08-19:
Hi @hunterlineage1  As a starting point in investigating your case, could you try unplugging the USB cable from the base of the L515 (not at the computer end), turning it around to the opposite orientation and re-inserting it (USB-C cables are two-way insertion at the micro-sized end).  Does this make the error disappear, please?

hunterlineage1, 2021-08-19:
> Hi @hunterlineage1 As a starting point in investigating your case, could you try unplugging the USB cable from the base of the L515 (not at the computer end), turning it around to the opposite [...]

Hi @MartyG-RealSense, I tried that, and got the same error. I looked at issue [#9374](https://github.com/IntelRealSense/librealsense/issues/9374). Some questions I will answer before you ask so that we can speed this up:

The firmware update did not coincide with the start of the problem -- I was using 1.5.8.1 this whole time. 
The RGB frame works perfectly. I doubt this problem is occurring only in the realsense-viewer because I tried to run a python script with openCV and the same input/output error occurs. 

Please help urgently.

MartyG-RealSense, 2021-08-19:
I wonder if it is getting confused by the detection of the integrated HD webcam that is claiming video0 and video1 (and counting it as a RealSense device, hence the 2 detected cameras).  

On a RealSense 400 Series camera at least, video0 would be used by the depth stream, video1 by infrared and video2 by RGB.  So if video2 is available to the RealSense camera but video0 and video1 are not, that could explain why RGB works but depth and infrared do not.

Could you try loading the uvcvideo module with the command below to see whether it corrects the problem please:

**sudo modprobe uvcvideo**

hunterlineage1, 2021-08-19:
> I wonder if it is getting confused by the detection of the integrated HD webcam that is claiming video0 and video1 (and counting it as a RealSense device, hence the 2 detected cameras).
> 
> On a RealSense 400 Series camera at least, video0 would be used by the depth stream, video1 by infrared and video2 by RGB. So if the /video channels are similarly enumerated for L515, and video2 is available to the RealSense camera but video0 and video1 are not, that could explain why RGB works but depth and infrared do not.
> 
> Could you try re-loading the uvcvideo module with the command below to see whether it corrects the problem please:
> 
> **sudo modprobe uvcvideo**

Hi @MartyG-RealSense I tried entering 'sudo modprobe uvcvideo'. The same error occurred. I then entered 'sudo modprobe -r uvcvideo' to disable my integrated webcam. The realsense-viewer then showed "Found 1 RealSense devices (mask 0xfe)" instead of showing 2 realsense devices.

However, the same error was encountered when I tried to activate the depth sensor:

types.h:310 - xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error

Please help immediately.

MartyG-RealSense, 2021-08-19:
Because of the symptoms in this case, I have referred it to Intel to seek advice from them.  I recommend putting the camera aside for now as there is nothing else that can be done in the meantime.

hunterlineage1, 2021-08-19:
> Because of the symptoms in this case, I have referred it to Intel to seek advice from them. I recommend putting the camera aside for now as there is nothing else that can be done in the meantime.

Thank you @MartyG-RealSense. Please let me know ASAP what can be done.

MartyG-RealSense, 2021-08-19:
After discussing your case with Intel, they requested that you return your L515 camera to wherever you purchased it from, using the **RMA (Return Merchandise Authorization)** returns process of that supplier.  

Could you also please email me the serial number of your L515 camera.  You can find this number either by using the **Info** option near the top of the options side-panel in the RealSense Viewer to display the camera's information.   The serial number is also printed on a small sticker on the camera's casing.  I can be emailed at:

martyx.grover@intel.com

MartyG-RealSense, 2021-08-31:
Hi @hunterlineage1 Do you require further assistance with making a camera return, please?

MartyG-RealSense, 2021-09-06:
Hi @hunterlineage1  Could you provide an update please about any problems that you are having with returning your camera if you still wish to do so?

hunterlineage1, 2021-09-18:
Hi Marty,

Sincere Apologies for the late reply. I am working out the RMA process with
my admin. I will get back to you as soon as possible.

Sincerely,
Sandilya

MartyG-RealSense, 2021-09-18:
No problem at all, Sandilya - thanks very much for the update!

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense