realsenseai/librealsense #9655: Depth frame transformation  to my world  coordinate system

okhsallam, 2021-08-19:
Dear all, I am working with the D455 to measure the depth of some surfaces, I aligned the camera tilted with 45 degree and captured the ceiling of my room as shown in the image.  Now I want to apply rotation transformation with 45 around y axis  to transform from the camera coordinates system to the world coordinate system to see the ceiling depth has constant. But i cannot find clear docs how to implement the transformation for the depth frame.
I am using python 

Thanks in advance 
[image]

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Wind 10  | 
| Kernel Version (Linux Only)    |  latest)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others}                          |

stevemartinov, 2021-08-19:
Man haven't you heard the news? Realsense is dead, no one will probably reply here from Intel

MartyG-RealSense, 2021-08-19:
Hi @okhsallam  I would recommend limiting the tilt angle to 30 degrees maximum if possible.  Whilst it is possible to tilt a 400 Series camera to 45 degrees, there may be unpredictable negative effects on the depth image beyond 30 degrees tilt.

If rotating the depth image is a required aspect of your project and you cannot change the tilt angle then you could generate a 3D point cloud depth image and then rotate the entire cloud (not just change the angle that the cloud is viewed from) using a process called **affine transform**.  In the RealSense SDK, an affine transform is provided by the instruction **rs2_transform_point_to_point**

Further information about rs2_transform_point_to_point in regard to Python can be found at https://github.com/IntelRealSense/librealsense/issues/5583

If you believe that simply changing the angle that the cloud is viewed from without changing the cloud itself is sufficient then the SDK Python example **opencv_pointcloud_viewer.py** includes code for doing so that you can refer to.  To highlight the parts of the code related to rotation, you can go to the script at the link below and do a browser 'Find' operation for the word **rotation**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

closed 2021-09-01 by MartyG-RealSense